lvgl/lvgl #3244: ESP32 PSRAM configure for LVGL

uldara1, 2022-04-01:
hello sir i use lvgl library with arduino ide 
the problem when i use cavas is show message 

`.dram0.bss' will not fit in region `dram0_0_seg'
region `dram0_0_seg' overflowed by 41512 bytes

 so i want to change to PSRAM but i don't know to change 
thank

uldara1, 2022-04-05:
Dear Sir @kisvegabor 

i got other error when i change configure 
pls check my configure correct or not 

[image]

here is error 
[image]

kisvegabor, 2022-04-05:
You should also set `LV_MEM_POOL_ALLOC` to an ESP-IDF header in which `pc_malloc` is declared.

uldara1, 2022-04-05:
Hello sir @kisvegabor 

thank for reply
i used Arduino ide not ESP-IDF
does it support ?

thank

kisvegabor, 2022-04-05:
I believe so, but I'm not using Arduino, and don't know how to reach this low level function from Arduino.

Zeron, 2022-04-05:
Haven't tried it, but in the Arduino environment ps_malloc is defined in esp32-hal-psram.h

Zeron, 2022-04-05:
I have tested this in Arduino - it is working:

`
/*=========================
   MEMORY SETTINGS
 *=========================*/

/*1: use custom malloc/free, 0: use the built-in `lv_mem_alloc()` and `lv_mem_free()`*/
#define LV_MEM_CUSTOM 0
#if LV_MEM_CUSTOM == 0
    /*Size of the memory available for `lv_mem_alloc()` in bytes (>= 2kB)*/
    #define LV_MEM_SIZE (128U * 1024U)          /*[bytes]*/

    /*Set an address for the memory pool instead of allocating it as a normal array. Can be in external SRAM too.*/
    #define LV_MEM_ADR 0     /*0: unused*/
    /*Instead of an address give a memory allocator that will be called to get a memory pool for LVGL. E.g. my_malloc*/
    #if LV_MEM_ADR == 0
        #define LV_MEM_POOL_INCLUDE <esp32-hal-psram.h>
        #define LV_MEM_POOL_ALLOC ps_malloc
    #endif

#else       /*LV_MEM_CUSTOM*/
    #define LV_MEM_CUSTOM_INCLUDE <stdlib.h>   /*Header for the dynamic memory function*/
    #define LV_MEM_CUSTOM_ALLOC   malloc
    #define LV_MEM_CUSTOM_FREE    free
    #define LV_MEM_CUSTOM_REALLOC realloc
#endif     /*LV_MEM_CUSTOM*/
`

kisvegabor, 2022-04-05:
Thanks, @Zeron!

pete-pjb, 2022-04-07:
This issue appears to be resolved so I will close it now @uldara1 feel free to reopen if the problem persists.

closed 2022-04-07 by pete-pjb

dizcza, 2022-10-20:
> You can use `LV_MEM_POOL_ALLOC ps_malloc(128*1024)` config in `lv_conf.h`.

Now (v9.0.0, master branch) it seems that I need to define it this way:

```c
#define LV_MEM_POOL_INCLUDE     "esp_heap_caps.h"
#define LV_MEM_POOL_ALLOC(size) heap_caps_malloc(size, MALLOC_CAP_SPIRAM)
```

Which works fine. But if I remove `LV_USE_BUILTIN_MALLOC` section in `lv_conf.h`

```diff
-#define LV_USE_BUILTIN_MALLOC 1
-#if LV_USE_BUILTIN_MALLOC
-    /*Size of the memory available for `lv_malloc()` in bytes (>= 2kB)*/
-    #define LV_MEM_SIZE (128U * 1024U)          /*[bytes]*/
-
-    /*Set an address for the memory pool instead of allocating it as a normal array. Can be in external SRAM too.*/
-    #define LV_MEM_ADR 0     /*0: unused*/
-    /*Instead of an address give a memory allocator that will be called to get a memory pool for LVGL. E.g. my_malloc*/
-    #if LV_MEM_ADR == 0
-        #undef LV_MEM_POOL_INCLUDE
-        #undef LV_MEM_POOL_ALLOC
-    #endif
-#endif  /*LV_USE_BUILTIN_MALLOC*/
+#define LV_MEM_POOL_INCLUDE     "esp_heap_caps.h"
+#define LV_MEM_POOL_ALLOC(size) heap_caps_malloc(size, MALLOC_CAP_SPIRAM)
```

I get 

```
In file included from /home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/src/misc/../lv_conf_internal.h:39,
                 from /home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/src/misc/lv_mem.h:16,
                 from /home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/src/misc/lv_mem.c:10:
/home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/src/misc/lv_mem.c: In function 'lv_malloc':
../main/lv_conf.h:54:25: error: implicit declaration of function 'lv_malloc_builtin'; did you mean 'lv_strlen_builtin'? [-Werror=implicit-function-declaration]
   54 | #define LV_MALLOC       lv_malloc_builtin
      |                         ^~~~~~~~~~~~~~~~~
/home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/src/misc/lv_mem.c:71:20: note: in expansion of macro 'LV_MALLOC'
   71 |     void * alloc = LV_MALLOC(size);
      |                    ^~~~~~~~~
../main/lv_conf.h:54:25: warning: initialization of 'void *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
   54 | #define LV_MALLOC       lv_malloc_builtin
      |                         ^~~~~~~~~~~~~~~~~
/home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/src/misc/lv_mem.c:71:20: note: in expansion of macro 'LV_MALLOC'
   71 |     void * alloc = LV_MALLOC(size);
      |                    ^~~~~~~~~
/home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/src/misc/lv_mem.c: In function 'lv_free':
../main/lv_conf.h:56:25: error: implicit declaration of function 'lv_free_builtin'; did you mean 'lv_strlen_builtin'? [-Werror=implicit-function-declaration]
   56 | #define LV_FREE         lv_free_builtin
      |                         ^~~~~~~~~~~~~~~
/home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/src/misc/lv_mem.c:104:5: note: in expansion of macro 'LV_FREE'
  104 |     LV_FREE(data);
      |     ^~~~~~~
/home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/src/misc/lv_mem.c: In function 'lv_realloc':
../main/lv_conf.h:55:25: error: implicit declaration of function 'lv_realloc_builtin'; did you mean 'lv_strlen_builtin'? [-Werror=implicit-function-declaration]
   55 | #define LV_REALLOC      lv_realloc_builtin
      |                         ^~~~~~~~~~~~~~~~~~
/home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/src/misc/lv_mem.c:125:20: note: in expansion of macro 'LV_REALLOC'
  125 |     void * new_p = LV_REALLOC(data_p, new_size);
      |                    ^~~~~~~~~~
../main/lv_conf.h:55:25: warning: initialization of 'void *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
   55 | #define LV_REALLOC      lv_realloc_builtin
      |                         ^~~~~~~~~~~~~~~~~~
/home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/src/misc/lv_mem.c:125:20: note: in expansion of macro 'LV_REALLOC'
  125 |     void * new_p = LV_REALLOC(data_p, new_size);
      |                    ^~~~~~~~~~
```

So it looks that defining `LV_MEM_POOL_ALLOC` is not enough to make lvgl happy, which raises a question: how `lv_conf.h` and configuration in ESP-IDF menuconfig are related? I've always been puzzled. There are things duplicated in `sdkconfig` and `lv_conf.h` and they bear different names...

kisvegabor, 2022-10-23:
Please do not use the `master` branch yet. See the [README](https://github.com/lvgl/lvgl#readme):
[image]

@embeddedt What do you think about changing the default branch to `release/v8.3` and  the default docs version to v8.3?

dizcza, 2022-10-23:
@kisvegabor the question is still valid for v8.3.3

embeddedt, 2022-10-23:
> What do you think about changing the default branch to release/v8.3 and the default docs version to v8.3?

I think it's okay. We just need to be extra vigilant not to merge features to `release/v8.3` since all PRs will get opened against that branch by default.

kisvegabor, 2022-10-25:
Meanwhile I added this in the `arch/driver` branch:
https://github.com/lvgl/lvgl/blob/cf08c98a477c9528471805380579e07d984c60d5/lvgl.h#L110-L112

It also might be sufficient for the code. For the docs we should really make the v8.3 the default, especially after `arch/driver` is merged.

SeppiBOT, 2024-12-20:
Hello, i wanted to ask in here how to implement Psram in LVGL v 9.2, now in lv_conf.h we no longer have the "MEMORY SETTINGS" area where the previous solution was suggested, it looks like the conf file has changed quite a bit since the previous 8.3... versions. I am having limited Sram problems, and need to find some kind of solution, be it heavily cutting down on Sram occupation via offloading to Sram, or any other solution, can anyone help? If you want me to provide any more information for anything, go ahead and point that out for me!

kisvegabor, 2024-12-22:
Probably @liamHowatt or @uLipe know it better, but there is a Kconfig option in ESP's menuconfig to make malloc use PSRAM.

uLipe, 2024-12-23:
Since LVGL can just use system's malloc and free, the PSRAM configuration is done directly at ESP-IDF level.

You can use the heap_caps_malloc API from IDF with MALLOC_CAPS_EXTERNAL it will allocate memory from the PSRAM.

You need also to enable CONFIG_SPIRAM on the ESPIDF Kconfig.

SeppiBOT, 2024-12-24:
Both suggestions are based around the esp idf, how can i apply those suggestions if im using arduinos IDE? Is there no native way of doing this in lvgl?
By the way thanks for the amazing work on lvgl!!

liamHowatt, 2025-01-08:
Normally LVGL's builtin malloc uses a statically reserved array which the linker assigns a region for, usually only in SRAM. However there's a config option which lets you specify a literal memory address which you know will be valid after some higher-level board initialization, such as initializing PSRAM.

https://github.com/lvgl/lvgl/blob/31bd7b61bae2b7790910b605d2cd30b773fa554d/lv_conf_template.h#L77-L78

If you can find out what address your Arduino IDE maps PSRAM to, you can specify it there. You will need to ensure PRSAM is not being managed by any other allocator or else there will be a runtime conflict that will have buggy effects.

If you can't disable the other allocator, you can do a hacky trick where you see what address `heap_caps_malloc` returns the first time it's called with a large enough size for your application (i.e. 2 MB), then set that value to `LV_MEM_ADR`, rebuild, and always assert that address was allocated at startup.

1. `assert(heap_caps_malloc(LV_MEM_SIZE, MALLOC_CAP_SPIRAM) == LV_MEM_ADR)`
2. `lv_init()`

Some experimenting will be required on your end.

Oliblc, 2025-09-02:
LVGL 9.3.0 using Arduino IDE, the following (in `LV_USE_STDLIB_MALLOC` section in `lv_conf.h`) works fine. It allocates memory in SPIRAM on an ESP32S3.
```
#define LV_MEM_POOL_INCLUDE     "esp_heap_caps.h"
#define LV_MEM_POOL_ALLOC(size) heap_caps_malloc(size, MALLOC_CAP_SPIRAM | MALLOC_CAP_8BIT)

```