lvgl/lvgl #3246: Slider bar left knob can not focused?

xfdr0805, 2022-04-03:
Hello everyone
I am using lvgl-8.3.0-dev version building with lv_sim_visual_studio
What should I do to focus on left knob object?
thanks a lot!

[image]

[image]

xfdr0805, 2022-04-05:
I found that the messagebox's close button also fails to focus when using the key
[image]

closed 2022-04-07 by xfdr0805

t-walker-wei, 2022-10-21:
@xfdr0805  Did you solve this problem? I have the same issue now. Thank you in advance.

kisvegabor, 2022-10-23:
I made no progress on it yet. For v9 we will rework the `group`s and we can take this issue into account as well. 

I added it to the ROADMAP: https://github.com/lvgl/lvgl/commit/b64dceae0ac5a840f03d038ac16441ce094482b0