lvgl/lvgl #3247: Weird anti-aliasing result with custom TTF font renderer + dark background

huming2207, 2022-04-04:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

I've implemented a font rendering wrapper that makes the `stb_truetype` to work with LVGL. It works, however, the antialiasing looks horrible if I use it for a `lv_label` in a darker background, particularly full black. However, it does work well in white background. 

Please refer to the images below.

### To Reproduce

- Here's my rendering wrapper library: https://github.com/huming2207/bisheng-fontmgr/blob/master/includes/stb_font_view.hpp
- Our environment:
  - Microcontroller: ESP32
  - Display: E-ink ED097TC2, 4-bit 16 level grayscale
  - LVGL version: master branch, commit at b171f7dde2a895142385ea1275f3f51255cb2811
  - ESP-IDF version: master branch, commit at e3c0229d2ec9bfd0203a3cab18d6642798457578
- To reproduce: 

1. Grab this header-only library to your codebase
2. Do something like:

```cpp
auto *font_view = new stb_font_view();
font_view->init(wqy_ttf_start, (wqy_ttf_end - wqy_ttf_start), 72);
```

...where `wqy_ttf_start` and `wqy_ttf_end` is a WenQuanYi font embedded to a ESP32 firmware.

3. Apply the style to a `lv_label`:

```cpp
static lv_style_t label_style;
lv_style_init(&label_style);
font_view->decorate_font_style(&label_style);
lv_style_set_text_color(&label_style, lv_color_white()); // Set text color to white, and assume the label is on a black background object
lv_obj_add_style(foo_label, &label_style, 0);
```

### Expected behavior

Anti-aliasing should just work fine.

### Screenshots or video

Properly anti-aliased font (when in white background):

[image]

[image]

Screwed up fonts (when in black background):

[image]

huming2207, 2022-04-04:
I've also got FreeType2 working on ESP32, and yes it generates the exact same result as the `stb_truetype`! So now I think there's a bug somewhere on the software blender side, and it seems like it's not the TTF renderer or the wrapper library's fault.

huming2207, 2022-04-04:
I've added a log at `lv_draw_label.c`, here's the result:

```
Letter: 0x32 "2", Color: 0xffff, Background: 0x24be
Letter: 0x31 "1", Color: 0xffff, Background: 0x24be
Letter: 0x31 "1", Color: 0xffff, Background: 0x24be
```

So I think this part looks correct, the color is white and the background is somewhat near black. Maybe something wrong in the software renderer?

kisvegabor, 2022-04-04:
I don't think it's a SW blend issue because the  non-FreeType fonts are using it as well and they are working fine. 

I suggest creating a dummy image for a letter. E.g. when the bitmap of `'a'` should be returned, return a known, simple array. E.g. increase the opacity in each line.

kisvegabor, 2022-04-04:
> Do you mean returning a bitmap from black to white from the get_glyph_bitmap_handler()?

Yes!

You can also try drawing a circle with LVGL maybe with some shadow to see if it looks well.

```c
  lv_obj_t * obj = lv_obj_create(lv_scr_act());
  lv_obj_set_size(obj, 50, 50);
  lv_obj_center(obj);

  lv_obj_set_style_radius(obj, LV_RADIUS_CIRCLE, 0);
  lv_obj_set_style_border_width(obj, 5, 0);
  lv_obj_set_style_border_color(obj, lv_color_hex3(0x333), 0);
  lv_obj_set_style_bg_color(obj, lv_color_hex3(0xccc), 0);
  lv_obj_set_style_shadow_width(obj, 40, 0);
```
[image]

huming2207, 2022-04-04:
@kisvegabor good idea, I will have a try tomorrow! Thanks in advance!

huming2207, 2022-04-04:
Hi @kisvegabor 

Here's the result:

1. With your code

```c
  lv_obj_t * obj = lv_obj_create(lv_scr_act());
  lv_obj_set_size(obj, 50, 50);
  lv_obj_center(obj);

  lv_obj_set_style_radius(obj, LV_RADIUS_CIRCLE, 0);
  lv_obj_set_style_border_width(obj, 5, 0);
  lv_obj_set_style_border_color(obj, lv_color_hex3(0x333), 0);
  lv_obj_set_style_bg_color(obj, lv_color_hex3(0xccc), 0);
  lv_obj_set_style_shadow_width(obj, 40, 0);
```
It generates something wrong:

[image]

2. I modified a bit to:

```c
    lv_obj_t * obj = lv_obj_create(lv_scr_act());
    lv_obj_set_size(obj, 100, 100);
    lv_obj_center(obj);

    lv_obj_set_style_radius(obj, LV_RADIUS_CIRCLE, 0);
    lv_obj_set_style_border_width(obj, 5, 0);
    lv_obj_set_style_border_color(obj, lv_color_hex3(0x111), 0);
    lv_obj_set_style_bg_color(obj, lv_color_hex3(0xccc), 0);
    lv_obj_set_style_shadow_width(obj, 40, 0);
```
It then generates:

[image]

[image]

3. I've also tried FreeType2 in white background and here's what it generates, this time it comes with the right anti-aliasing:

[image]

huming2207, 2022-04-05:
Meanwhile, here's a pre-rendered PNG image that writes to the display as a `lv_img` object:

[image]

The anti-aliasing works in this case. That's another reason why I suspect probably there are some issues with our configuration and/or the LVGL software renderer rather than the display driver.

huming2207, 2022-04-05:
Here's the gradient bitmap test. I've added the code below at: https://github.com/huming2207/bisheng-fontmgr/blob/master/includes/stb_font_view.hpp#L62
... so that to override the bitmap buffer with a dummy gradient (from full black to full white) bitmap:

```c
            for (size_t h_idx = 0; h_idx < height; h_idx += 1) {
                uint8_t color = (uint8_t)(((float)h_idx / (float)height) * (float)255);
                for (size_t idx = 0; idx < width; idx += 1) {
                    size_t buf_idx = h_idx * width + idx;
                    ctx->font_buf[buf_idx] = color;
                }
            }
```

Here's what it be like with white background & black text:

[image]

Here's the black background & white text:

[image]

I think both of these two scenarios supposed to be black on top and gradually turn to be white at bottom. But apparently they are not behaving like that.

huming2207, 2022-04-05:
I fixed it! Turned out to be that the opacity in my `set_px()` callback is completely wrong! 

I think the reason why the PNG works is that PNG images are treated as non-opaque stuff while those anti-aliased pixels are somewhat opaque in this scenario.

I've searched around again and eventually I found this issue: https://github.com/lvgl/lvgl/issues/2863 

So I followed that and reimplemented the way of handling the `opa` value.

Here's my original implementation:

```cpp
void tcon_lv::disp_set_px_handler(lv_disp_drv_t *disp_drv, uint8_t *buf, lv_coord_t buf_w, lv_coord_t x, lv_coord_t y, lv_color_t color, lv_opa_t opa)
{
    uint32_t idx = ((y + 1) * buf_w) + x;
    if (opa < 16) {
        buf[idx] = 0xff;
    } else {
        // Algorithm originally from here: https://stackoverflow.com/a/58451121
        uint32_t red_0 = ((((uint32_t)color.ch.red << 8) / opa) * 3843) >> 14;
        uint32_t green_0 = (((((uint32_t)color.ch.green << 8) / opa) * 11718) >> 14) >> 1;
        uint32_t blue_0 = ((((uint32_t)color.ch.blue << 8) / opa) * 1183) >> 14;
        uint16_t gray_0 = red_0 + green_0 + blue_0;

        buf[idx] = (uint8_t)(((gray_0 << 11) + (gray_0 << 6) + gray_0) >> 8) >> 4;
    }
}
```

Here's the new implementation after I read that issue topic:

```cpp
void tcon_lv::disp_set_px_handler(lv_disp_drv_t *disp_drv, uint8_t *buf, lv_coord_t buf_w, lv_coord_t x, lv_coord_t y, lv_color_t color, lv_opa_t opa)
{
    uint32_t idx = ((y + 1) * buf_w) + x;

    // Algorithm originally from here: https://stackoverflow.com/a/58451121
    uint32_t red_0 = (((uint32_t)color.ch.red) * 3843) >> 14;
    uint32_t green_0 = ((((uint32_t)color.ch.green) * 11718) >> 14) >> 1;
    uint32_t blue_0 = (((uint32_t)color.ch.blue) * 1183) >> 14;
    uint16_t gray_0 = red_0 + green_0 + blue_0;

    uint8_t pix_in_buf = buf[idx];
    uint8_t incoming_pix = (uint8_t)(((gray_0 << 11) + (gray_0 << 6) + gray_0) >> 8) >> 4;
    uint8_t blended_pix = (pix_in_buf * (LV_OPA_COVER - opa) + incoming_pix * opa) / LV_OPA_COVER;
    buf[idx] = blended_pix;
}
```
 
Also, sorry for that previously I kept thinking it was the SW renderer issue.  It's actually my fault lol...

closed 2022-04-05 by huming2207

kisvegabor, 2022-04-05:
> I think the reason why the PNG works is that PNG images are treated as non-opaque stuff while those anti-aliased pixels are somewhat opaque in this scenario.

The PNG images are treated as a "color map" + "alpha (mask) map". While letter are a single color  + "alpha (mask) map". 

Anyway happy to hear that you have found the issue! :slightly_smiling_face:

huming2207, 2022-04-05:
> The PNG images are treated as a "color map" + "alpha (mask) map". While letter are a single color + "alpha (mask) map".

Oh okay, thx for the info. But if the image is in RGB888 (and no alpha), does it still has that alpha map?

kisvegabor, 2022-04-05:
> But if the image is in RGB888 (and no alpha), does it still has that alpha map?

Nope, it's only for ARGB images. With ARGB images LVGL separates the color and alpha data and uses the alpha channel as a mask during blending.

huming2207, 2022-04-05:
> Nope, it's only for ARGB images. With ARGB images LVGL separates the color and alpha data and uses the alpha channel as a mask during blending.

Understood, thanks.