lvgl/lvgl #3255: Unexpected Scroll Behavior

cagrig, 2022-04-05:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

Scroll part of a list or object gets a scroll event from another unrelated object.

### To Reproduce

At first, two list objects on the same screen were affecting each other while one is being cleaned and dynamically added new items into it. List objects are changed by base objects and the issue continues. At the same time, it is observed that the message box component affects also scroll parts of the objects (video is attached).

To simplify or reproduce the problem, list object example is taken from docs and applied. The same behavior is observed. When items of a list dynamically are changed or the message box is closed, the other list takes a scroll event.

lvgl version: 8.1.0
tested on: simulator

### Expected behavior

When the other scrollable objects are updated or message box is closed, the remaining / active scrollable object should not scroll.
### Screenshots or video

https://user-images.githubusercontent.com/32802359/161767773-aa300e7b-4d5f-4bc5-850e-119012522883.mp4

kisvegabor, 2022-04-11:
Is there a default group set? (`lv_group_set_default?`) Probably it happens because when the message box if focused and it's closed a button if the list gets focused, and LVGL automatically scrolls to that button.

cagrig, 2022-04-13:
Thank you for your interest. I could not look for the group set, yet. After working on it, I am going to reply here.

closed 2022-05-06 by lvgl-bot

cagrig, 2022-05-16:
@kisvegabor thank you again. As you said, the reason is the default group. Adding a new default group to the lists solved the issue.

kisvegabor, 2022-05-16:
Great, thanks for the feedback!