lvgl/lvgl #3258: Sleep interval (time to next tick)

simplejack-src, 2022-04-08:
### Introduce the problem
I'm looking to have the CPU processing be a little leaner (i.e. tickless/sleep when nothing to do). Currently, I've tweaked the Wayland driver backend to stub out a ```void wait(int timeout)``` function that waits until something Wayland-relevant occurs (input/resize/etc.). I've read the 'Sleep' section of the documentation, and implemented my main loop accordingly (see example). However, I'm thinking I'm missing something. I'm new to LVGL, but I'm wondering if there's a function to query when the next required tick interval is? Specifically, I'm thinking at bare minimum I need to know when the next timer is going to expire (so I can ensure I wakeup at exactly the right time to handle it).

### Examples and cases
My main loop:
```
#include <stdio.h>
#include <errno.h>
#include <time.h>
#include "lvgl.h"
#include "lv_drivers/wayland/wayland.h"

int main(void)
{
   struct timespec ts = { 0x00 };
   int ret;
   lv_disp_t *disp;
   uint32_t tick_sleep = 0;

   lv_init();
   lv_wayland_init();

   disp = lv_wayland_create_window(320, 240, "My Window", NULL);

   /* TODO: Add lv_* GUI objects here */

   while (0x01) {
      lv_tick_inc(tick_sleep);
      lv_task_handler();

      if (lv_anim_count_running() ||
          (lv_disp_get_inactive_time(NULL) < LV_DISP_DEF_REFR_PERIOD)) {
         /* 5ms sleep (while display is still 'active') */
         tick_sleep = 5;
         ts.tv_nsec = (5 * 1000000);
         do {
            ret = nanosleep(&ts, &ts);
         } while (ret && (errno == EINTR));
      } else {
         tick_sleep = LV_DISP_DEF_REFR_PERIOD;
         lv_wayland_wait(10000); // How could I calculate this timeout value (10s)?
      }
   }

   lv_wayland_deinit();
   return 0x00;
}
```

### Suggested solution
As shown in the example above I'm using a high timeout (10 seconds), which in my quick testing is working, I'm thinking this is obviously too high (at bare minimum to service any timer thats created below that). Ideally, I'd like to be able to query a ```get_next_tick()``` function to get an appropriate timeout value to use (the above ```wayland_wait()``` function already returns early if something occurs). Does such a function already exist? If not, could I request such a function (feature request) ?

Thank-you

P.S. Thank-you to all the LVGL developers, I'm just starting to look through this library and it looks fantastic!

embeddedt, 2022-04-08:
`lv_task_handler` returns the number of milliseconds till it should be called again (e.g. if an animation is running) or `LV_NO_TIMER_READY` if you should block and wait for input.

However, since LVGL polls for input, it ends up returning `LV_INDEV_DEF_READ_PERIOD` a lot of the time since that task is always running. I didn't get around to implementing event-driven input, however, with a quick glance at the way the code is structured now, if you have your instance of `lv_indev_drv_t` it should have a `read_timer` handle on it that you can use with `lv_timer_pause` and `lv_timer_resume`.

I hope that was clear enough! Let me know if you have questions.

simplejack-src, 2022-04-09:
Well I've modified the Wayland driver and my test case accordingly. Seems to somewhat work...I'm currently debugging why a pointer click isn't processed (screen updated) until a subsequent pointer move.

Test case:
```
#include <stdio.h>
#include <errno.h>
#include <time.h>
#include "lvgl.h"
#include "lv_drivers/wayland/wayland.h"

int main(void)
{
   struct timespec start;
   struct timespec end;
   lv_disp_t *disp;
   uint32_t sleep = -1;

   lv_init();
   lv_wayland_init();

   disp = lv_wayland_create_window(320, 240, "My Window", NULL);

   /* TODO: Add lv_* GUI objects here */

   while (0x01) {
      clock_gettime(CLOCK_MONOTONIC, &start);
      lv_tick_inc(sleep);
      lv_task_handler();

      lv_wayland_pause_timers();
      sleep = lv_timer_handler();
      /* Sleep until next timer expiry or until a Wayland event */
      lv_wayland_wait((sleep == LV_NO_TIMER_READY) ? -1 : sleep);
      lv_wayland_resume_timers();

      clock_gettime(CLOCK_MONOTONIC, &end);
      sleep = (((end.tv_sec - start.tv_sec) * 1000) +
               ((end.tv_nsec - start.tv_nsec) / 1000000));
      sleep = ((sleep == 0) ? 1 : sleep);
   }

   lv_wayland_deinit();
   return 0x00;
}
```

embeddedt, 2022-04-09:
You should avoid calling `lv_tick_inc` in the same loop as that causes time to not move until after the sleep call returns. Since you're on Linux, you can use the `LV_TICK_CUSTOM` config option and skip calling it entirely. [Here](https://github.com/lvgl/lv_port_linux_frame_buffer) is a Linux project that does that so you have an example of how to configure it.

Also, I don't think this is the cause of the issue, but you're calling `lv_timer_handler` twice and ignoring the return value from the first call. `lv_task_handler` is an alias because that was the old name.

simplejack-src, 2022-04-09:
I've moved the tick out to use ```LV_TICK_CUSTOM``` (using the below function**):
```
uint32_t current_time(void)
{
   static uint32_t last = 0;
   uint32_t now; 
   struct timespec ts;

   clock_gettime(CLOCK_MONOTONIC, &ts);

   now = ((ts.tv_sec * 1000ULL) + (ts.tv_nsec / 1000000ULL));
   now = ((last == now) ? (last + 1) : now);
   last = now;
   return now;
}
```

I also enabled some INFO logs (using ```lv_example_anim_1()```) as my testbed. Got the below logs (shows LVGL registered the button press but didn't update the layout/flush the display with changes until a subsequent cycle/mouse move):
```
[Info]	(151765.708, +151765708)	 lv_init: begin 	(in lv_obj.c line #99)
[Info]	(151765.710, +2)	 lv_obj_create: begin 	(in lv_obj.c line #201)
[Info]	(151765.709, +-1)	 lv_obj_create: begin 	(in lv_obj.c line #201)
[Info]	(151765.710, +1)	 lv_obj_create: begin 	(in lv_obj.c line #201)
[Info]	(151765.710, +0)	 lv_label_create: begin 	(in lv_label.c line #75)
[Info]	(151765.709, +-1)	 lv_switch_create: begin 	(in lv_switch.c line #75)
[Info]	(151765.710, +1)	 lv_obj_update_layout: Layout update begin 	(in lv_obj_pos.c line #311)
[Info]	(151765.709, +-1)	 lv_obj_update_layout: Layout update begin 	(in lv_obj_pos.c line #311)
[Info]	(151765.710, +1)	 lv_obj_update_layout: Layout update begin 	(in lv_obj_pos.c line #311)
[Info]	(151765.709, +-1)	 lv_obj_update_layout: Layout update begin 	(in lv_obj_pos.c line #311)
[Info]	(151791.564, +25855)	 indev_proc_press: pressed at x:176 y:119 	(in lv_indev.c line #810)
[Info]	(151793.853, +2289)	 indev_proc_release: released 	(in lv_indev.c line #958)
[Info]	(151793.910, +57)	 lv_obj_update_layout: Layout update begin 	(in lv_obj_pos.c line #311)
[Info]	(151793.939, +29)	 lv_obj_update_layout: Layout update begin 	(in lv_obj_pos.c line #311)
[Info]	(151793.969, +30)	 lv_obj_update_layout: Layout update begin 	(in lv_obj_pos.c line #311)
...
```

** Example you linked doesn't quite seem to be correct (with the description/usage of ```LV_TICK_CUSTOM```), as their example seems to return the delta-time as opposed to the current time.

simplejack-src, 2022-04-09:
I think I've found my problem. The Wayland driver creates a cycle timer (to update LVGL with Wayland event values), this timer ends up running after the ```read_cb()``` poll (which means another timer cycle has to run through to pick up these values). A quick change (to make the Wayland driver run this cycle when my ```lv_wayland_wait()``` finishes) and things work properly. So I think I'll probably look to rejig the Wayland driver backend to have an ```#ifdef LV_WAYLAND_TIMER_HANDLER``` which will do the timers pause/resume/handler in one function and simplify my main loop down to:
```
while (1) {
   lv_wayland_timer_handler();
}
```

One sticking point I still have though. I still need to run ```lv_timer_handler()``` twice. Once for it's proper use and again after the Wayland input timers have been disabled (to determine when the next timer, if one is still active, need to run). Although, perhaps I could set a high ```LV_INDEV_DEF_READ_PERIOD``` and on wakeup for each timer do a ```lv_timer_ready()``` ... Will look to experiment with this...

simplejack-src, 2022-04-10:
Well experimentation seems to have yielded results! I've now reduced my effort to an ```lv_wayland_run()``` function** (which runs in a Wayland/LVGL event loop until the last window closes). I went the ```lv_timer_ready()``` route (and using a sufficiently high poll rate, think > 60s) I'm effectively operating fully event driven (sleeping when nothing has occurred).

@embeddedt  Big thank-you for all your insight here, made diving into Wayland/LVGL a lot easier. This issue can be closed (all the tools were present, just had to line them up straight).

** I'll look to make a pull request on lv_drivers with the Wayland changes I've made (after looking through the contribution guidelines and a bit of cleanup).

embeddedt, 2022-04-10:
> Example you linked doesn't quite seem to be correct (with the description/usage of LV_TICK_CUSTOM), as their example seems to return the delta-time as opposed to the current time.

It should still be returning the current time; it's just adjusted so that the first call returns 0ms instead of milliseconds since the epoch. In other words, the epoch is adjusted to be when the application is started. That way `lv_tick_get()` will return 0ms the first time it's called.

It looks like your callback is returning the delta though (you seem to always set `last` to equal `now`)?

simplejack-src, 2022-04-10:
Ah yes...I didn't pick up on that (sensible to delay overflow as long as possible). In my original example, I was aiming to force a movement in time (in my experience a lot of embedded libraries that need a time reference, call for it once and expect some sort of movement between new calls), however, LVGL doesn't seem to suffer from this problem (and in fact takes makes multiple ```lv_tick_get()``` calls in a single handler cycle). I've changed my custom tick function accordingly:

```
uint32_t current_time(void)
{
   static unsigned long long start_ms = 0;
   unsigned long long now;
   struct timespec ts;

   if (start_ms == 0) {
      clock_gettime(CLOCK_MONOTONIC, &ts);
      start_ms = ((ts.tv_sec * 1000ULL) + (ts.tv_nsec / 1000000ULL));
   }

   clock_gettime(CLOCK_MONOTONIC, &ts);
   now = ((ts.tv_sec * 1000ULL) + (ts.tv_nsec / 1000000ULL));

   return (now - start_ms);
}
```

simplejack-src, 2022-04-10:
Assuming ```lv_drivers``` [#204 ](https://github.com/lvgl/lv_drivers/pull/204) get picked up (example in PR) LVGL sleep/tickless/event-driven with the Wayland backend should be good.

closed 2022-04-10 by simplejack-src