lvgl/lvgl #3261: function "lv_obj_get_text_color_filtered" return the wrong color

wey4321, 2022-04-08:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

### To Reproduce

### Expected behavior

### Screenshots or video

wey4321, 2022-04-08:
[image]
i fix the issue like this, and the return color is correct.
maybe there are better ways to fix it!

embeddedt, 2022-04-08:
Setting filtered properties was a bug in the autogenerated API. There should only be a getter. In the next version of LVGL `lv_obj_set_text_color_filtered` will be removed. You should just set the regular property with `lv_obj_set_text_color`.

closed 2022-04-08 by embeddedt

wey4321, 2022-04-11:
@embeddedt   the code of  'lv_obj_get_style_prop' of master branch is like this
[image]
is seems, this bug is fixed. and whether the next version of LVGL will be availabel at branch release_v8.2?

kisvegabor, 2022-04-11:
We will release v8.3 in a few months.

wey4321, 2022-04-12:
@kisvegabor so the branch release_v8.2 is not long-term stable version, and it will be replaced by release v8.3 ? and the bugs of v8.2 will only be fixed at master branch?

embeddedt, 2022-04-12:
I think there is some confusion here. There isn't a bug from what I can tell, aside from the fact that you were able to call `lv_obj_set_style_text_color_filtered` in previous versions (which you should not have been able to, and the API does not exist in v8.3 anymore). Filtering is intended to be applied inside a getter only based on other style properties. You should not be setting the filtered version of a property.

wey4321, 2022-04-13:
@embeddedt  ok, i got it. thanks a lot!

embeddedt, 2022-04-13:
Glad I could help!