lvgl/lvgl #3262: Setting the file cache to 512 will cause the font to fail to load

alec-2018, 2022-04-08:
### Perform all steps below and tick them with [x]
- [ ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

hi @kisvegabor ,I Setting the file cache to 512 will cause the font to fail to load，The emulator version is the latest（lv_port_win_visual_studio-master）
### To Reproduce

In the emulator, setting LV_FS_WIN32_CACHE_SIZE to 512 reproduces the problem
```c
    lv_font_t* font = lv_font_load("C:\\font.bin");
    lv_obj_t* label = lv_label_create(lv_scr_act());
    lv_label_set_text(label, "abcdefghigklmnopqrst");
    lv_obj_set_style_text_font(label, font, 0);
```
### Expected behavior

[image]

### Screenshots or video

[image]

kisvegabor, 2022-04-14:
Are using the latest version of LVGL from the master branch?

If so, please attach your `font.bin` to reproduce the issue.

alec-2018, 2022-04-14:
> Using the latest version of LVGL from the master branch?
> yes
> If so, please attach yours `font.bin`to reproduce the issue.
[font.zip](https://github.com/lvgl/lvgl/files/8489496/font.zip)
Sorry, I lost my previous font file.
I can reproduce the problem using this font file；It's strange that I put cache_size = 512, no error occurs, but cache_size = 1024, the problem will reproduce
I modified the code here: "file_p->cache->start = file_p->cache->end + 1" to "file_p->cache->start = file_p->cache->end" ，cache_size = 1024 OK

kisvegabor, 2022-04-21:
Hi,

I've tracked it down and found the root of the issue: an extra `+ 1`... 

Once I was there I added a simple unit test for `lv_fs.c` which revealed that cached mode doesn't handle if EOF is reached. So I fixed that too. 

Let me know if it works on your end as well.

closed 2022-04-21 by kisvegabor

alec-2018, 2022-04-25:
> 你好，
> 
> 我已经追踪它并找到了问题的根源：一个额外的`+ 1`......
> 
> 一旦我在那里，我添加了一个简单的单元测试，`lv_fs.c`它表明如果达到 EOF，缓存模式将无法处理。所以我也解决了这个问题。
> 
> 让我知道它是否也适用于您。

> Hi,
> 
> I've tracked it down and found the root of the issue: an extra `+ 1`...
> 
> Once I was there I added a simple unit test for `lv_fs.c` which revealed that cached mode doesn't handle if EOF is reached. So I fixed that too.
> 
> Let me know if it works on your end as well.

thanks, it works for me

commit 98660a86 (2022-04-25): fix(fs): fix cached read and add unit test for lv_fs

fixes: #3262