lvgl/lvgl #3265: Wrong function return result

Barto22, 2022-04-10:
### Perform all steps below and tick them with [x]
- [X] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [X] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [X] Update lvgl to the latest version
- [X] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

### To Reproduce

### Expected behavior

### Screenshots or video

embeddedt, 2022-04-10:
Thanks for noting this! I'm not sure why there is no warning on CI (or locally, for that matter) when returning a boolean without casting. I'll need to look into this further.

embeddedt, 2022-04-11:
Upon doing some further research, it appears that the lack of a CI error is caused by the fact that many compilers allow implicit conversion of the integral value `0` to a pointer without a warning, e.g.:

```c
void *test_fn(void)
{
    return 0;
}
```

There doesn't seem to be a way to force the compiler to error in this specific scenario, and `false` will be equal to NULL on practically every platform, so I don't think it's an edge case worth being concerned about.

Returning `true` in that situation would cause an error as expected.

commit 06fff4b9 (2022-04-11): fix(event) avoid using a boolean as a pointer

Fixes #3265
closed 2022-04-11 by embeddedt