lvgl/lvgl #3266: help,slips to bottom so hard

Pingye007, 2022-04-11:
### Introduce the problem
I wrote a demo to test whether it can slip well or not without setting flex flow "LV_FLEX_FLOW_COLUMN",
and it worked well on Windows simulator.But when I downloaded the program on POS based on Linux,
I found it scrolled to bottom from top was so hard,and scrolled top from bottom was so easy.And I cannot 
figure it out,because I just set the configuration according to the giving examples. Where may I be wrong?

### Examples and cases
Here are my codes,I think it's simple.

typedef struct _widget_attrs_t {
	uint8_t* widget_name;
	lv_dir_t scroll_dir;
}widget_attrs_t;

typedef enum{
    SCAN_NULL = -1,
    SCAN_SINGLE_MODE = 0,
    SCAN_CONTINUOUS
}scan_mode_t;

#define TEST_UNIT 200

lv_obj_t* gen_mode_screen(scan_mode_t scan_mode)
{
	if (scan_mode == SCAN_NULL)
	{
		return NULL;
	}
	static widget_attrs_t settings_attrs, scan_attrs, gallery_attrs;
	lv_obj_t* ret_obj = lv_obj_create(NULL);
	lv_obj_t* panel = lv_obj_create(ret_obj);
	//lv_obj_set_size(panel, LV_SIZE_CONTENT, LV_SIZE_CONTENT);
	lv_obj_set_size(panel, 200, 200);

	/* Settings */
	lv_obj_t* settings_obj = lv_obj_create(panel);
	lv_obj_set_size(settings_obj, TEST_UNIT,TEST_UNIT);
	lv_obj_t* settings_label = lv_label_create(settings_obj);
	lv_label_set_text(settings_label, "Settings Label");
	lv_obj_set_style_bg_color(settings_obj, lv_palette_main(LV_PALETTE_BLUE), 0);
	lv_obj_set_size(settings_label, lv_pct(100), LV_SIZE_CONTENT);
	lv_obj_center(settings_label);
	settings_attrs.scroll_dir = LV_DIR_TOP;
	settings_attrs.widget_name = "settings";
	printf("SETTINGS ATTRS:DIR = %d \n", settings_attrs.scroll_dir);
	//lv_obj_add_event_cb(settings_obj, move_limit_cb, LV_EVENT_PRESSED, &settings_attrs);

	/* Scan Mode */
	lv_obj_t* scan_obj = lv_obj_create(panel);
	lv_obj_set_size(scan_obj, TEST_UNIT, TEST_UNIT);
	lv_obj_set_style_bg_color(scan_obj, lv_palette_main(LV_PALETTE_RED), 0);
	lv_obj_t* scan_label = lv_label_create(scan_obj);
	lv_obj_set_size(scan_label, lv_pct(100), LV_SIZE_CONTENT);
	lv_obj_center(scan_label);
	scan_attrs.scroll_dir = LV_DIR_ALL;
	scan_attrs.widget_name = "scan";
	//lv_obj_add_event_cb(scan_obj, move_limit_cb, LV_EVENT_PRESSED, LV_DIR_ALL);
	//lv_obj_add_event_cb(scan_obj, move_limit_cb, LV_EVENT_PRESSED, &scan_attrs);

	if (scan_mode == SCAN_SINGLE_MODE)
	{
		lv_label_set_text(scan_label, "Scan Single");
	}
	else if (scan_mode == SCAN_CONTINUOUS)
	{
		lv_label_set_text(scan_label, "Scan Continuous");
	}

	/* Gallery */
	lv_obj_t* gallery_obj = lv_obj_create(panel);
	lv_obj_set_size(gallery_obj, TEST_UNIT,TEST_UNIT);
	lv_obj_set_style_bg_color(gallery_obj, lv_palette_main(LV_PALETTE_YELLOW), 0);
	lv_obj_t* gallery_label = lv_label_create(gallery_obj);
	lv_obj_set_size(gallery_label, lv_pct(100), LV_SIZE_CONTENT);
	lv_label_set_text(gallery_label, "Gallery Label");
	lv_obj_center(gallery_label);

	lv_obj_center(scan_obj);
	lv_obj_align(scan_obj,LV_ALIGN_TOP_LEFT,0,0);
	lv_obj_align_to(settings_obj, scan_obj, LV_ALIGN_CENTER, 0, TEST_UNIT);
	lv_obj_align_to(gallery_obj, scan_obj, LV_ALIGN_CENTER, 0, 0-TEST_UNIT);

	lv_obj_set_style_pad_all(scan_obj, 0, 0);
	lv_obj_set_style_pad_all(settings_obj, 0, 0);
	lv_obj_set_style_pad_all(gallery_obj, 0, 0);
	lv_obj_set_style_border_width(scan_obj, 0, 0);
	lv_obj_set_style_border_width(settings_obj, 0, 0);
	lv_obj_set_style_border_width(gallery_obj, 0, 0);
	lv_obj_set_scrollbar_mode(panel,LV_SCROLLBAR_MODE_OFF);
	lv_obj_center(panel);

	//lv_obj_scroll_to_view_recursive(scan_obj, LV_ANIM_ON);

	// lv_obj_add_flag(panel, LV_OBJ_FLAG_SCROLLABLE);
	// lv_obj_set_scroll_snap_x(panel, LV_SCROLL_SNAP_CENTER);
	lv_obj_set_scroll_snap_y(panel, LV_SCROLL_SNAP_CENTER);
	lv_obj_add_flag(panel,LV_OBJ_FLAG_SCROLL_ONE);

	lv_obj_update_snap(panel,LV_ANIM_OFF);
	//lv_obj_add_event_cb(panel, entry_move_cb, LV_EVENT_SCROLL_BEGIN, &scan_attrs);

	return ret_obj;

}

void test_scan_mode()
{
	scan_mode_t test_mode;

#if 1
	test_mode = SCAN_SINGLE_MODE;
#else
	test_mode = SCAN_CONTINUOUS;
#endif

	lv_obj_t* test_scr = gen_mode_screen(test_mode);
	lv_obj_t* test_label = lv_label_create(lv_scr_act());
	lv_label_set_text(test_label, "TEST LEBEL");
	lv_obj_center(test_label);

	lv_disp_load_scr(test_scr);
}

### Suggested solution
hope someboy can help me!And I am very interested in LVGL,fix this means a lot for me.
Thx!!!

FASTSHIFT, 2022-04-11:
I tested your code on Linux and it seems to work, is it a touchpad driver issue?

https://user-images.githubusercontent.com/26767803/162734145-3af94a5c-586d-44a2-bcc9-8194e820b079.mp4

kisvegabor, 2022-04-11:
@Pingye007  Just to be sure, area using the latest version of LVGL (from the master branch)?

Pingye007, 2022-04-12:
@FASTSHIFT 
Yes,it runs on simulator well and I can't find problems at all.
Might be drivers...

Pingye007, 2022-04-12:
@FASTSHIFT 
V8.2.0 from tags,And I found that slipped to top from bottom so smooth,so I rebuilt my arrangement by setting a widget as top widget,and setting other widgets below it one by one.Then slipped to bottom (from other widgets) became smooth too,this solved my problem,for now I guess.

closed 2022-05-03 by lvgl-bot