lvgl/lvgl #3268: Add layout finished event callback in disp_drv

FASTSHIFT, 2022-04-11:
### Introduce the problem
My screen driver has two fullscreen buffers and lvgl uses `direct_mode` to render directly on the buffers. When the buffer switches to the frame to be rendered by lvgl, the dirty area of ​​the previous frame needs to be copied to the frame to be rendered.

### Examples and cases
If the frame to be rendered is redrawn in full screen, the buffer synchronization operation can be omitted (because the buffer is completely covered).

### Suggested solution
An `layout_finished_cb` event is required to inform the display driver whether it needs to synchronize the buffers of the old and new frames.

**lv_refr.c:**
```c
/**
 * Refresh the joined areas
 */
static void lv_refr_areas(void)
{
    px_num = 0;

    if(disp_refr->inv_p == 0) return;

    /*Find the last area which will be drawn*/
    int32_t i;
    int32_t last_i = 0;
    for(i = disp_refr->inv_p - 1; i >= 0; i--) {
        if(disp_refr->inv_area_joined[i] == 0) {
            last_i = i;
            break;
        }
    }

    /* Notify the display driver that the layout is complete */
    if(disp_refr->driver->layout_finished_cb)
    {
        disp_refr->driver->layout_finished_cb(disp_refr->driver, disp_refr);
    }

    disp_refr->driver->draw_buf->last_area = 0;
    disp_refr->driver->draw_buf->last_part = 0;
    disp_refr->rendering_in_progress = true;

    for(i = 0; i < disp_refr->inv_p; i++) {
        /*Refresh the unjoined areas*/
        if(disp_refr->inv_area_joined[i] == 0) {

            if(i == last_i) disp_refr->driver->draw_buf->last_area = 1;
            disp_refr->driver->draw_buf->last_part = 0;
            lv_refr_area(&disp_refr->inv_areas[i]);

            px_num += lv_area_get_size(&disp_refr->inv_areas[i]);
        }
    }

    disp_refr->rendering_in_progress = false;
}
```

**my_disp_port.c:**
```c
static void my_disp_layout_finished_cb(lv_disp_drv_t* disp_drv, lv_disp_t* disp_refr)
{
    lv_coord_t hor_res = LV_HOR_RES;
    lv_coord_t ver_res = LV_VER_RES;

    for(uint16_t i = 0; i < disp_refr->inv_p; i++)
    {
        if(disp_refr->inv_area_joined[i] == 0)
        {
            const lv_area_t* area_p = &disp_refr->inv_areas[i];

            /* If a full screen redraw is detected, ignore dirty areas sync */
            if(lv_area_get_width(area_p) == hor_res
            && lv_area_get_height(area_p) == ver_res)
            {
                return;
            }
        }
    }

    /* Sync the dirty area of ​​the previous frame */
    my_disp_sync_areas();
}
```

kisvegabor, 2022-04-11:
Probably I misunderstood something, but if `buf1` is active (i.e. shown on the display) and LVGL renders to `buf2` and `(10;10;20;20)` will be redrawn, do you copy e.g. the `(10;10;20;20)` area from `buf1` to `buf2`?

If so, it's strange because you copy and area which will be overwritten by rendering.  

Copying the areas which WERE rendered (not will be rendered) seems more logical, because e.g. if a text was unhidden it should be shown in both buffers.

Am I missing something?

FASTSHIFT, 2022-04-13:
@kisvegabor 
I used a queue to manage the actions of lvgl rendering and screen sending.
The queue contains two elements, the content of which is a pointer to the buffer to be sent.

On the screen side, the screen driver scans the head of the queue every 16ms, and writes the address of the buffer to be sent stored in the head of the queue into the DMA register to start sending data. The sending process takes 7ms. After the sending is complete, the hardware will trigger an interrupt to notify the upper layer. After the transmission is over, the buffer address is removed from the queue after the transmission is over.

On the lvgl rendering side, the rendering thread scans the queue for free space, and if so, renders a new frame and adds it to the end of the queue.

FASTSHIFT, 2022-04-13:
In order to avoid the buffer data being modified by the rendering thread when the DMA is sent, I postpone the buffer synchronization operation until the start of a new frame rendering to ensure that the written buffer is in an idle state

FASTSHIFT, 2022-04-14:
@kisvegabor 
I cached `inv_areas` in last `flush_cb`. So when a new frame is about to be drawn, the dirty area synchronization will be performed according to the `inv_areas` recorded in the previous frame.
The process is as follows:
Frame 0: lvgl laoyout ->  no sync `inv_areas` -> lvgl rendering -> last flush_cb -> cache `inv_areas` -> set `buf1` to DMA
Frame 1: lvgl laoyout -> sync `inv_areas`(Frame 0) -> lvgl rendering -> last `flush_cb` -> cache inv_areas -> set `buf2` to DMA
Frame 2: lvgl laoyout -> sync `inv_areas`(Frame 1) -> lvgl rendering -> last `flush_cb` -> cache inv_areas -> set `buf1` to DMA
Frame 3: lvgl laoyout -> sync `inv_areas`(Frame 2) -> lvgl rendering -> last `flush_cb` -> cache inv_areas -> set `buf2` to DMA
...

kisvegabor, 2022-04-14:
Ah, I see now. It wasn't clear from the code. 

We can have a callback like this but I'd rather call it `render_start_cb()`. It also could be used to wait for VSYNC to start rendering. It's useful if rendering is faster then the an VSYNC period.

FASTSHIFT, 2022-04-14:
> Ah, I see now. It wasn't clear from the code.
> 
> We can have a callback like this but I'd rather call it `render_start_cb()`. It also could be used to wait for VSYNC to start rendering. It's useful if rendering is faster then the an VSYNC period.

Yes, this will be useful :)

kisvegabor, 2022-04-14:
Great, could you send a PR?

FASTSHIFT, 2022-04-14:
> Great, could you send a PR?

OK

closed 2022-04-29 by FASTSHIFT