lvgl/lvgl #3269: Snapshot widgets to buffer them for drawing

kisvegabor, 2022-04-11:


kisvegabor, 2022-04-11:
I'm trying to understand this topic better and how a more advanced render pipeline should be designed. The current implementation (just can see above) is just an early exploration so don't take it too seriously. 

As usual, I was thinking about how rendering on web works. We can start from the point when we know what and where we want to draw (i.e. styles, layouts and hierarchy are resolved).  

This is how I was thinking:

**We have a screen with a button, a label on the button** 
If we had infinite RAM (like on PC/mobile) we created  a texture (== LVGL image for the sake of simplicity) from each object and composed them into a final layer which is sent to the display.  If the button is pressed, the screen's image doesn't need to be updated, but button should, and if the text color/font/etc is also changed then the label's image also should be updated. When all done compose to a layer again, send to the display, done.

**What if the button is rotated by the 30 deg?**
 The button's image can be rotated when blended to the final layer. The label also should be rotated around the same pivot.

**What if the button is rotated by the 30 deg and label by 20 deg?**
The transformation parameters should be cascaded, i.e. the label should be rotated by 30+20 deg.

**What if the button has 50% opa ?**
Now the button is drawn with 50% opa and after that the label is drawn by 50% opa too. But is not correct. The button and the label should be composed to an intermediate layer (considering their rotations too) and that single layer should be blended with 50% opa to the screen's image. On web the related term is the [Stacking context](
https://developer.mozilla.org/en-US/docs/Web/CSS/CSS_Positioning/Understanding_z_index/The_stacking_context).

**What if the button has OVERFLOW_VISIBLE and the label overflows it?**
If there is no global opa, just blend the label and the button after each other without clipping the label to the button.

**And if the button has 50% opa too?**
The texture of the button should be large enough to hold all the children.

**Blurred backdrop-filter?**
Like [this](https://developer.mozilla.org/en-US/docs/Web/CSS/backdrop-filter#result). Even if only one pixels changes on the background, the whole blurred box above that pixel needs to be updated, as that one pixel affects the whole blur. So probably check all children when part of the wallpaper changes. But the box is just a tricky rectangle which reads the current composed layer beneath itself and changes it.  

**CONCLUSION**
- All widgets are cached into textures without their children. So (bg+border+shadow as one texture, not 3 separate)
- Rendering the final image means composing the widgets' textures
- If a widget changes, only its texture needs to be updated and the final layer should be recomposed
- An intermediate layer should be added in special cases, e.g. when a widget has opa!=100%

**THOUGHTS**
- If all widgets store their own texture, the textures can not be reused. I.e. 100 buttons looking the same, have 100 textures. Maybe there should be a global texture buffer, from where widgets can reference same-looking textures, counting references, etc...
- It's still a lot of RAM... We shouldn't cache e.g. a white screen background and all texts.

**IMPLEMENTATION IDEA**
- `lv_obj_invalidate()` should request texture update too. If there is an intermediate layer on the object's parent, requests update on it too. 
- At the beginning of `disp_refr_timer` 
   - Update invalid textures   
   - Compose intermediate layers
- During rendering:
   - If a widget is replaced by an intermediate layer, just draw it that already composed layer. 
   - Else draw the widget's texture
  
Probably it's not worth to store texture for all widgets. Instead there can be some smart policies, e.g.
- Create texture automatically, if the object (or any of its parents is rotated, scaled, etc)
- Create a texture if requested manually (i.e. the widget has a large shadow and would be slow to compute, so ask LVGL to cache it) 
- Estimate the draw complexity of the widget, and buffer if somethings seems too slow to draw.
- Or just buffer everything if you are on a desktop-like hardware. 

The need for intermediate layers should be detected and they should be created automatically. 
It sounds complicated but the intermediate layers could be crated/rendered line-by-line, just like how LVGL renders the screen line-by-line into a small buffer.

---

That's it  in "short". :slightly_smiling_face: I cc a few user who might be interested in this topic, but for anyone, feel free tell your thoughts!

@embeddedt  @X-Ryl669 @xiaoxiang781216 @FASTSHIFT @mariotaku

mariotaku, 2022-04-12:
The way you want to move to composition based rendering will be pretty nice for devices with larger RAM. Here are some of my thoughts. 

To save a view to snapshot, we need to consider what will change frequently and what not. Let's take a few kinds of widgets as an example:
* lv_obj: Shaped background should be saved. Except for the object doesn't have a radius and has a solid background.
* lv_img: The whole image widget should be saved.
* lv_slider: The knob and the track should be separated.
* lv_label: Similar to obj, but for the characters we need to think more.

kisvegabor, 2022-04-12:
> lv_obj: Shaped background should be saved. Except for the object doesn't have a radius and has a solid background.

I agree.

> lv_img: The whole image widget should be saved.

It's already saved as the image is already an image :slightly_smiling_face: 

> lv_slider: The knob and the track should be separated.

I was thinking about it too.  Snapshoting sliders with the same styles but different value into different textures doesn't make much sense. So snapshoting should be a handled by the widget, and one widget might produce and reference multiple textures.  It'd work something like: 
- `widget_update_texture(slider1)`: send a texture update event to the slider which snapshots what it wants (e.g. do not snapshot the indicator)
- In `LV_EVENT_DRAW_MAIN` of the slider: it know about itself that it created textures, so during drawing it looks for textures too. Or `lv_draw_rect/text/arc()` searches for textures automatically.  (I think you did the same in the SDL renderer)

BTW, it also affects bar, checkbox, button matrix, arc, and many more. 

> lv_label: Similar to obj, but for the characters we need to think more.

I think characters buffering/caching should be managed by the font engine. On this level we should save the whole label.

---

Do you think this approach can be used to simplify the SDL renderer. I think the best would be if it used SW rendering, and used SDL only to blend textures.

---

X-Ryl669, 2022-04-12:
What you are describing is similar to CSS's box layout. However, there are few gotchas here:
1. When applying a transform (rotation, zoom, perspective ?) you need to store a matrix per object. This implies a change to the box's size, position and inner pixels. So a point that's tested for hit-test must be multiplied by the matrices successively (like you said, if you have a children inside a children, you'll need to apply each matrix for each point to test and for drawing). 
2. Storing to texture is just a fancy name for dealing with composition buffers and damage management and a Z-buffer somehow (or at least use the painter's algorithm). In embedded world, this means either a lot of RAM usage (no-no-no) or using a lot of CPU processing by having a vector based drawing engine that can deal successfully (and fast) with a 3D scene (layers of 2D layers *= textures* transformed by a matrix). A vector based engine needs to apply a matrix operation (at least 3x2) for each point drawn, not sure if embedded CPU can deal with such constraint in realtime. Typically, you need to compute the combined matrix for each visible object, then invert the matrix and use this matrix for each destination point to find the source point. 
3. If you add alpha to the mix, then the complexity implodes since a transparent pixel does not hide anymore the objects below, so you can end up drawing each object for each frame.

That being said, there are a lot of tries to implement such typical rendering pattern in embedded world and CPU are getting faster every year (with multiple core and so on). The best optimization I've seen so far use a tiling approach. With a tiling approach, you can split the rendering screen in tiles and store a bitmap for the state of the tiles. When some content is modified, the hit test algorithm is able to map the object to the tiles that need to be re-rendered (that fast, because you're dealing with large tiles, not pixels here and you only need to hit test the corner of the objects). Then you'll set up the window/ROI for the screen redraw and only re-render the damaged tiles before clearing the tiles' bitmap. That avoids having to store bitmaps of the objects and doing the composition based on textures and deal with texture management. This also has the advantage to be easily parallel (each core rendering its own tile, and if using a GPU, it maps perfectly with the GPU core scheduler).
That's how servo is doing its composition too.

XuNeo, 2022-04-12:
I have an idea.

Based on current canvas object, add the ability to draw any obj on it by using same tech snapshot uses. Canvas object acts like a virtual display in current display, since it's an image on current display, user can do any transform.

RAM must be a concern on real device, which makes the canvas approach more practical. 

Considering a real scenario that a list item scales while it scrolls, I just create a canvas obj as list item and use it as a container and set image transform parameter accordingly. 

See: 
[image]

Being a virtual display, it should also relay all indev events to act user inputs, which can be done simply registering a virtual indev driver to the display.

The style may cause some trouble especially when adding style to the canvas that has effect on children.

kisvegabor, 2022-04-12:
@X-Ryl669 
> When applying a transform (rotation, zoom, perspective ?) you need to store a matrix per object. This implies a change to the box's size, position and inner pixels. So a point that's tested for [...]

True, but it already works in  the proof of concept version. See the slider in my first post. 

>  lot of RAM usage (no-no-no) 

 I agree that LVGL should support single MCU scenarios too, but having 16..32MB external SRAM is not that rare today. E.g. [i.MX RT595](https://www.nxp.com/products/processors-and-microcontrollers/arm-microcontrollers/i-mx-rt-crossover-mcus/i-mx-rt500-crossover-mcu-with-arm-cortex-m33-dsp-and-gpu-cores:i.MX-RT500) has 5MB internal SRAM (can store 50+ 200x200 16 bit images). ESP modules also come with plenty of SRAM.

> a lot of CPU processing by having a vector based drawing engine that can deal successfully (and fast) with a 3D scene (layers of 2D layers = textures transformed by a matrix). 

Actually the current SW renderer is a simplified vector renderer. as it can draw e.g. a rounded rectangle border with a given radius. Okay, it can draw it with 30 deg rotation. That's why it's simplified. :slightly_smiling_face:  Anyway, now we have a render engine which is fast enough to draw all layer the on each screen update without buffering/collapsing anything to images. 

> A vector based engine needs to apply a matrix operation (at least 3x2) for each point drawn, not sure if embedded CPU can deal with such constraint in realtime.

Now it works only for images, but I can confirm that it work in runtime. I'm working on some optimization in image transformation too, and in the current state rotating and zooming as many 100x100 ARGB images that fill a 800x480 screen (~40) takes:
- 60 ms is there is no anti aliasing
- 110 ms with anti aliasing

on a 200 MHz STM32F769. Blending without transformation took 35 ms.
E.g. i.MX RT 595 and 1170 has VG-Lite GPU supporting rotation support.. BTW, I should have cc @GorgonMeducer too in this issue. We are working on a image transformation API for LVGL to allow using ARM GPU too. So event SW transformation is fast now, and more and more MCU will appear with complex GPUs. 
No perspective transformation support now, but I find it very important and plan to add it in the future.

> The best optimization I've seen so far use a tiling approach. 

Actually LVGL is doing something like that. Not dividing the the screen into NxM tiles but refresh only the dirty areas. It searches for the "top object" (which fully covers that area, it's screen in the worst case) and starts rendering from there.

@XuNeo   

I imagined it the same way! The demo you can see in the first comment, stores a snapshot (~ canvas) in each object and renders those canvases as "mini displays".

mariotaku, 2022-04-12:
@kisvegabor Actually for SDL it is pretty close to snapshots. The major difference is SDL implementation is shape based rather than object based, but for most objects, they are really similar.

* lv_obj: Shaped backgrounds are saved as a whole
* lv_img: Pre-processed images are saved
* lv_slider: The knob and the track are saved separately
* lv_label: Glyphs are drawn by font rendering engine

Another idea is to add some flag to mark an object to draw on a texture first. For a view that rarely changes, saving to snapshot will significantly improve performance when it comes to scrolling/scaling.

X-Ryl669, 2022-04-12:
> Actually LVGL is doing something like that. Not dividing the the screen into NxM tiles but refresh only the dirty areas. It searches for the "top object" (which fully covers that area, it's screen in the worst case) and starts rendering from there.

So if you have a label in the screen that's on the bottom right corner and its text is modified (or the position is moved by 1px left), the whole screen will redraw ?
In a tile based approach, if you have 50px tiles (for example), only the last 50px will redraw, not the whole screen.

XuNeo, 2022-04-12:
@kisvegabor I mean to add "snapshot" capability directly to lv_canvas object, not the base lv_obj so we can use the virtual display driver(holds by lv_canvas) and only updating dirty areas calculated from real display.

kisvegabor, 2022-04-12:
@mariotaku 
> For a view that rarely changes, saving to snapshot will significantly improve performance when it comes to scrolling/scaling.

I agree! 

@X-Ryl669 
> So if you have a label in the screen that's on the bottom right corner and its text is modified (or the position is moved by 1px left), the whole screen will redraw ?

Nope. By 
> It searches for the "top object" and starts rendering from there.

I meant ...and start drawn from there but updating only the dirty area.

E.g. is a label is moved from `10;10 100;20` to `10;11 100;21` then `10;10 100;21` area will be updated. If it was on a button with 100% opacity, draw this area of then button and then the label. If the button has 50% opacity, draw this area of the screen, the button and the label. 

@XuNeo 
>  mean to add "snapshot" capability directly to lv_canvas object, not the base lv_obj so we can use the virtual display driver(holds by lv_canvas) and only updating dirty areas calculated from real display.

I see. It's also an interesting approach as it brings all the power of "display"s. I can imagine it on the "intermediate" layers I described in the first post.

GorgonMeducer, 2022-04-12:
> I agree that LVGL should support single MCU scenarios too, but having 16..32MB external SRAM is not that rare today. E.g. [i.MX [...]

Recently, I got some feedback about LVGL (when supporting people using LVGL via cmsis-pack), people are complaining that the latest LVGL is less friendly for MCUs with small memory than before (v5 / v6 they mentioned).  For IoT edge devices, cost pressure is high and equipping SDRAM is not as popular as we want. 

Based on my understanding, the main reasons people choose LVGL are:
1. it is free to use
2. it has a small memory footprint and is friendly for MCU

On the other hand, DTCM is usually small (64K~512K) and SRAMs with mega-byte level sizes are usually slower, not to mention the PRAMs and DRAMs (wait state introduced are from 10 to hundreds of cycles).

In possible, please keep this "low-memory-footprint" feature.

mariotaku, 2022-04-13:
@GorgonMeducer I would agree on providing 2 ways to draw. One is draw-on-demand and another is buffering.

GorgonMeducer, 2022-04-13:
@mariotaku Even buffering, it is better to keep a low-memory footprint.

FASTSHIFT, 2022-04-13:
@kisvegabor 
Great improvement! Can we use the img cache idea to automatically reclaim the memory occupied by snapshots of widgets that are not frequently updated? This is useful for MCUs with relatively sufficient memory (16~32MB) to prevent the memory usage from exceeding the physical memory size.

kisvegabor, 2022-04-13:
Things started to clean up in my head. Now I see 3 **independent** features:
1. Improve SW render engine
    - The user requests to draw 30 deg rotated rectangle and render engine should do it somehow. 
    - If it needs to draw the rectangle into a buffer first to rotate it later, fine. Let's say this is how the current SW renderer works. Probably an SVG engine would do it differently.
    - The SW renderer might be configured the save these (or any other) temporary buffers to make itself faster
    - All these are internal SW render stuff and not visible from outside
    - I think it's the least important now
2. "Intermediate layer"  
    - Needed if opa!=100% and when a "subtree" needs to be rendered to make some operations on it. It'd be faster to perform masking on the result image only once, instead of all children
    - To handle opa and masking a 200x200 image can be drawn in 10 20x200 parts, i.e. it can be rendered in small chunks to avoid using large buffers
    - If a widget is transformed, has perspective, other image-like effect, we can't render in parts but the whole widget needs to available for transformation
    - So the images are rendered non-transformed, non-filtered, non-masked, and all these will be applied while drawing the rendered buffer as an image.
    - All these can be handled during rendering: 
       - As LVGL redraws the widgets, it can see that the next one has e.g. opa=50%
       - Then LVGL sets a new `draw_ctx` with a new buffer and continues the rendering  with the new ctx
       - It can use a smaller buffer and do the rendering in more steps
       - When the rendering is ready use the buffer as images and draw them transformed, translucent, masked, etc
       - Nested intermediate layers can be handled this way too
    - Not considering transformations, invalidation is simple too, as only the  dirty area of the intermediate layers will be redrawn. 
    - With transformations (e.g. if a label has changed on a rotated button), it seems also possible to update only the changed area but it's fully clear now.
    - The buffers used for rendering can be dropped after the rendering and drawing is finished
    - I think it's the most important update as is allows proper opa handling and sub tree transformation
3. Buffer a sub tree on widget level
    - Save the opacity, transformations, masks too
    - Almost like having an intermediate layer on demand, however it needs 2 intermediate layers: one to draw the image non-transformed, and an other which is destination of transformations, filtering, etc. And the latter should be saved.
    - The same effect can be achieved by having a wrapper object around the  subtree to save and tell to not drop the intermediate layer of the wrapper after rendering.
    - Point 2) is required for it, and after that it's quite simple. So it can be done after 2)

---

@GorgonMeducer 
> Recently, I got some feedback about LVGL (when supporting people using LVGL via cmsis-pack), people are complaining that the latest LVGL is less friendly for MCUs with small memory than before (v5 / v6 they mentioned). 

Is it flash or RAM usage? Dis you also get feedback about the improvement (or regression) of performance, API, ease-of-use and flexibility? 

> If possible, please keep this "low-memory-footprint" feature.

Yes, all these should be optional.

@FASTSHIFT 
> Great improvement! Can we use the img cache idea to automatically reclaim the memory occupied by snapshots of widgets that are not frequently updated? This is useful for MCUs with relatively sufficient memory (16~32MB) to prevent the memory usage from exceeding the physical memory size.

Yes, it'd be great to automatically forget unused image buffers. I think it applies to point 3) in my list above.

GorgonMeducer, 2022-04-13:
@kisvegabor most of the feedback is about memory footprint, especially about RAM usage. Didn't hear anything about performance, API, ease-of-use or flexibility.

kisvegabor, 2022-04-14:
Thanks! One more thing: RAM usage of the widgets, or in general?

GorgonMeducer, 2022-04-14:
@kisvegabor In general

kisvegabor, 2022-04-26:
I've implemented and merged ["Intermediate layer"](https://github.com/lvgl/lvgl/issues/3269#issuecomment-1097954232). See the docs here and this example.

The next step could be the caching snapshots. Here I'm thinking about how to handle very large widgets. Maybe some kind of progressive rendering? (cache what is visible +/- 100 px or so). I'll check how HTML caches the content of e.g. `<html>` tag which can be really large. Any input is very welcome on it. :)

kisvegabor, 2022-04-27:
An other use case of this widget rotation feature:
https://docs.lvgl.io/master/widgets/core/arc.html#simple-arc

kisvegabor, 2022-05-14:
My interest was the any widget transform feature and its implemented. I'm also interested in the snapshot but now I also try to make some progress with the driver architecture.

I close this issue for now, we can reopen it if this topic comes up again.

closed 2022-05-14 by kisvegabor

espressif2022, 2022-12-20:
Is this method applicable to the scene where the entire page is moved as title_view ？
Currently the frame rate of my UI drops very quickly during the move（33fps ->10fps），do you have any good suggestions to make title_view switching smoother?

kisvegabor, 2022-12-21:
I'll comment in https://github.com/lvgl/lvgl/issues/3878