lvgl/lvgl #3272: Missing scroll events if animation is off.

Viatorus, 2022-04-13:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

There seems to exist 3 different scroll event behaviors when scrolling is done programmatically (e.g. using lv_obj_scroll_to):

1.  No event is fired if scrolling isn't necessary
2.  SCROLL_BEGIN, SCROLL and SCROLL_END is fired when scrolling is animated
3.  Only SCROLL is fired when scrolling isn't animated

The 3rd behavior doesn't feel right (maybe because I came across this by search for a "bug" in our application).
I thought: If I use non-animated scrolling either no event is fired or all.

### To Reproduce
```py
#!/opt/bin/lv_micropython -i

import lvgl as lv

import display_driver

def callback_scroll_begin(evt):
    print("scroll_begin")

def callback_scroll(evt):
    print("scroll")

def callback_scroll_end(evt):
    print("scroll_end")

panel = lv.obj(lv.scr_act())
panel.set_size(100, 100)
panel.center()

child = lv.obj(panel)
child.set_pos(0, 0)
label = lv.label(child)
label.set_text("Zero")
label.center()

panel.add_event_cb(callback_scroll_begin, lv.EVENT.SCROLL_BEGIN, None)
panel.add_event_cb(callback_scroll, lv.EVENT.SCROLL, None)
panel.add_event_cb(callback_scroll_end, lv.EVENT.SCROLL_END, None)

panel.scroll_to_x(1, lv.ANIM.OFF)
# panel.scroll_to_x(1, lv.ANIM.ON)
```

https://sim.lvgl.io/v8.3/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/016099e09c3bccdf6b0dd5269c1ec4d00eb07b86/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/016099e09c3bccdf6b0dd5269c1ec4d00eb07b86/examples/scroll/lv_example_scroll_1.py&script_direct=1c22ce183931df1f4eeb99179ab48e2713f53049

### Expected behavior

Either:
* add SCROLL_BEGIN, SCROLL__END event to `lv_obj_scroll_to` if animation is off
* remove all events
* document this behavior

kisvegabor, 2022-04-14:
Hi,

it should be fixed here: https://github.com/lvgl/lvgl/commit/777fe1ea706f38b82ab8ee180548ecb85334a469

closed 2022-04-14 by kisvegabor

Viatorus, 2022-04-14:
Wow. Fast as always! Thank you.

kisvegabor, 2022-04-14:
Not always, but happy that you feel this way :smile: