lvgl/lvgl #3276: Why I can not compile the code? It show No such file or directory

tigerya214, 2022-04-15:
Why it show as follow???

> Executing task: ninja  <

[1/22] Performing build step for 'bootloader'
FAILED: bootloader-prefix/src/bootloader-stamp/bootloader-build bootloader/bootloader.elf bootloader/bootloader.bin bootloader/bootloader.map 
cmd.exe /C "cd /D C:\Users\tigerya\Desktop\DIY_WATCH\lv_port_esp32-master\build\bootloader && D:\software\ESP-IDF\tools\cmake\3.20.3\bin\cmake.exe --build ."   
系统找不到指定的路径。
[8/22] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj      
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj 
ccache D:\software\ESP-IDF\tools\xtensa-esp32s3-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-gcc.exe -DHAVE_CONFIG_H -DLV_CONF_INCLUDE_SIMPLE -DLV_LVGL_H_INCLUDE_SIMPLE -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -Iconfig -ID:/software/ESP-IDF/esp-idf/components/newlib/platform_include -ID:/software/ESP-IDF/esp-idf/components/freertos/include -ID:/software/ESP-IDF/esp-idf/components/freertos/include/esp_additions/freertos -ID:/software/ESP-IDF/esp-idf/components/freertos/port/xtensa/include -ID:/software/ESP-IDF/esp-idf/components/freertos/include/esp_additions -ID:/software/ESP-IDF/esp-idf/components/esp_hw_support/include -ID:/software/ESP-IDF/esp-idf/components/esp_hw_support/include/soc -ID:/software/ESP-IDF/esp-idf/components/esp_hw_support/include/soc/esp32s3 -ID:/software/ESP-IDF/esp-idf/components/esp_hw_support/port/esp32s3/. -ID:/software/ESP-IDF/esp-idf/components/esp_hw_support/port/esp32s3/private_include -ID:/software/ESP-IDF/esp-idf/components/heap/include -ID:/software/ESP-IDF/esp-idf/components/log/include -ID:/software/ESP-IDF/esp-idf/components/lwip/include/apps -ID:/software/ESP-IDF/esp-idf/components/lwip/include/apps/sntp -ID:/software/ESP-IDF/esp-idf/components/lwip/lwip/src/include -ID:/software/ESP-IDF/esp-idf/components/lwip/port/esp32/include -ID:/software/ESP-IDF/esp-idf/components/lwip/port/esp32/include/arch -ID:/software/ESP-IDF/esp-idf/components/soc/include -ID:/software/ESP-IDF/esp-idf/components/soc/esp32s3/. -ID:/software/ESP-IDF/esp-idf/components/soc/esp32s3/include -ID:/software/ESP-IDF/esp-idf/components/hal/esp32s3/include -ID:/software/ESP-IDF/esp-idf/components/hal/include -ID:/software/ESP-IDF/esp-idf/components/hal/platform_port/include -ID:/software/ESP-IDF/esp-idf/components/esp_rom/include -ID:/software/ESP-IDF/esp-idf/components/esp_rom/include/esp32s3 -ID:/software/ESP-IDF/esp-idf/components/esp_rom/esp32s3 -ID:/software/ESP-IDF/esp-idf/components/esp_common/include -ID:/software/ESP-IDF/esp-idf/components/esp_system/include -ID:/software/ESP-IDF/esp-idf/components/esp_system/port/soc -ID:/software/ESP-IDF/esp-idf/components/esp_system/port/public_compat -ID:/software/ESP-IDF/esp-idf/components/xtensa/include -ID:/software/ESP-IDF/esp-idf/components/xtensa/esp32s3/include -ID:/software/ESP-IDF/esp-idf/components/driver/include -ID:/software/ESP-IDF/esp-idf/components/driver/esp32s3/include -ID:/software/ESP-IDF/esp-idf/components/esp_pm/include -ID:/software/ESP-IDF/esp-idf/components/esp_ringbuf/include -ID:/software/ESP-IDF/esp-idf/components/efuse/include -ID:/software/ESP-IDF/esp-idf/components/efuse/esp32s3/include -ID:/software/ESP-IDF/esp-idf/components/vfs/include -ID:/software/ESP-IDF/esp-idf/components/esp_wifi/include -ID:/software/ESP-IDF/esp-idf/components/esp_event/include -ID:/software/ESP-IDF/esp-idf/components/esp_netif/include -ID:/software/ESP-IDF/esp-idf/components/esp_eth/include -ID:/software/ESP-IDF/esp-idf/components/tcpip_adapter/include -ID:/software/ESP-IDF/esp-idf/components/esp_phy/include -ID:/software/ESP-IDF/esp-idf/components/esp_phy/esp32s3/include -ID:/software/ESP-IDF/esp-idf/components/esp_ipc/include -ID:/software/ESP-IDF/esp-idf/components/app_trace/include -ID:/software/ESP-IDF/esp-idf/components/esp_timer/include -ID:/software/ESP-IDF/esp-idf/components/mbedtls/port/include -ID:/software/ESP-IDF/esp-idf/components/mbedtls/mbedtls/include -ID:/software/ESP-IDF/esp-idf/components/mbedtls/esp_crt_bundle/include -ID:/software/ESP-IDF/esp-idf/components/app_update/include -ID:/software/ESP-IDF/esp-idf/components/spi_flash/include -ID:/software/ESP-IDF/esp-idf/components/bootloader_support/include -ID:/software/ESP-IDF/esp-idf/components/nvs_flash/include -ID:/software/ESP-IDF/esp-idf/components/pthread/include -ID:/software/ESP-IDF/esp-idf/components/esp_gdbstub/include -ID:/software/ESP-IDF/esp-idf/components/esp_gdbstub/xtensa -ID:/software/ESP-IDF/esp-idf/components/esp_gdbstub/esp32s3 -ID:/software/ESP-IDF/esp-idf/components/espcoredump/include -ID:/software/ESP-IDF/esp-idf/components/espcoredump/include/port/xtensa -ID:/software/ESP-IDF/esp-idf/components/wpa_supplicant/include -ID:/software/ESP-IDF/esp-idf/components/wpa_supplicant/port/include -ID:/software/ESP-IDF/esp-idf/components/wpa_supplicant/esp_supplicant/include -ID:/software/ESP-IDF/esp-idf/components/ieee802154/include -ID:/software/ESP-IDF/esp-idf/components/console -ID:/software/ESP-IDF/esp-idf/components/asio/asio/asio/include -ID:/software/ESP-IDF/esp-idf/components/asio/port/include -ID:/software/ESP-IDF/esp-idf/components/cbor/port/include -ID:/software/ESP-IDF/esp-idf/components/unity/include -ID:/software/ESP-IDF/esp-idf/components/unity/unity/src -ID:/software/ESP-IDF/esp-idf/components/cmock/CMock/src -ID:/software/ESP-IDF/esp-idf/components/coap/port/include -ID:/software/ESP-IDF/esp-idf/components/coap/libcoap/include -ID:/software/ESP-IDF/esp-idf/components/nghttp/port/include -ID:/software/ESP-IDF/esp-idf/components/nghttp/nghttp2/lib/includes -ID:/software/ESP-IDF/esp-idf/components/esp-tls -ID:/software/ESP-IDF/esp-idf/components/esp-tls/esp-tls-crypto -ID:/software/ESP-IDF/esp-idf/components/esp_adc_cal/include -ID:/software/ESP-IDF/esp-idf/components/esp_hid/include -ID:/software/ESP-IDF/esp-idf/components/tcp_transport/include -ID:/software/ESP-IDF/esp-idf/components/esp_http_client/include -ID:/software/ESP-IDF/esp-idf/components/esp_http_server/include -ID:/software/ESP-IDF/esp-idf/components/esp_https_ota/include -ID:/software/ESP-IDF/esp-idf/components/esp_lcd/include -ID:/software/ESP-IDF/esp-idf/components/esp_lcd/interface -ID:/software/ESP-IDF/esp-idf/components/protobuf-c/protobuf-c -ID:/software/ESP-IDF/esp-idf/components/protocomm/include/common -ID:/software/ESP-IDF/esp-idf/components/protocomm/include/security -ID:/software/ESP-IDF/esp-idf/components/protocomm/include/transports -ID:/software/ESP-IDF/esp-idf/components/mdns/include -ID:/software/ESP-IDF/esp-idf/components/esp_local_ctrl/include -ID:/software/ESP-IDF/esp-idf/components/sdmmc/include -ID:/software/ESP-IDF/esp-idf/components/esp_serial_slave_link/include -ID:/software/ESP-IDF/esp-idf/components/esp_websocket_client/include -ID:/software/ESP-IDF/esp-idf/components/expat/expat/expat/lib -ID:/software/ESP-IDF/esp-idf/components/expat/port/include -ID:/software/ESP-IDF/esp-idf/components/wear_levelling/include -ID:/software/ESP-IDF/esp-idf/components/fatfs/diskio -ID:/software/ESP-IDF/esp-idf/components/fatfs/vfs -ID:/software/ESP-IDF/esp-idf/components/fatfs/src -ID:/software/ESP-IDF/esp-idf/components/freemodbus/common/include -ID:/software/ESP-IDF/esp-idf/components/idf_test/include -ID:/software/ESP-IDF/esp-idf/components/idf_test/include/esp32s3 -ID:/software/ESP-IDF/esp-idf/components/jsmn/include -ID:/software/ESP-IDF/esp-idf/components/json/cJSON -ID:/software/ESP-IDF/esp-idf/components/libsodium/libsodium/src/libsodium/include -ID:/software/ESP-IDF/esp-idf/components/libsodium/port_include 
-ID:/software/ESP-IDF/esp-idf/components/mqtt/esp-mqtt/include -ID:/software/ESP-IDF/esp-idf/components/openssl/include -ID:/software/ESP-IDF/esp-idf/components/perfmon/include -ID:/software/ESP-IDF/esp-idf/components/spiffs/include -ID:/software/ESP-IDF/esp-idf/components/usb/include -ID:/software/ESP-IDF/esp-idf/components/ulp/include -ID:/software/ESP-IDF/esp-idf/components/wifi_provisioning/include -I../components/lvgl -I../components/lvgl/src -I../components/lv_examples 
-I../components/lvgl_esp32_drivers -I../components/lvgl_esp32_drivers/lvgl_tft -mlongcalls -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -fmacro-prefix-map=C:/Users/tigerya/Desktop/DIY_WATCH/lv_port_esp32-master=. -fmacro-prefix-map=D:/software/ESP-IDF/esp-idf=IDF -f [... cut ...] ain.dir\main.c.obj.d -o esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj -c ../main/main.c
../main/main.c:34:18: fatal error: lv_examples/src/lv_demo_widgets/lv_demo_widgets.h: No such file or directory
         #include "lv_examples/src/lv_demo_widgets/lv_demo_widgets.h"
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
终端进程“C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command ninja ”已终止，退出代码: 1。

kisvegabor, 2022-04-16:
Are you using LVGL v7 or v8? 

cc @C47D

C47D, 2022-04-16:
From the provided information I can assume you're using ESP32S3 chip as target. @tigerya214 is this a custom project or lv_port_esp32?

tigerya214, 2022-04-19:
It is  lv_port_esp32, I cloned the repo by this code
`git clone --recurse-submodules https://github.com/lvgl/lv_port_esp32.git`

closed 2022-05-12 by lvgl-bot