lvgl/lvgl #3281: Select raw format  in Online Image Converter will create header.cf LV_IMG_CF_RAW_CHROMA undefine

acer1204, 2022-04-18:
i use LVGL v8.2 in stm32h743.
it work fine now.
but i try to use lvgl's gif decoder function.
when i use gif example, it work.
but if i create my gif file in to online image convert and select raw with c array format.
it will download a *.c file. 
buf the header format is LV_IMG_CF_RAW_CHROMA, and the lvgl v8.2 show this is undefine.
if i just change LV_IMG_CF_RAW_CHROMA to LV_IMG_CF_RAW. it compile work.
but download my board, the gif not show anything.
just a white screen.
is this is a bug with  online image convert ?

embeddedt, 2022-04-19:
Are you using the GIF object as described [here](https://docs.lvgl.io/master/libs/gif.html)?

acer1204, 2022-04-19:
> 

yes.
use this like example *.c file can run it.
but if i create the *.c file by this link [LVGL's online converter](https://lvgl.io/tools/imageconverter).
select c array and raw format.
it will not work at all.

it create header format is LV_IMG_CF_RAW_CHROMA,
and this define is not work in lvgl 8.2.