lvgl/lvgl #3286: rotation is not working with frambuffer

embexus, 2022-04-20:
Hello I'm trying to get rotation working with frambuffer but so far no luck, here is the code I used :

```
diff --git a/main.c b/main.c
index 33cd734..88f652c 100644
--- a/main.c
+++ b/main.c
@@ -11,6 +11,8 @@
 
 int main(void)
 {
+    lv_disp_rot_t rot=LV_DISP_ROT_NONE;
+
     /*LittlevGL init*/
     lv_init();
 
@@ -31,7 +33,7 @@ int main(void)
     disp_drv.flush_cb   = fbdev_flush;
     disp_drv.hor_res    = 800;
     disp_drv.ver_res    = 480;
-    lv_disp_drv_register(&disp_drv);
+    lv_disp_t *lv_displ=lv_disp_drv_register(&disp_drv);
 
     evdev_init();
     static lv_indev_drv_t indev_drv_1;
@@ -53,6 +55,11 @@ int main(void)
     /*Create a Demo*/
     lv_demo_widgets();
 
+    /*Rotate Display*/
+    lv_disp_set_rotation(lv_displ, LV_DISP_ROT_90);
+    rot=lv_disp_get_rotation(lv_displ);
+    printf("rotation is %d\n", rot);
+
     /*Handle LitlevGL tasks (tickless mode)*/
     while(1) {
         lv_timer_handler();
``

Display rotation doesn't work but it looks like the touch with evdev has rotated !

embeddedt, 2022-04-20:
The framebuffer driver doesn't have hardware rotation support, so I think you need to set `disp_drv.sw_rotate = 1` as well.

embexus, 2022-04-20:
@embeddedt thanks so much it is working now with follwing chnages (had to invert also x/y resolution:

```
diff --git a/main.c b/main.c
index 33cd734..df0394b 100644
--- a/main.c
+++ b/main.c
@@ -11,6 +11,8 @@
 
 int main(void)
 {
+    lv_disp_rot_t rot=LV_DISP_ROT_NONE;
+
     /*LittlevGL init*/
     lv_init();
 
@@ -29,9 +31,10 @@ int main(void)
     lv_disp_drv_init(&disp_drv);
     disp_drv.draw_buf   = &disp_buf;
     disp_drv.flush_cb   = fbdev_flush;
-    disp_drv.hor_res    = 800;
-    disp_drv.ver_res    = 480;
-    lv_disp_drv_register(&disp_drv);
+    disp_drv.hor_res    = 480;
+    disp_drv.ver_res    = 800;
+    disp_drv.sw_rotate  = 1;
+    lv_disp_t *lv_displ=lv_disp_drv_register(&disp_drv);
 
     evdev_init();
     static lv_indev_drv_t indev_drv_1;
@@ -53,6 +56,11 @@ int main(void)
     /*Create a Demo*/
     lv_demo_widgets();
 
+    /*Rotate Display*/
+    lv_disp_set_rotation(lv_displ, LV_DISP_ROT_90);
+    rot=lv_disp_get_rotation(lv_displ);
+    printf("rotation is %d\n", rot);
+
     /*Handle LitlevGL tasks (tickless mode)*/
     while(1) {
         lv_timer_handler();
```

closed 2022-04-20 by embexus