lvgl/lvgl #3289: Errors related unit test when compiled with mbed-os

zhiyong-ft, 2022-04-23:
### Introduce the problem
When included as library into a project that already has unity included, a lot of errors will be thrown because of duplicated unity files.

### Examples and cases
When LVGL was included into my mbed-os based project, I get numerous errors like the following during compilation.
```
[Error] unity.c@24,43: invalid use of undefined type 'struct _Unity'
[Error] unity.c@24,77: invalid use of undefined type 'struct _Unity'
[Error] unity.c@1005,22: invalid use of undefined type 'struct _Unity'
[Error] unity.c@1014,18: invalid use of undefined type 'struct _Unity'
[Error] unity.c@1018,14: invalid use of undefined type 'struct _Unity'
[Error] unity.c@1030,35: unknown type name '_UU32'
[Error] unity.c@1032,35: unknown type name 'UNITY_LINE_TYPE'
[Error] unity.c@1034,5: unknown type name '_UU32'
[Error] unity.c@24,43: invalid use of undefined type 'struct _Unity'
[Error] unity.c@24,77: invalid use of undefined type 'struct _Unity'
[Error] unity.c@1045,22: invalid use of undefined type 'struct _Unity'
[Error] unity.c@1054,18: invalid use of undefined type 'struct _Unity'
[Error] unity.c@1058,14: invalid use of undefined type 'struct _Unity'
[Error] unity.c@1071,41: unknown type name '_UU32'
[Error] unity.c@1073,41: unknown type name 'UNITY_LINE_TYPE'
[Error] unity.c@1075,5: unknown type name '_UU32'
[Error] unity.c@24,43: invalid use of undefined type 'struct _Unity'
[Error] unity.c@24,77: invalid use of undefined type 'struct _Unity'
```

### Suggested solution

For the time being, we just deleted the tests folder inside lvgl, after that we can compile and run our code. It would be nice if some more graceful way can be implemented to manage this.

embeddedt, 2022-04-25:
These files are wrapped in `LV_BUILD_TEST` ([example](https://github.com/lvgl/lvgl/blob/07b7eea56c048a0654c254cadebee8caf5f7933b/tests/unity/unity.c#L6)), so in theory they shouldn't be compiled in other projects at all. Is it our `unity.c` file that produces these errors or yours?

zhiyong-ft, 2022-05-02:
> These files are wrapped in `LV_BUILD_TEST` ([example](https://github.com/lvgl/lvgl/blob/07b7eea56c048a0654c254cadebee8caf5f7933b/tests/unity/unity.c#L6)), so in theory they shouldn't be compiled in other projects at all. Is it our `unity.c` file that produces these errors or yours?

The build errors were in the other library that also contains unity.c. Once we added the following line into lvgl_conf.h
```
#define LV_BUILD_TEST 0
```
Compilation with tests folder present in lvgl succeeded without error.

closed 2022-05-02 by zhiyong-ft