lvgl/lvgl #3293: Update PlatformIO library to v8.2.0 ?

vshymanskyy, 2022-04-25:
It's still on 8.1.0: https://registry.platformio.org/libraries/lvgl/lvgl

embeddedt, 2022-04-25:
Something must have gone wrong; @kisvegabor apparently [did attempt to update it](https://github.com/lvgl/lvgl/issues/2790#issuecomment-1065936264) back in March. Ideally we would automate it as part of a release, but if I recall correctly, there were issues with that.

kisvegabor, 2022-04-26:
Yes, I really updated it and got confirmation email about it succeeded. I'm looking into it.

kisvegabor, 2022-04-26:
I tried to publish it again and got this:
```
kisvegabor ~/projects/lvgl/lv_platformio/.pio/libdeps/emulator_64bits/lvgl $pio package publish
Preparing a package...
Type:     library
Owner:    kisvegabor
Name:     lvgl
Version:  8.2.0
Error: The package `kisvegabor/lvgl@8.2.0` is already published in the registry
sh: parse_git_branch: command not found
```

Even the star count sits on an the 5 month old value.
[image]

vshymanskyy, 2022-04-26:
@ivankravets please take a look

vshymanskyy, 2022-04-26:
@kisvegabor I just reached out to Ivan, he should be able to check it soon

kisvegabor, 2022-04-26:
It turned out that I'm very good at messing up publishing at Pio, so probably I made some mistake. :fearful:

ivankravets, 2022-04-26:
Hi all, thanks for mentioning me!

@kisvegabor, I've removed duplicated package `kisvegabor/lvgl@8.2.0`. You are the owner of [LVGL](https://registry.platformio.org/search?q=owner%3Alvgl) organization. 

You MUST pass `--owner lvgl` to the `pio package publish` command. So, please use 

```
pio package publish --owner lvgl
```

kisvegabor, 2022-04-26:
Thank our for the super quick reply, Ivan! I've just published it with that command and it's already updated.

Now I saved that command to my cheat sheet. I hope we don't need to bother you with similar issue in future. :slightly_smiling_face:

ivankravets, 2022-04-26:
No problem, thanks for your amazing project!

P.S: I also manually updated the release date for 8.2.0 https://registry.platformio.org/libraries/lvgl/lvgl/versions

closed 2022-04-26 by kisvegabor

vshymanskyy, 2022-04-26:
All is good! Thanks, everyone.
Now my [PIO + LVGL + Blynk.Edgent example](https://github.com/blynkkk/edgent-examples-pio/tree/main/PIO_Edgent_Wio_Terminal) works flawlessly on Seeed Wio Terminal:
[image]

kisvegabor, 2022-04-27:
Looks great!