lvgl/lvgl #3297: imgbtn positioning problem

landerega, 2022-04-26:
### Perform all steps below and tick them with [x]
- [X] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [X] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [X] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
When I use image buttons the image button itself is not positioned correctly. I use the "_lv_obj_align_to_" function to align the image to the position I want but the showing position is not OK

### To Reproduce

### Expected behavior
I expect the image to show in the middle of the screen but it shows with some offset. I cannot confirm this but it looks to me that this offset is the same as the image width and height. 

During the execution (in onInputEvent function) I do the following:
    	    lv_obj_set_size(self->btnPrg1, wine.header.w, wine.header.h);
    	    lv_obj_align(self->btnPrg1, LV_ALIGN_CENTER, 0, 0);

I set the size of the object to be the same as the image and then I recenter the image. This way the position is set correctly.

Is there any reason the imgbtn object does not take the size of the image from the beginning automatically?

kisvegabor, 2022-05-01:
The height of the image button is set the to the image's height, but you need to set the width manually. 
Actually this the purpose of the image button, is that the center part is repeated to the width you set.

landerega, 2022-05-02:
Ok thank you, If that is the way the image button is supposed to work I will set the width of each button manually. I was afraid that I was doing something wrong or some configuration was missing.

Thank you, I will close the issue

closed 2022-05-02 by landerega