lvgl/lvgl #3298: v9 reminders

kisvegabor, 2022-04-26:
Collect the refactoring and other ideas  here for v9.

- `lv_style_set_size()` should have separate width and height parameters
- `lv_img_set_src()` use "type-aware" parameter. See [here](https://github.com/lvgl/lvgl/tree/arch/img-decode-rework)
- Rename `LV_IMG_CF_TRUE_COLOR_*` to `LV_IMG_CF_RGB/RGBA`etc
- Update the `lv_img_dsc_t` to support images larger than 2048x2048
- `lv_mem_alloc/realloc/free` to `lv_malloc/realloc/free`?
- Drop `lv_mem_buf_get` as tlsf should be fast enough for normal allocations too. Fragmentation is also lower if processes can completely clean up after themselves.

The list will be longer as new ideas come. Feel free to comment if you also have one. :slightly_smiling_face:

KLVN, 2022-05-25:
- 3D transformations on images (https://forum.lvgl.io/t/rotate-image-image-in-the-z-direction/3963/6)

kisvegabor, 2022-05-26:
Perspective distortion can be added in any minor version as it doesn't require a conceptual change. It's just an other option next to scale and rotation.

Cuixudong, 2022-06-01:
Add control animation group to reduce the amount of independent control animation code.
for example：https://www.bilibili.com/video/BV1EV411s7tK   and   https://www.bilibili.com/video/BV1644y1z7MS

Cuixudong, 2022-06-02:
> For this I checked GSAP stagger as inspiration/reference: https://greensock.com/docs/v3/Staggers What do you think about it?

Looks great.

kisvegabor, 2022-06-03:
I added a reminder about staggers

GorgonMeducer, 2022-06-14:
Placeholder: 

- Introduce 9patch-images widgets/style
- Introduce an option to replace RGB565 with RGB565-A8 when `LV_COLOR_DEPTH` to `16` in benchmark

kisvegabor, 2022-06-14:
> Introduce an option to replace RGB565 with RGB565-A8 when LV_COLOR_DEPTH to 16 in benchmark

Can't we do it now in v8.3?

I added the other two to the list.

kctan0805, 2022-06-18:
Our platform supports HW acceleration on image format RGB565 and RGBA8888.
If I set LV_COLOR_DEPTH as 16, the LV_IMG_CF_TRUE_COLOR_ALPHA image will be RGB565-A8.
If I set LV_COLOR_DEPTH as 32, both LV_IMG_CF_TRUE_COLOR and LV_IMG_CF_TRUE_COLOR_ALPHA images will be RGBA8888.
Could we have an option to set LV_IMG_CF_TRUE_COLOR_ALPHA image as RGBA8888 format even LV_COLOR_DEPTH == 16?

weblqb, 2022-07-03:
Build-in thread-safe if it is practical. Thanks! ^_^

kisvegabor, 2022-07-06:
I merged the [ROADMAP](https://github.com/lvgl/lvgl/blob/master/docs/ROADMAP.md) with the list of ideas in the first post. 

I collected the  API breaking changes in to the "Plannel for v9" section. "Planned in general" contains ideas that can be added in any minor version therefore they not critical for v9.

Please take a look at the list and tell what you think. 

My approach is to start with with the simple tasks (naming, minor refactoring) and let the few complex ones for later (masking, color formats, label features).

Cuixudong, 2022-07-09:
what about PicoGL?

GorgonMeducer, 2022-07-09:
@kisvegabor did you forget the 9paches-image widgets?

kisvegabor, 2022-07-09:
@Cuixudong For for what purpose do you mean PicoGL?

@GorgonMeducer It's already added [here](https://github.com/lvgl/lvgl/blob/master/docs/ROADMAP.md#styles).

Mair, 2022-07-13:
@kisvegabor please  keep me updated to the progress (I'll also check in periodically). I'll start updating the course as soon as its stable :)

kisvegabor, 2022-07-19:
I'm thinking about going back to a flatter directory structure. It means e.g. moving `extra/widgets` to `src/widgets`. The goal of this change would be make the features more visible and unified. Storing widgets in two places look odd in the practice. 

Do see any problems with it?

pete-pjb, 2022-07-19:
Hi @kisvegabor ,

Having all the widgets in one place makes good sense and I gets my vote 🙂 

Cheers,

Pete

GorgonMeducer, 2022-07-19:
@kisvegabor Once you make this happen, please could you also update the PDSC file?

https://github.com/lvgl/lvgl/blob/61048554914eee36630f492ebe4e4742b51fa2c1/env_support/cmsis-pack/LVGL.lvgl.pdsc#L444-L474

kisvegabor, 2022-07-19:
@GorgonMeducer 
Yes, no problem. 
FYI, I"ve also started to refactor `lv_conf.h` which probably affects the CMSIS pack too.

kisvegabor, 2022-07-20:
@GorgonMeducer I update the directory structure and LVGL.lvgl.pdsc too. a16e6a83

geert-KLA-BE, 2022-08-11:
I would like to have vertical aligment of labels. Having another object to around it do that is rather wasteful.

kisvegabor, 2022-08-15:
What is the use case when you can't simply position the label in y relative to some other object?

geert-KLA-BE, 2022-08-17:
> What is the use case when you can't simply position the label in y relative to some other object?

For example I want a simple list of fixed heigth label buttons. I simply want to vertical align the text. This can only be done if I wrap all of them in a seperate object with a label in them. This is rather wasteful on resources just for vertical alignment of the text.

[image]

geert-KLA-BE, 2022-08-17:
Currently it seems impossible to add custom (pointer) input events. For example double tapping on a button, adding custom gestures.
Having this it would also be useful to have multi touch support. So an application could add custom gestures for that like a pinch to zoom, two finger scrolling, 2 finger rotation. It might not make sense for low end embedded devices but on powerful embedded devices it is possible to use these.

Cuixudong, 2022-08-19:
I think the list item can be added to the left sliding menu to display the top and delete buttons, which is easier than using composite controls to achieve.
[image]

h13-0, 2022-08-29:
Will lvgl V9 consider supporting methods such as `lv_tileview_set_anim_time` and `lv_tileview_set_edge_flash` in lvgl V7?

kisvegabor, 2022-08-31:
There is an [anim](https://docs.lvgl.io/master/overview/style-props.html#anim) style property which an/should be used on more places to provide better control over the animation. I think we could use it for the tileview too.

Jpe230, 2022-09-13:
With the removal of byte swap for RGB565, what's the recommended approach to swap bytes for displays that requires it? At a driver level?

GorgonMeducer, 2022-09-13:
@Jpe230 Yes, swap hi-low bytes in `disp_flush()`

GorgonMeducer, 2022-09-13:
@xiaoxiang781216 how many bytes per pixel? 3 or 4?

xiaoxiang781216, 2022-09-13:
Source may any format supported by lvgl, but the display surface is 3bytes.

kisvegabor, 2022-09-15:
> With the removal of byte swap for RGB565, what's the recommended approach to swap bytes for displays that requires it? Swap them at a driver level

In `master` there is [disp_drv->color_format](https://github.com/lvgl/lvgl/blob/master/src/hal/lv_hal_disp.h#L107) that you can set to `LV_COLOR_FORMAT_NATIVE_REVERSE`. I've tested that it's almost as fast as `LV_COLOR_16_SWAP` was in v8. 

It's not documented yet as it might still change in the future.

> Can we support 24bit RGB display driver interface?

Yes, definitely. It'd be very important to add support to it in v9. I'm not sure about the performance, as addressing 3 bytes probably won't be that fast. Converting RGBX8888 to RGB888 at the end might be faster. I'm not sure yet.

h13-0, 2022-09-17:
# About lv_tileview
## How to enable Elastic scroll?
I want to achieve this effect:  
[image]

When I use
```C
  lv_obj_t* tileView = lv_tileview_create(lv_scr_act());
  lv_obj_add_flag(tileView, LV_OBJ_FLAG_SCROLL_ELASTIC);
```
but it didn't work:
[image]

## How to set momentum or sliding switching speed?
When I switch pages and release the mouse, I want to control the automatic homing speed of tileview.  
For example, I hope tileview is always so fast:
[image]

instead of:  
[image]

I use:
```C
  lv_obj_set_style_anim_speed(tileView, 1000, 0);
  lv_obj_set_style_anim_time(tileView, 1, 0);
  lv_obj_clear_flag(tileView, LV_OBJ_FLAG_SCROLL_MOMENTUM);
```
but nothing happend.

## My code here
```C
  lv_obj_t* tileView = lv_tileview_create(lv_scr_act());
  lv_obj_add_flag(tileView, LV_OBJ_FLAG_SCROLL_ELASTIC);
  lv_obj_set_style_anim_speed(tileView, 1000, 0);
  lv_obj_set_style_anim_time(tileView, 1, 0);
  lv_obj_clear_flag(tileView, LV_OBJ_FLAG_SCROLL_MOMENTUM);

  lv_obj_t* page1 = lv_tileview_add_tile(tileView, 0, 0, LV_DIR_BOTTOM);
  lv_obj_t* text1 = lv_label_create(page1);
  lv_label_set_text(text1, "Page 1");
  lv_obj_set_style_text_font(text1, &lv_font_montserrat_24, 0);
  
  
  lv_obj_t* page2 = lv_tileview_add_tile(tileView, 0, 1, LV_DIR_TOP | LV_DIR_RIGHT); 
  lv_obj_t* text2 = lv_label_create(page2);
  lv_label_set_text(text2, "Page 2");
  lv_obj_set_style_text_font(text2, &lv_font_montserrat_24, 0);
```
Tested on lvgl8.2 and 8.3.  
Should i change the implementation or wait for the lvgl9?
Thank you.

kisvegabor, 2022-09-19:
@h13-0 
If you don't add a direction to a tile scrolling will be disabled into that direction. So you can do this:
```c
  lv_obj_t* page1 = lv_tileview_add_tile(tileView, 0, 0, LV_DIR_TOP | LV_DIR_BOTTOM);

  lv_obj_t* page2 = lv_tileview_add_tile(tileView, 0, 1, LV_DIR_BOTTOM | LV_DIR_TOP | LV_DIR_RIGHT);
```

@xlbao123 
How do you mean it in more detail?

xlbao123, 2022-09-19:
I'm just getting started with lvgl and I see that it seems the image cache mechanism is still relatively simple, do we have any plans to provide a consistent cache management module for image and glyph bitmap

xlbao123, 2022-09-19:
> What I want is to be able to call a unified trace callback on critical paths and branches of ui engine execution when the trace switch is turned on, so that I can access logs or a module like systrace to analyze performance bottlenecks

GorgonMeducer, 2022-09-19:
> I'm just getting started with lvgl and I see that it seems the image cache mechanism is still relatively simple, do we have any plans to provide a consistent cache management module for image and glyph bitmap

When doing this, please keep in mind that LVGL also serves MCU with small RAM.

sukesh-ak, 2022-11-14:
@kisvegabor What is the approx timeline for v9 release?
I have a project in ESP-IDF 4.x and planning to move to IDF 5.x
Currently the project runs with lvgl v9 but I am sure it won't be stable.

So if release timeline is far for lvgl v9, I will move back to v8.x for now.

kisvegabor, 2022-11-14:
Still there is a lot to do so I suggest sticking with v8 :slightly_smiling_face:

sukesh-ak, 2022-11-14:
> Still there is a lot to do so I suggest sticking with v8 🙂

Sure thank you :)

LuWB, 2022-12-08:
In lv_draw_img_dsc_t,replace 'zoom' by 'scaling_h' and 'scaling_v'.Sometimes,lv_img need fiil object.

kisvegabor, 2022-12-08:
I agree and it's planned. See the ROADMAP:
https://github.com/lvgl/lvgl/blob/master/docs/ROADMAP.md#styles

LonerDan, 2023-01-10:
It could be nice to rethink and unify the configuration layer to make it easier to keep lv_conf.h in sync with Kconfig. We could perhaps use the Kconfig to define all the config options, their dependencies and default values and then use a script to generate the lv_conf_template.h/lv_conf_internal.h in a similar fashion we now generate lv_conf_internal.h from lv_conf_template.h.

kisvegabor, 2023-01-11:
We were thinking about it but IIRC the problem was that Kconfig doesn't support "text" value type. E.g. for
```c
#define LV_MEM_CUSTOM_MALLOC    malloc
```

So `lv_conf.h` seems superior and we could generate Kconfig from it. We "just" need to parse the decoration comments to identify the levels for Kconfig. In theory it could work but in the practice a lot of custom hacks were required to have even a hand written Kconfig file.

Anyway, maybe we just need someone who spends some time with to find a solution.  Would you be interested to look into it? :slightly_smiling_face:

LonerDan, 2023-01-11:
> We were thinking about it but IIRC the problem was that Kconfig doesn't support "text" value type. E.g. for
Per the format definition, Kconfig supports (among others) 'string' type, so it should be usable afaik
https://www.kernel.org/doc/html/latest/kbuild/kconfig-language.html

sukesh-ak, 2023-01-11:
> We were thinking about it but IIRC the problem was that Kconfig doesn't support "text" value type. E.g. for
>
> ```c
> #define LV_MEM_CUSTOM_MALLOC    malloc
> ```
>
> So `lv_conf.h` seems superior and we [...]

It can be old school `ini` file format too. The one mostly used in Linux.
I use this but its `C++` though
https://github.com/Rookfighter/inifile-cpp

sukesh-ak, 2023-01-11:
kconfig is difficult usually to navigate when there are more settings in it.
Header file is easy to do text search.

xiaoxiang781216, 2023-01-11:
Kconfig can describe the dependence, choice, if/endif and comment/help, it's more expressible than the simple ini file. I would suggest that the reverse path which convert Kconfig to other file format instead.

LonerDan, 2023-01-11:
> kconfig is difficult usually to navigate when there are more settings in it. Header file is easy to do text search.

I don't see why the header file would be any easier to search through than Kconfig file. Both are well-structured text files. It also doesn't make sense to introduce any new format, since the Kconfig won't probably go away when it's used at least by esp-idf afaik.

sukesh-ak, 2023-01-11:
> I don't see why the header file would be any easier to search through than Kconfig file. Both are well-structured text files. It also doesn't make sense to introduce any new format, since the [...]

I use kconfig for ESP-IDF projects. 
But difficulty I mentioned is about navigating through menu system. That's how people usually use.

I am not suggesting to move to `ini` file but header is easier in the current context for `lvgl`.
kconfig when its touched, runs a full compile of the project so I don't usually prefer for frequently changed config elements.
Though, in the case of `lvgl`, frequent changes happen only while testing perf tweaks.

LonerDan, 2023-01-11:
> I use kconfig for ESP-IDF projects. But difficulty I mentioned is about navigating through menu system. That's how people usually use.

The tooling around Kconfig is extensive and the most commonly used 'menuconfig' is not the only way to create/modify config. There are available multiple graphical frontends that even allow you to search and other advanced features. Have a look at kconfig-frontends project.

> I am not suggesting to move to `ini` file but header is easier in the current context for `lvgl`. kconfig when its touched, runs a full compile of the project so I don't usually prefer for [...]

Kconfig only creates config file with defines. What you do with it next is up to you. With a good integration with the build system you use, you can rebuild only the files affected by the config change.

kisvegabor, 2023-01-11:
I also find `.h` simpler but as a large number of people and systems are using Kconfig we should support it as well. 

The question is from which format can we generate the other.  As Konfig is not able to describe all types  we need I think we can't start from it.

sukesh-ak, 2023-01-11:
> Kconfig only creates config file with defines. What you do with it next is up to you. With a good integration with the build system you use, you can rebuild only the files affected by the config change.

I agree but was talking about ESP-IDF.
Any changes to configuration through `menuconfig` (which uses kconfig) updates `sdkconfig` which triggers a full compile.

LonerDan, 2023-01-11:
@kisvegabor I see what you mean, Kconfig can't produce unquoted literals without preprocessing by some other tool.
Since those #defines are predominantly used to specify custom platform abstraction (which is in most cases not that frequently changed compared to other options), we could use Kconfig configuration to specify the name of a header file that contains those definitions.
Basically, split the lvgl configuration into 'lv_conf.h' for generic options and 'lv_platform.h' with platform abstraction/customization. Kconfig then could be used to both generate 'lv_conf.h' and specify name of the platform specific header to include. We could even host 'lv_platform.h' files for common platforms.

kisvegabor, 2023-01-12:
Something like this could work, but I can't what's the problem with generating Kconfig from `lv_conf.h`. In a header file we have infinite flexibility and we can add any markup (comments) to indicate types, hierarchy or so and our custom converter can parse these markup.

LonerDan, 2023-01-12:
Why introduce our custom markup, create custom parser for it and introduce extra effort to maintain it when we can leverage already existing tooling and libraries ([kconfiglib](https://github.com/ulfalizer/Kconfiglib) comes to mind, although it currently deals with [its author dissapearing](https://github.com/ulfalizer/Kconfiglib/issues/121))
Also, splitting the configuration header file may be beneficial from a separation of concerns point of view, since single config file could get quite bloated with addition of new features/backends/platforms down the line.

kisvegabor, 2023-01-13:
> Why introduce our custom markup, create custom parser for it and introduce extra effort to maintain it when we can leverage already existing tooling and libraries [...]

The advantage of a header file is that it's "tooling free". Just like LVGL: not special build system, tools, black magic, just add the files and compile them as you wish/can. I'd like to keep this as default, because it's the most flexible. We can build something on top of it though.

So in the end we can't avoid having one or more `.h` config files which needs to be handwritten or generated.

Anyway, part of `lv_conf.h` can be really generated from Kconfig, but still need a generator which parsed the Kconfig file create a config file from it that we like.

Splitting `lv_conf.h` to more files could also work. It seems we will need these files:
- `lv_conf_platform.h`- memory manager, big endian, etc
- `lv_conf_driver.h` - we will merge the driver in to LVGL they can have a separate file too. E.g. enable ILI934, or Linux fbdev driver. Maybe can be merged into `lv_conf_platform.h`.
- `lv_conf_features.h`  - widgets, other modules
```  

The last two seems Kconfig compatible.

LonerDan, 2023-01-27:
Hi @kisvegabor
What is your stance on build system integration in lvgl? There seem to be project files for Make, CMake, SConscript, whose up-to-date state seems to be based primarily on efforts of people using/needing them for their projects. I needed to integrate lvgl into my [meson](https://mesonbuild.com) based project so I created a supporting wrapper for it. One of the nice things meson allows when adding subprojects that don't use meson themselves is the ability to [add the build files for it separately, out-of-tree](https://mesonbuild.com/Wrap-dependency-system-manual.html#wrapfile-with-meson-build-patch). It also has a kind of [dependency repository](https://mesonbuild.com/Wrapdb-projects.html) containing definitions for many project (both those using meson themselves and those with community added support).
My question is, would you be open to me submitting meson support to lvgl directly or would you prefer I submit it as an out-of-tree wrap to WrapDB instead?

kisvegabor, 2023-01-28:
Hi @LonerDan,

As I see, meson's popularity is increasing and I'm absolutely open to add meson build system support to LVGL.  

> My question is, would you be open to me submitting meson support to lvgl directly or would you prefer I submit it as an out-of-tree wrap to WrapDB instead?

I'm not familiar with WrapDB. What would be the pros and cons of having meson support directly in LVGL compared to adding it WrapDB? Does it makes sense to do both?

LonerDan, 2023-01-28:
> I'm not familiar with WrapDB. What would be the pros and cons of having meson support directly in LVGL compared to adding it WrapDB? Does it makes sense to do both?

Yes, in case of native meson support right in lvgl it would still make sense to add it to wrapdb. We would just submit a wrap file (similar to other types of manifest files, example [here](https://github.com/mesonbuild/wrapdb/blob/master/subprojects/tracy.wrap)). That would allow users to simply add lvgl to their project by calling `meson wrap install lvgl`. I am still trying to figure out how to best handle the configuration of lvgl via meson (ie. the least hacky way that leverages strengths of the build system). I think it would work better if we had the configuration Kconfig centric as discussed previously since meson has module named 'keyval' that can load Kconfig output quite easily. I will try few experiments and let you know.

kisvegabor, 2023-01-28:
Sounds great! :relaxed: 

Let me know if I can help with something.

[69 more replies]

closed 2023-12-04 by kisvegabor