lvgl/lvgl #3299: Latest Master Commits today appear to be causing a problem if an object has OPA value less than 100%.

pete-pjb, 2022-04-26:
Since the latest master commits today when LVGL attempts to render an object with an OPA value less than 100%, the following message is generated if logging is enabled:

[Warn]  (120.944, +167)  refr_obj: Rendering this widget needs LV_COLOR_SCREEN_TRANSP 1         (in lv_refr.c line #872)

The object concerned is also not being rendered correctly.

Just as an experiment I have tried LV_COLOR_SCREEN_TRANSP 1 in lv_conf.h and rebuilt the library

This fixes the problem in the simulator with the test code below, but on my target hardware I just get a blank black screen with the same configuration.  
I also tried adding the following lines to my driver initialisation code but I still just get a blank black screen on my target system.

```c 
lv_obj_set_style_bg_opa(lv_scr_act(), LV_OPA_TRANSP, LV_PART_MAIN | LV_STATE_DEFAULT);
lv_disp_set_bg_opa(NULL, LV_OPA_TRANSP);
````

Some test code:
```c
#define NO_MSG_SLOTS 		10
#define MSG_B_SIZEX			1014
#define MSG_B_SIZEY			40
#define MSG_ANIM_TIME		1000
#define MSG_DISP_TIME		        5000

static void anim_x_cb(void * var, lv_coord_t v) {

    lv_obj_set_x(var, v);
}
static void anim_x_cb1(void * var, lv_coord_t v) {	// This is required as you can't call the same function from two different animation parts!

    lv_obj_set_x(var, v);
}
char *sys_msgs[] = {
		"System message 1",
		"System message 2",
		"System_message 3",
};

static void show_sys_message( uint16_t message_id ) {

	static uint8_t				msg_pos = 0, first_call = true;
	static lv_obj_t				*mbox[NO_MSG_SLOTS];
	static lv_style_t			mbox_style;
	static lv_anim_t 			a1[NO_MSG_SLOTS], a2[NO_MSG_SLOTS];
	static lv_anim_timeline_t	*anim_timeline[NO_MSG_SLOTS];

	if( first_call ) {
		lv_style_init(&mbox_style);
		lv_style_set_radius(&mbox_style, 3);
		lv_style_set_opa(&mbox_style, LV_OPA_70);
		lv_style_set_border_width(&mbox_style, 2);
		lv_style_set_text_align(&mbox_style, LV_TEXT_ALIGN_CENTER);
		for( uint8_t i = 0; i < NO_MSG_SLOTS; i++ ) {

			mbox[i] = lv_textarea_create(lv_scr_act());

			lv_textarea_set_one_line(mbox[i], true);
			lv_obj_add_style(mbox[i], &mbox_style, LV_PART_MAIN);
			lv_obj_set_width( mbox[i], MSG_B_SIZEX);
			lv_obj_set_pos(mbox[i], SDL_HOR_RES, (105 + (MSG_B_SIZEY * i)) );
			lv_anim_init(&a1[i]);
			lv_anim_set_var(&a1[i], mbox[i]);
			lv_anim_set_values(&a1[i], SDL_HOR_RES, (SDL_HOR_RES - MSG_B_SIZEX - 10));
			lv_anim_set_time(&a1[i], MSG_ANIM_TIME);
			lv_anim_set_exec_cb(&a1[i], (lv_anim_exec_xcb_t)anim_x_cb);
			lv_anim_set_path_cb(&a1[i], lv_anim_path_ease_in);
			lv_anim_init(&a2[i]);
			lv_anim_set_var(&a2[i], mbox[i]);
			lv_anim_set_values(&a2[i], (SDL_HOR_RES - MSG_B_SIZEX - 10), SDL_HOR_RES);
			lv_anim_set_time(&a2[i], MSG_ANIM_TIME);
			lv_anim_set_exec_cb(&a2[i], (lv_anim_exec_xcb_t) anim_x_cb1);
			lv_anim_set_path_cb(&a2[i], lv_anim_path_ease_in);

			anim_timeline[i] = lv_anim_timeline_create();
			lv_anim_timeline_add(anim_timeline[i], 0, &a1[i]);
			lv_anim_timeline_add(anim_timeline[i], MSG_DISP_TIME, &a2[i]);

		}
		first_call = false;
	}
	lv_textarea_set_text(mbox[msg_pos], sys_msgs[message_id]);
	lv_obj_move_foreground( mbox[msg_pos] );
	lv_anim_timeline_start(anim_timeline[msg_pos]);
	if( msg_pos++ == NO_MSG_SLOTS ) msg_pos = 0;

}

show_sys_message( 0 );
```
Prior to todays updates I have never had to use LV_COLOR_SCREEN_TRANSP 1 in lv_conf.h maybe this is an ongoing mistake on my part but either way I can't seem to get things to work correctly since this update.  @kisvegabor if you have any suggestions as to where I should be looking for a solution to this I would very much appreciate some pointers.

Kind Regards,

Pete

kisvegabor, 2022-04-26:
Huh, happy to hear that it works. :slightly_smiling_face: 
I was excited that how this large update will work for others.

Have you noticed any regression in performance?

pete-pjb, 2022-04-26:
No noticeable performance change on my platform (Zynq running WXGA+ 1440 X 900)😃
Thanks again.👍

Kind Regards,

Pete.