lvgl/lvgl #3301: spinbox: cannot set widget position in MicroPython

ting-liu, 2022-04-27:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
In MicroPython, 'lv.spinbox.set_pos' is mapped to C function 'lv_spinbox_set_pos', instead of 'lv_obj_set_pos'. Should C function 'lv_spinbox_set_pos' be renamed, or any other more graceful way to avoid the name conflict?

### To Reproduce
Running the spinbox example on LVGL simulator:
https://docs.lvgl.io/master/examples.html#spinbox
add oneline code to set spinbox position.
```python
spinbox = lv.spinbox(lv.scr_act())
spinbox.set_pos(145, 233)
```
-->
there is error "TypeError: argument num/types mismatch"

### Expected behavior
setting spinbox position works normally

### Screenshots or video

kisvegabor, 2022-05-01:
I cc @amirgon, the author of the MicroPython binding. I'm sure he know what could be the problem.  

FYI, it works:
```c
>>> spinbox.set_x(30)
>>> spinbox.set_y(30)
```

kisvegabor, 2022-05-01:
At first read I didn't understand the real issue. Got it now :slightly_smiling_face: 

Yes,  `lv_spinbox_set_pos` should be renamed, but it'd be an API breaking change so we can do it only in v9, or accept this minor API break in the next minor version. @embeddedt?

As a new name `lv_spinbox_set_cursor_pos` could be used. At least it's in line with `lv_textarea_set_cursor_pos`.

amirgon, 2022-05-01:
Renaming is definitely the right solution.  
As a temporary workaround, you can cast spinbox to obj and then call set_pos:
```python
lv.obj.__cast__(spinbox).set_pos(145, 233)
```

embeddedt, 2022-05-02:
This rename could be handled instantly by a `sed` command so I think it's fine to do in a minor release (with appropriate documentation, of course).

Although, thinking about it some more, wouldn't it be possible to `#define lv_spinbox_set_pos lv_spinbox_set_cursor_pos`. and completely avoid incompatibility at the source level? The MicroPython binding ignores preprocessor macros, correct?

amirgon, 2022-05-02:
> The MicroPython binding ignores preprocessor macros, correct?

The Micropython binding uses the preprocessed LVGL sources so it's affected by the preprocessor. But it does not provide an API for macros because they are already gone on preprocessed sources.

kisvegabor, 2022-05-04:
I fixed it as @embeddedt suggested.

commit a99eb6bb (2022-05-04): fix(spinbox): rename lv_spinbox_set_pos to lv_spinbox_set_cursor_pos

fixes #3301
closed 2022-05-04 by kisvegabor