lvgl/lvgl #3302: Display rotation 90 needs to provide something like "rounder_cb" to support more screens

alec-2018, 2022-04-27:
### Perform all steps below and tick them with [x]
- [ ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

hi，I've come across screens that can be oddly refreshed in a way that requires odd rows to start and even rows to end and odd columns to start and even columns to end，and the screen cannot be hardware rotated.so, use sw rotate , and LV_DISP_ROT_MAX_BUF is not the frame buffer size, it is possible to call flush_cb multiple times, and the calculated area will not meet the conditions of odd-numbered start and even-numbered ends.Therefore, it is necessary to add a method similar to "rounder_cb" to modify the area of "flush_cb" each time
### To Reproduce

### Expected behavior

### Screenshots or video

kisvegabor, 2022-05-04:
Oh, that's unfortunate. But this even-odd issue is related to a specific display controller chip, right? Can't you use that chip to rotate the display?

alec-2018, 2022-05-05:
> Oh, that's unfortunate. But this even-odd issue is related to a specific display controller chip, right? 
yes, it  is related to a specific display controller chip.

>Can't you use that chip to rotate the display?
Incredible, this chip can't rotate the display, but it's real. So it needs lvgl to do the rotation.
In response to this problem, when rotating 0° or 180°, lvgl only calls "flush_cb" once, and "rounder_cb" can handle it, but when rotating 90° or 270°, lvgl calls "flush_cb" multiple times, so that "rounder_cb" does will be invalid
The following is the specific implementation of my "rounder_cb"
```
static void disp_rounder(struct _lv_disp_drv_t * disp_drv, lv_area_t * area)
{
	if (area->x1 % 2 != 0) {
		area->x1 -= 1;
	}

	if (area->y1 % 2 != 0) {
		area->y1 -= 1;
	}

	if (area->x2 % 2 == 0) {
		area->x2 += 1;
	}

	if (area->y2 % 2 == 0) {
		area->y2 += 1;
	}
}
```

alec-2018, 2022-05-06:
Maybe this controller chip is really strange. I close this issue

closed 2022-05-06 by alec-2018