lvgl/lvgl #3304: LVGL + 2DGPU isn't very efficient with large amount of small mask blending

AfflatusX, 2022-04-28:
### Introduce the problem
I recently tried to add a 2DGPU support to LVGL 8.2. (btw, the 8.2 refactoring is amazing, thanks fo the work!) The GPU pass is fully working now - I can redirect all the blending calls to GPU with no issue and tested with the music player demo, everything looks great!

However, when digging deeper by adding some tracing to the fill_normal routine, I noticed it gets called quite often. I created a very simple demo which rotates a line every second (similar to a clock's second hand) and measured the call counts. For every rotation, it does a couple of hundreds of small blending.

This is fine with sw_renderer, but when using GPU there is a cost to talk to the GPU, and they add up quickly. The latency would negate the benefit of using the GPU to actually do the blending.

My current thinking is to batch these blending ops and execute them at the end of the frame. However, these mask objects seems to be created/destroyed, so they are no longer valid after the blending call. There are workarounds to that, but I want to see if I'm on the right path or there are better ways to leverage GPU more efficiently.

### Examples and cases
see above description. 

### Suggested solution
Considering support GPU batching natively during the draw pipeline.

kisvegabor, 2022-04-28:
Hi,

Thanks for the feedback! 

Have you rotated a `line` widget or an image?

AfflatusX, 2022-04-28:
A line widget, I also made sure it's properly positioned so that it only occupies the region it meant to.

for example - a line from (50,50) to (100,100) actually occupies (0,0) to (100,100), i positioned it on 50,50 and translate the point_2 to (50,50).

kisvegabor, 2022-04-28:
I see. The line is really drawn in smaller parts to keep the memory usage low. Now the buffer size is "1 display horizontal resolution pixels". You can change [this line](https://github.com/lvgl/lvgl/blob/d8622f544fe8884f7336198a97bd47fe63597a52/src/draw/sw/lv_draw_sw_line.c#L379) to:
```c
size_t mask_buf_size = LV_MIN(lv_area_get_size(&blend_area), 5 * hor_res);
```

Does it make any difference?

AfflatusX, 2022-04-28:
Ah yes, it does get lower (in my app it drops from 590-ish to 280 in best case), although it looks like it depends on the angle of the line.

AfflatusX, 2022-04-28:
I think with this and some batching the perf can get much better, let me try a couple of things and report back.

Looks like image/letter has similar logic - that probably explains why when I added some text and animating on that the perf drops.

AfflatusX, 2022-04-28:
@kisvegabor yeah with batching and the mask buffer size increase I'm hitting 60FPS again. I'm closing the issue, thanks for the pointer!

closed 2022-04-28 by AfflatusX

kisvegabor, 2022-05-04:
Thanks for the feedback, I'll consider making these buffer sizes adjustable.