lvgl/lvgl #3305: Add automatic/manual options for invalidate area

FASTSHIFT, 2022-04-29:
### Introduce the problem
The current lvgl will automatically mark a redraw area every time the widget attribute is modified. Some redraw area calculations may take a long time and may be repeated (such as `lv_img_set_zoom/angle`), resulting in a certain performance impact.

### Examples and cases
For example, there is an icon interface that almost fills the full screen, and I want to re-zoom every time I move all the icons. I tried two ways:
1. Use lvgl's native interface `lv_img_set_zoom` to traverse each icon to set the zoom value
2. Write the zoom value directly into `img->zoom`, and then mark the entire screen as a redraw area.

Measuring the time-consuming of both methods, it is found that the second method can reduce the execution time by about 5ms.

### Suggested solution
Is it possible to give the user an interface for setting automatic and manual redraw area calculations to reduce repeated redraw area calculations.
I suggest adding a `LV_OBJ_FLAG_MANUAL_INVALID` flag, which is checked every time the widget attribute is set. If the user sets it to manual mode, lvgl does not automatically redraw the area calculation.

kisvegabor, 2022-04-29:
I see the problem. What about handling it globally. E.g. temporarily disable adding invalidated areas to display.

Something like:
```c
lv_disp_enable_area_invalidation(disp, true/false);
```

FASTSHIFT, 2022-04-29:
It is a good idea, but some functions may have to add a `lv_disp_get_enable_area_invalidation()` judgment to bypass the redraw area calculation, such as:
```c
void lv_img_set_zoom(lv_obj_t * obj, uint16_t zoom)
{
    lv_img_t * img = (lv_img_t *)obj;
    if(zoom == img->zoom) return;

    if(zoom == 0) zoom = 1;

    if(!lv_disp_get_enable_area_invalidation(lv_disp_get_default())) {
        img->zoom = zoom;
        return;
    }

    lv_obj_update_layout(obj);  /*Be sure the object's size is calculated*/
    lv_coord_t w = lv_obj_get_width(obj);
    lv_coord_t h = lv_obj_get_height(obj);
    lv_area_t a;
    _lv_img_buf_get_transformed_area(&a, w, h, img->angle, img->zoom >> 8, &img->pivot);
    a.x1 += obj->coords.x1 - 1;
    a.y1 += obj->coords.y1 - 1;
    a.x2 += obj->coords.x1 + 1;
    a.y2 += obj->coords.y1 + 1;
    lv_obj_invalidate_area(obj, &a);

    img->zoom = zoom;
    lv_obj_refresh_ext_draw_size(obj);

    _lv_img_buf_get_transformed_area(&a, w, h, img->angle, img->zoom, &img->pivot);
    a.x1 += obj->coords.x1 - 1;
    a.y1 += obj->coords.y1 - 1;
    a.x2 += obj->coords.x1 + 1;
    a.y2 += obj->coords.y1 + 1;
    lv_obj_invalidate_area(obj, &a);
}
```

kisvegabor, 2022-04-29:
Good idea! 

A few days ago I noticed that actually all these extra calculations in `lv_img_set_zoom/angle/pivot` are useless because `lv_obj_refresh_ext_draw_size` sets a draw area which involves the transformed image in all directions and also invalidates that area. For example [here](https://docs.lvgl.io/master/widgets/core/img.html#rotate-and-zoom) a very large area is invalidated regardless the of the transform functions try to invalidate only the transformed area. You can see it if `LV_USE_REFR_DEBUG` is enabled. 

The solution could be to disable invalidation for `lv_obj_refresh_ext_draw_size` to invalidate only the tight transformed areas. But this way there can be nested invalidation disable/enable calls. So we should count the enable/disable calls like:
```c
void lv_disp_enable_area_invalidation(bool en)
{
   //disp->inv_en_cnt = 1 by default 
   disp->inv_en_cnt += en ? 1: -1; 
}

...

//Invalidate only if disp->inv_en_cnt > 0

```

For the function names these might be better: `lv_disp_enable_invalidation` and `lv_disp_is_invalidation_enabled`. What do you think?

FASTSHIFT, 2022-05-04:
@kisvegabor 
I agree with this.

kisvegabor, 2022-05-07:
For `lv_obj_refresh_ext_draw_size` the invalidation should be disabled.

kisvegabor, 2022-05-23:
Hi,
Are there any updates on this?

kisvegabor, 2022-05-28:
The PR is merged so I close this issue. Thank you @FASTSHIFT

closed 2022-05-28 by kisvegabor