lvgl/lvgl #3321: benchmark will crash on v8.1 or newer

chenyinchang, 2022-05-04:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
I trasnplant v8.1 and v8.2 to esp32s3. When I run benchmark example. But Especially  the benchmark run to "static void arc_thick_cb(void)", then the device will reboot.Sometime once reboot and go to the end, but sometime it will reboot forever.

### To Reproduce

when I redefine "ARC_WIDTH_THICK" to 13 on benchmark.c , the app will crash. If I change number to  <= 10, will normall

### Expected behavior

### Screenshots or video

the crash photo over here. I find the "lv_draw_mask_free_param" on lv_draw_mask.c  which be added  on v8.1 and newer, I guess something wrong on this.
[image]

kisvegabor, 2022-05-04:
It seems working for me in the simulator. Can you reproduce it on PC?

Doe it change something if you increase `LV_MEM_SIZE`?

chenyinchang, 2022-05-04:
> It seems working for me in the simulator. Can you reproduce it on PC?
> 
> Doe it change something if you increase `LV_MEM_SIZE`?

Thank you for your help. I use [lv_port_esp32](https://github.com/lvgl/lv_port_esp32) as template to transplant v8.1 and v8.2. The template default set `LV_MEM_SIZE` is 32. I have try change to `64 `which seem didn't work. And then I change to `1024`, the crash change place below photo. (I run the whole guiTask on FreeRTOS and give the task ram is `4096*2`, in [lv_port_esp32](https://github.com/lvgl/lv_port_esp32) /main/main.c line in 67 which you will see too)
[image]

nicusorcitu, 2022-05-04:
maybe the same root cause as #3324 ?

chenyinchang, 2022-05-04:
> maybe the same root cause as #3324 ?

It maybe look like same.Can you describe sth wrong when you run benchmark? When crash?
Can you try run benchmark again when you block `static void sub_arc_cb(void)` and  `static void arc_thick_cb(void)` in `lv_demo_benchmark.c` like below photo?
[image]
[image]

nicusorcitu, 2022-05-04:
I do not get an error while running the benchmark and I do not mean to reproduce exactly your scenario.
What I mean is that maybe there is the same rootcause on the allocator while trying to realocate a size that does not fit on the memory reserved. And it will lead to writing to an invalid block.

To detect that I suggest to enable the traces:

#define LV_LOG_TRACE_MEM        1
#define LV_LOG_LEVEL LV_LOG_LEVEL_TRACE

kisvegabor, 2022-05-04:
The log says:
```

[Trace]	(765.535, +7)	 lv_obj_clean: finished (delete 0x3fc9c444) 	(in lv_obj_tree.c line #105)
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.

Core  1 register dump:
PC      : 0x420079ac  PS      : 0x00060330  A0      : 0x82015510  A1      : 0x3fce2430  
0x420079ac: scene_next_task_cb at /home/admin123/esp32s3_example/lv_port_esp32-master/build/../components/lvgl/demos/benchmark/lv_demo_benchmark.c:737
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x000006c0  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x0000069c  
A10     : 0x3fc9c444  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x00060d23  
A14     : 0x00060023  A15     : 0x00000001  SAR     : 0x00000004  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffc  

Backtrace:0x420079a9:0x3fce24300x4201550d:0x3fce2550 0x420155cb:0x3fce2570 0x420065fe:0x3fce2590 0x40381d59:0x3fce25d0 
0x420079a9: scene_next_task_cb at /home/admin123/esp32s3_example/lv_port_esp32-master/build/../components/lvgl/demos/benchmark/lv_demo_benchmark.c:737
```

It can happen that there is a diviision by zero in some special cases but [line 737](https://github.com/lvgl/lvgl/blob/master/demos/benchmark/lv_demo_benchmark.c#L737) is a comment. Can you localize the issue more precisely? If it was enabled, disabling optimization can help.

chenyinchang, 2022-05-06:
> The log says:
>
> ```
>
> [Trace]	(765.535, +7)	 lv_obj_clean: finished (delete 0x3fc9c444) 	(in lv_obj_tree.c line #105)
> Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was [...]

OK, I will try again later.And I think I may find out crash reason which is I don't define LV_HOR_RES and LV_VER_RES and only define LV_VER_RES_MAX and LV_HOR_MAX(Because when I define two parameters, the crash never happen until now).Except the two parameters, I init display HOR and VER is set right value. I check the code, if user didn't define two parameter, it will get default value. Maybe get default value go wrong?

kisvegabor, 2022-05-09:
> And I think I may find out crash reason which is I don't define LV_HOR_RES and LV_VER_RES and only define LV_VER_RES_MAX and LV_HOR_RES_MAX(

`LV_HOR_RES` and `LV_VER_RES ` uses the display driver's config [by default](https://github.com/lvgl/lvgl/blob/master/src/core/lv_disp.h#L205-L217). And  `LV_VER_RES_MAX` and `LV_HOR_RES_MAX` are not used in v8.

`disp_drv.hor_res` and `disp_drv.ver_res` should have a sane [default value](https://github.com/lvgl/lvgl/blob/master/src/hal/lv_hal_disp.c#L85-L86). So even is you don't set them it should work from LVGL's point of view.

closed 2022-06-01 by lvgl-bot

brightproject, 2024-11-27:
> And I think I may find out crash reason which is I don't define LV_HOR_RES and LV_VER_RES and only define LV_VER_RES_MAX and LV_HOR_RES_MAX(

Good day @kisvegabor 
I apologize for posting a question here, but I have already posted 2 questions on the [forum](https://forum.lvgl.io/) and have not received answers to them.
I also understand that you are not obliged to give answers to everyone and to the first request.
So, sorry.
I have a similar problem, although the advice that I read here in the forum threads did not help me.

I draw a sector using an arch, which will ultimately act as an arrow.

[image]

```
// speed scale
 ui_speedline = lv_img_create(ui_Screen0);
 lv_img_set_src(ui_speedline, &ui_img_speedscale360_png);
 lv_obj_set_width(ui_speedline, 180);
 lv_obj_set_height(ui_speedline, 360);
 lv_obj_set_x(ui_speedline, -91); // left(-), right(+)
 lv_obj_set_y(ui_speedline, 0); // down(+), up(-)
 lv_obj_set_align(ui_speedline, LV_ALIGN_CENTER);
 lv_obj_add_flag(ui_speedline, LV_OBJ_FLAG_ADV_HITTEST); /// Flags
 lv_obj_clear_flag(ui_speedline, LV_OBJ_FLAG_SCROLLABLE); /// Flags
 lv_img_set_pivot(ui_speedline, 100, 0);
 lv_obj_set_style_bg_color(ui_speedline, lv_color_hex(0xFFFFFF), LV_PART_MAIN | LV_STATE_DEFAULT);
 lv_obj_set_style_bg_opa(ui_speedline, 255, LV_PART_MAIN | LV_STATE_DEFAULT);

 // speed arrow in the form of a sector
 ui_SpeedARC = lv_arc_create(ui_Screen0);
 lv_obj_set_width(ui_SpeedARC, 340);
 lv_obj_set_height(ui_SpeedARC, 340);
 lv_obj_set_align(ui_SpeedARC, LV_ALIGN_CENTER);
 lv_arc_set_range(ui_SpeedARC, 0, 0);
 lv_obj_add_flag(ui_SpeedARC, LV_OBJ_FLAG_ADV_HITTEST); /// Flags
 // lv_arc_set_value(ui_SpeedARC, 0);
 // lv_arc_set_bg_angles(ui_SpeedARC, 153, 215);
 // set the color of the arch here
// lv_obj_set_style_arc_color(ui_SpeedARC, lv_color_hex(0x00FF00), LV_PART_INDICATOR | LV_STATE_DEFAULT); // green
lv_obj_set_style_arc_color(ui_SpeedARC, lv_color_hex(0xFF0000), LV_PART_INDICATOR | LV_STATE_DEFAULT); // red
lv_obj_set_style_arc_opa(ui_SpeedARC, 255, LV_PART_INDICATOR | LV_STATE_DEFAULT);
lv_obj_set_style_bg_color(ui_SpeedARC, lv_color_hex(0x000000), LV_PART_INDICATOR | LV_STATE_DEFAULT);
lv_obj_set_style_bg_opa(ui_SpeedARC, 0, LV_PART_INDICATOR | LV_STATE_DEFAULT);
// make the knob (square) color at the beginning and end of the slider transparent
lv_obj_set_style_bg_color(ui_SpeedARC, lv_color_hex(0xFFFFFF), LV_PART_KNOB | LV_STATE_DEFAULT);
lv_obj_set_style_bg_opa(ui_SpeedARC, 0, LV_PART_KNOB | LV_STATE_DEFAULT);
// arrow length
lv_obj_set_style_arc_width(ui_SpeedARC, 60, LV_PART_INDICATOR | LV_STATE_DEFAULT);
```

When I swipe the screen with my finger, or hold my finger for a while, or sometimes just touch it - the `esp32` reboots.
The logs are as follows ... division by zero occurs:

> [Trace] (4.348, +1)      event_send_core: Sending event 21 to 0x3fc991f8 with 0x3fc98740 param  (in lv_event.c line #428)
> [Trace] (4.349, +1)      event_send_core: Sending event 26 to 0x3fc991f8 with 0x3fcebbf8 param  (in lv_event.c line #428)
> [Trace] (4.350, +1)      event_send_core: Sending event 27 to 0x3fc991f8 with 0x3fcebbf8 param  (in lv_event.c line #428)
> [Trace] (4.351, +1)      event_send_core: Sending event 22 to 0x3fc991f8 with 0x3fc98740 param  (in lv_event.c line #428)
> [Trace] (4.352, +1)      event_send_core: Sending event 23 to 0x3fc991f8 with 0x3fc98740 param  (in lv_event.c line #428)
> [Trace] (4.353, +1)      event_send_core: Sending event 24 to 0x3fc991f8 with 0x3fc98740 param  (in lv_event.c line #428)
> [Trace] (4.354, +1)      event_send_core: Sending event 25 to 0x3fc991f8 with 0x3fc98740 param  (in lv_event.c line #428)
> [Trace] (4.355, +1)      event_send_core: Sending event 20 to 0x3fc9923c with 0x3fc98740 param  (in lv_event.c line #428)
> [Trace] (4.356, +1)      event_send_core: Sending event 21 to 0x3fc9923c with 0x3fc98740 param  (in lv_event.c line #428)
> [Trace] (4.357, +1)      event_send_core: Sending event 26 to 0x3fc9923c with 0x3fcebbf8 param  (in lv_event.c line #428)
> [Trace] (4.358, +1)      event_send_core: Sending event 27 to 0x3fc9923c with 0x3fcebbf8 param  (in lv_event.c line #428)
> [Trace] (4.359, +1)      event_send_core: Sending event 22 to 0x3fc9923c with 0x3fc98740 param  (in lv_event.c line #428)
> [Trace] (4.360, +1)      event_send_core: Sending event 23 to 0x3fc9923c with 0x3fc98740 param  (in lv_event.c line #428)
> [Trace] (4.361, +1)      event_send_core: Sending event 24 to 0x3fc9923c with 0x3fc98740 param  (in lv_event.c line #428)
> [Trace] (4.362, +1)      event_send_core: Sending event 25 to 0x3fc9923c with 0x3fc98740 param  (in lv_event.c line #428)
> [Trace] (4.363, +1)      call_flush_cb: Calling flush_cb on (115;96)(243;261) area with 0x3fcb2304 image pointer        (in lv_refr.c line #1317)
> [Trace] (4.364, +1)      lv_mem_free: freeing 0x3fc9b30c        (in lv_mem.c line #171)
> [Trace] (4.366, +2)      lv_mem_free: freeing 0x3fc9a490        (in lv_mem.c line #171)
[... 35 lines cut ...]
>       (inlined by) lv_indev_read_timer_cb at lib/lvgl/src/core/lv_indev.c:101
>   #6  0x42010453 in lv_timer_exec at lib/lvgl/src/misc/lv_timer.c:313
>       (inlined by) lv_timer_handler at lib/lvgl/src/misc/lv_timer.c:109
>   #7  0x42002cd8 in loop() at src/main.cpp:166
>   #8  0x4202fc98 in loopTask(void*) at .platformio/packages/framework-arduinoespressif32@src-98cb8087edcca5875bec152528484d34/cores/esp32/main.cpp:74
[... cut ...]
> Build:Mar 27 2021
> rst:0xc (RTC_SW_CPU_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
> Saved PC:0x4037a266
>   #0  0x4037a266 in esp_cpu_wait_for_intr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/cpu.c:121
> 
> SPIWP:0xee
> mode:DIO, clock div:1
> load:0x3fce3818,len:0x109c
> load:0x403c9700,len:0x4
> load:0x403c9704,len:0xb50
> load:0x403cc700,len:0x2fe4
> entry 0x403c98ac
> Hello Arduino! V8.3.11
> I am LVGL_Arduino

I'd be happy to chat in the discussion🙂

kisvegabor, 2024-12-18:
Hi,

I'm sorry for the super late reply. Please open a new issue here so that it can receive more views. 

Can you try the same in a simulator with a Debugger? Debuggers usually can stop on division by zero errors, so that you can exactly see what's happening. 

Updating LVGL to `master` also might help, as this issue might be already fixed.

brightproject, 2024-12-18:
Thanks for the response, I don't understand `visual studio` well enough, so I have some difficulties with launching the `simulator`.
I also tried to launch the simulator through `codespace`, but there are errors there too.
Therefore, I compile the code through `vscode` + `platformio` and flash the device and look at the errors on it.

> https://forum.lvgl.io/t/setup-lvgl-in-a-simulator-with-4-clicks-in-2-minutes/11317/17?u=brightproject

The device rebooted due to incorrect slider arch dimensions.
Everything is fine now🙂