lvgl/lvgl #3323: Wrong drawing result when LV_COLOR_DEPTH=16 and LV_COLOR_SCREEN_TRANSP=1

FASTSHIFT, 2022-05-04:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
I want to use the style_opa property to set the transparency of the widget, but some drawing errors occur.
This is my configuration:
```c
#define LV_COLOR_DEPTH 16
#define LV_COLOR_SCREEN_TRANSP 1
#define LV_USE_THEME_DEFAULT 0
```

### To Reproduce
```c
void style_opa_test(void)
{
  lv_obj_t* obj = lv_obj_create(lv_scr_act());
  lv_obj_remove_style_all(obj);
  lv_obj_set_style_bg_color(obj, lv_palette_main(LV_PALETTE_BLUE), 0);
  lv_obj_set_style_bg_opa(obj, LV_OPA_COVER, 0);
  lv_obj_center(obj);

  lv_obj_t* label = lv_label_create(obj);
  lv_obj_set_style_text_color(label, lv_color_white(), 0);
  lv_obj_set_style_text_font(label, LV_FONT_DEFAULT, 0);
  lv_obj_center(label);

  lv_obj_fade_out(obj, 1000, 1000);
}
```

### Expected behavior
I switched to this version before https://github.com/lvgl/lvgl/commit/318146a2c25362eabf258470be263a4cfeaefe87 and it can be drawn normally.

https://user-images.githubusercontent.com/26767803/166660013-1b100d0f-845d-46fa-b3c2-e70e26663e7f.mp4

### Screenshots or video

https://user-images.githubusercontent.com/26767803/166659209-28e8cf31-4e29-413d-8ea2-55165fd5d3d5.mp4

kisvegabor, 2022-05-04:
Hi, 
Do you use the latest version from maser?

FASTSHIFT, 2022-05-04:
> Hi, Do you use the latest version from maser?

Yes, I can be sure.
Problem disappears when setting `LV_USE_THEME_DEFAULT=1`.

kisvegabor, 2022-05-04:
Okay, I'm looking into it.

kisvegabor, 2022-05-04:
I found the problem: it's because no theme are loaded and by default the screen is transparent.

You can either do this:
```c
    lv_obj_set_style_bg_opa(lv_scr_act(), LV_OPA_COVER, 0);
    lv_obj_set_style_bg_color(lv_scr_act(), lv_color_black(), 0);
```

or use `LV_THEME_BASIC` which set some sane defaults but nothing fancy.

FASTSHIFT, 2022-05-04:
@kisvegabor 
Thanks for your answer :)

closed 2022-05-04 by FASTSHIFT

kisvegabor, 2022-05-04:
You are very welcome! :slightly_smiling_face: