lvgl/lvgl #3328: bug with lv_chart widget? Y- and X- axis

epikao, 2022-05-05:
Hello

I am not sure if this is a bug, I have already mentioned this in the How-To forum however without satisfactory answer.
I am using a STM32F7xx board with LVGL version 8.3.

Basically I have two problems.

In the **first problem**, when using the following code, "000" is displayed instead of "1000".
`lv_chart_set_range(ui.screen_chart_1,LV_CHART_AXIS_PRIMARY_Y, 0, 1000);`

But when using 1001 it is displayed correctly, see picture below.

[image]

Then the **second problem**:
If set tick X-axis as following code, the function `lv_chart_set_value_by_id()` is more or less fast, but if set axis_tick to second following code, the function takes many processor load and around 30ms... if use the function lv_chart_set_next_value() instead of the id function it freezes:
1:
`lv_chart_set_axis_tick(ui.screen_chart_1, LV_CHART_AXIS_PRIMARY_X, 10, 5, 17, 1, true, 30);`
2:
lv_chart_set_axis_tick(ui.screen_chart_1, LV_CHART_AXIS_PRIMARY_X, 10, 5, 900, 1, true, 30);

( I use 900 points: `lv_chart_set_point_count(ui.screen_chart_1, 900);` )

From my understanding he ticks on the X-axis have to be set only once (at initialization), why does this have an influence on the graph set function and therefore to the processor load?

pete-pjb, 2022-05-05:
Hi @epikao,

I can help with your first problem it looks like you should increase the value of draw_size on your call to ```lv_chart_set_axis_tick()``` I would try increasing the value to 50 initially to see if there is now room to display "1000"

Hope that helps!

Kind Regards,

Pete

epikao, 2022-05-05:
> I would try increasing the value to 50 initially to see if there is now room to display "1000"

thank you, but why should it be related to the "draw_size" ? because for "1001" it has enough room?

```
	lv_obj_set_pos(ui->screen_chart_1, 60, 300);
	lv_obj_set_size(ui->screen_chart_1, 950, 200);
```

I changed "60" to "80", without change...

epikao, 2022-05-05:
> I can help with your first problem it looks like you should increase the value of draw_size on your call to
thank you, that worked (for problem 1) :-) ...

epikao, 2022-05-06:
> Then the **second problem**:
Is my second problem explained understandably?

epikao, 2022-05-18:
> I'd be interested in check the performance issue, but please add a codesnippet that I can copy/paste to reproduce the problem.

I have noticed that the following function also takes a relatively long time. The chart function together with the meter function makes then partly already more than 15ms... but only if a changing value is passed. 
How many ms would be so usual? With a STM32F7xx ? Only so approximately?

`lv_meter_set_indicator_value(ui.screen_meter_1, screen_meter_1_scale_1_ndimg_0, value);`

Thank you

kisvegabor, 2022-05-19:
What is the indicator type? (arc, needle line, needle image, scale lines?)

Invalidating an arc takes the longest time but on an STM32F7 it shouldn't be noticeable.

epikao, 2022-05-19:
> I'd be interested in check the performance issue, but please add a codesnippet that I can copy/paste to reproduce the problem.

I will first try to measure the elapsed time of every six functions listed below with the oscilloscope...
Would it be normal if these functions (at least the ones with lv_rand) move in the ms range on a STM32F7xx?

```
    while (1)
    {
        //HAL_Delay(500);
    	
    	//if( ( millis()-lastMillis1 ) > 50){
    		
    		HAL_GPIO_TogglePin(GPIOI, GPIO_PIN_8); //LED 101
    		
    		lv_meter_set_indicator_value(ui.screen_meter_1, screen_meter_1_scale_1_ndimg_0, lv_rand(0,1000) );
    		lv_meter_set_indicator_value(ui.screen_meter_1, screen_meter_1_scale_1_ndimg_0, 500 );
    		lv_arc_set_value(ui.screen_arc_1, lv_rand(0,1000));
    		lv_arc_set_value(ui.screen_arc_1, 500);
    		lv_chart_set_value_by_id(ui.screen_chart_1, screen_chart_1_0, countChart, lv_rand(0,1000));
    		lv_chart_set_value_by_id(ui.screen_chart_1, screen_chart_1_0, countChart, 500);
    		
    		//lastMillis1 = millis();
    	//}
}
```

epikao, 2022-05-20:
I have now checked all the above functions individually and measured how the LED toggles (see attached images).

First I commented out all functions, the LED toggles more or less regularly every 2ms.

lv_meter (random value) has the longest interruptions, partly up to 30ms!
lv_meter (fixed value) has partly 10ms interruptions
lv_arc (random value) has partly 10ms interruptions
lv_arc (fixed value) is OK/regular,
lv_chart (random value) is OK/regular
lv_chart (fixed value) is OK/regular

So different from assumed is not lv_chart my problem, but lv_meter and lv_arc.... 
Are these long interruptions normal?

Thanks a lot

**NO function:**
[image]

**lv_meter (random):**
[image]

**lv_meter (fix):**
[image]

**lv_arc (random):**
[image]

**lv_arc (fix):**
[image]

**lv_chart (random):**
[image]

**lv_chart (fix):**
[image]

kisvegabor, 2022-05-24:
Thanks for the detailed measurements. 

Please send the exact code of the meter and arc creation and settings so that I can reproduce it.

epikao, 2022-05-24:
> Please send the exact code of the meter and arc creation and settings so that I can reproduce it.

**Here my arc code:**

```
	//Write codes screen_arc_1
	ui->screen_arc_1 = lv_arc_create(ui->screen);
	lv_obj_set_pos(ui->screen_arc_1, 180, -5);
	lv_obj_set_size(ui->screen_arc_1, 414, 540);

	//Write style state: LV_STATE_DEFAULT for style_screen_arc_1_main_main_default
	static lv_style_t style_screen_arc_1_main_main_default;
	lv_style_reset(&style_screen_arc_1_main_main_default);
	lv_style_set_bg_color(&style_screen_arc_1_main_main_default, lv_color_make(0xf6, 0xf6, 0xf6));
	lv_style_set_bg_grad_color(&style_screen_arc_1_main_main_default, lv_color_make(0xf6, 0xf6, 0xf6));
	lv_style_set_bg_grad_dir(&style_screen_arc_1_main_main_default, LV_GRAD_DIR_VER);
	lv_style_set_bg_opa(&style_screen_arc_1_main_main_default, 0);
	//lv_style_set_border_width(&style_screen_arc_1_main_main_default, 2);
	//lv_style_set_border_color(&style_screen_arc_1_main_main_default, lv_color_make(0xff, 0x00, 0x00));
	lv_style_set_arc_color(&style_screen_arc_1_main_main_default, lv_color_make(0x90, 0x00, 0x00));
	//lv_style_set_arc_width(&style_screen_arc_1_main_main_default, 12);
	lv_obj_add_style(ui->screen_arc_1, &style_screen_arc_1_main_main_default, LV_PART_MAIN|LV_STATE_DEFAULT);

	//Write style state: LV_STATE_DEFAULT for style_screen_arc_1_main_indicator_default
	static lv_style_t style_screen_arc_1_main_indicator_default;
	lv_style_reset(&style_screen_arc_1_main_indicator_default);
	lv_style_set_arc_color(&style_screen_arc_1_main_indicator_default, lv_color_make(0xff, 0x00, 0x00));
	lv_style_set_arc_width(&style_screen_arc_1_main_indicator_default, 12);
	lv_obj_add_style(ui->screen_arc_1, &style_screen_arc_1_main_indicator_default, LV_PART_INDICATOR|LV_STATE_DEFAULT);

	//Write style state: LV_STATE_DEFAULT for style_screen_arc_1_main_knob_default
	static lv_style_t style_screen_arc_1_main_knob_default;
	lv_style_reset(&style_screen_arc_1_main_knob_default);
	lv_style_set_bg_color(&style_screen_arc_1_main_knob_default, lv_color_make(0x21, 0x95, 0xf6));
	lv_style_set_bg_grad_color(&style_screen_arc_1_main_knob_default, lv_color_make(0x21, 0x95, 0xf6));
	lv_style_set_bg_grad_dir(&style_screen_arc_1_main_knob_default, LV_GRAD_DIR_VER);
	lv_style_set_bg_opa(&style_screen_arc_1_main_knob_default, 0);
	lv_style_set_pad_all(&style_screen_arc_1_main_knob_default, 5);
	lv_obj_add_style(ui->screen_arc_1, &style_screen_arc_1_main_knob_default, LV_PART_KNOB|LV_STATE_DEFAULT);

	//Write style state: LV_STATE_DISABLED for style_screen_arc_1_main_knob_disabled
	static lv_style_t style_screen_arc_1_main_knob_disabled;
	lv_style_reset(&style_screen_arc_1_main_knob_disabled);
	lv_style_set_bg_color(&style_screen_arc_1_main_knob_disabled, lv_color_make(0xff, 0x00, 0x00));
	lv_style_set_bg_grad_color(&style_screen_arc_1_main_knob_disabled, lv_color_make(0xff, 0x00, 0x00));
	lv_style_set_bg_grad_dir(&style_screen_arc_1_main_knob_disabled, LV_GRAD_DIR_VER);
	lv_style_set_bg_opa(&style_screen_arc_1_main_knob_disabled, 0);
	lv_style_set_pad_all(&style_screen_arc_1_main_knob_disabled, 5);
	lv_obj_add_style(ui->screen_arc_1, &style_screen_arc_1_main_knob_disabled, LV_PART_KNOB|LV_STATE_DISABLED);
	lv_arc_set_bg_angles(ui->screen_arc_1, 180, 360);
	lv_arc_set_angles(ui->screen_arc_1, 180, 270);
	lv_arc_set_rotation(ui->screen_arc_1, 0);
	lv_obj_set_style_pad_top(ui->screen_arc_1, 20, LV_STATE_DEFAULT);
	lv_obj_set_style_pad_bottom(ui->screen_arc_1, 20, LV_STATE_DEFAULT);
	lv_obj_set_style_pad_left(ui->screen_arc_1, 20, LV_STATE_DEFAULT);
	lv_obj_set_style_pad_right(ui->screen_arc_1, 20, LV_STATE_DEFAULT);
	lv_obj_set_style_arc_rounded(ui->screen_arc_1, 0,  LV_PART_INDICATOR|LV_STATE_DEFAULT);
	lv_obj_set_style_arc_rounded(ui->screen_arc_1, 0, LV_STATE_DEFAULT);
	lv_arc_set_range(ui->screen_arc_1, 0,5000);
	lv_arc_set_value(ui->screen_arc_1, 0);
	lv_obj_clear_flag(ui->screen_arc_1, LV_OBJ_FLAG_CLICKABLE);

    lv_obj_set_style_arc_width(ui->screen_arc_1, 30, LV_PART_MAIN); // Changes background arc width
    lv_obj_set_style_arc_width(ui->screen_arc_1, 30, LV_PART_INDICATOR); // Changes set part width
```

**Here my meter code:**

```
	//Write codes screen_meter_1
[... 23 lines cut ...]
    lv_obj_set_style_bg_opa(ui->screen_meter_1, LV_OPA_COVER, LV_PART_INDICATOR);
    lv_obj_set_style_bg_color(ui->screen_meter_1, lv_color_make(0xff, 0x00, 0x00), LV_PART_INDICATOR); //lv_palette_darken(LV_PALETTE_GREY, 4)
    lv_obj_set_style_outline_color(ui->screen_meter_1, lv_color_make(0xff, 0x00, 0x00), LV_PART_INDICATOR); //lv_color_white()
    lv_obj_set_style_outline_width(ui->screen_meter_1, 3, LV_PART_INDICATOR);
    lv_obj_set_style_border_color(ui->screen_meter_1, lv_color_hex(0xFF0000), LV_PART_MAIN); //change color circle
    lv_obj_set_style_border_width(ui->screen_meter_1, 0, LV_PART_MAIN); //remove circle

	lv_meter_scale_t *screen_meter_1_scale_1;
	screen_meter_1_scale_1 = lv_meter_add_scale(ui->screen_meter_1);

	//lv_meter_indicator_t *screen_meter_1_scale_1_ndimg_0;
	screen_meter_1_scale_1_ndimg_0 = lv_meter_add_needle_line(ui->screen_meter_1, screen_meter_1_scale_1, 3, lv_color_make(0xff, 0x00, 0x00), -30);
	//screen_meter_1_scale_1_ndimg_0 = lv_meter_add_needle_img(ui->screen_meter_1.screen_meter_1, screen_meter_1_scale_1, &_needle1_alpha_70x19, -4, 9);
	//screen_meter_1_scale_1_ndimg_0 = lv_meter_add_needle_img(ui->screen_meter_1.screen_meter_1, screen_meter_1_scale_1, &_punching_needle_alpha_120x40, 0, 20);
[... cut ...]
	lv_meter_set_scale_range(ui->screen_meter_1, screen_meter_1_scale_1, 0, 100, 180, 180);

	//add arc for screen_meter_1_scale_1
	lv_meter_indicator_t *screen_meter_1_scale_1_arc_1;
	screen_meter_1_scale_1_arc_1 = lv_meter_add_arc(ui->screen_meter_1, screen_meter_1_scale_1, 4, lv_color_make(0xff, 0xff, 0x00), 2);
	lv_meter_set_indicator_start_value(ui->screen_meter_1, screen_meter_1_scale_1_arc_1, 0);
	lv_meter_set_indicator_end_value(ui->screen_meter_1, screen_meter_1_scale_1_arc_1, 360);
	lv_meter_set_indicator_value(ui->screen_meter_1, screen_meter_1_scale_1_ndimg_0, 0 );
```

kisvegabor, 2022-05-24:
I could test it only on PC now, but in a profiler the function related to invalidation wasn't really visible. 

How fast the rendering is if a simple invalidation takes 30ms?

epikao, 2022-05-24:
> I could test it only on PC now, but in a profiler the function related to invalidation wasn't really visible.
> 
> How fast the rendering is if a simple invalidation takes 30ms?

I am not very familiar with LVGL. Which function is meant by "function related to invalidation"?
Or how can I check the rendering time?

Thank you

kisvegabor, 2022-05-25:
I've added these lines to the end of `lv_example_meter_1()`:
```
    volatile uint32_t t = lv_tick_get();
    uint32_t i;
    for(i = 0; i < 1000; i++) {
    	lv_meter_set_indicator_value(meter, indic, lv_rand(0, 100));
    }
    volatile uint32_t elaps = lv_tick_elaps(t);
    printf("%d ms\n", elaps);
```

and tested it on STM32F769 Discovery.

It printed `75 ms` (for 1000 invalidation). So 1 invalidation took 75 us.

epikao, 2022-05-25:
Thank you. I still have the problem with lv_meter/lv_arc. Recopy the whole lvgl-library did not help.
If I put the following Code in the while loop, I receive the times see screenshot below.

```
        for(i = 0; i < 1000; i++) {
        	lv_meter_set_indicator_value(guider_ui.screen_meter_1, screen_meter_1_scale_1_ndimg_0, lv_rand(0, 100));
        }
        volatile uint32_t elaps = lv_tick_elaps(t);
        //printf("%d ms\n", elaps);
    	char buf8[20];
    	lv_snprintf(buf8, sizeof(buf8), "%lu\n\r", elaps);
    	//lv_snprintf(buf8, sizeof(buf8), "%d ms\n", elaps);
        HAL_UART_Transmit(&huart3,buf8,sizeof(buf8),1);
```

[image]

kisvegabor, 2022-05-25:
There are several strange thing in the log:
- Why the new lines are not handled on every line?
- Why are there so large difference between the values?

It seems like your system is doing something else in the background too. If normal profiling doesn't work for any reason sometimes I do it manually. :slightly_smiling_face: That is pause the Debug 10 times to see what the MCU is doing. 

BTW, I'm using this project: https://github.com/lvgl/lv_port_stm32f769_disco

epikao, 2022-05-25:
Ok thank you, however I am at the end of my Latin...., it's too much...
the "elaps" variable always has the same value in the debugger, but the buffer variable changes... whereas the serial monitor also doesn't always show the same...see screenshot below.

I use the lv_port_stm32f746... all demos works without problem...

[image]

epikao, 2022-05-25:
> It printed `75 ms` (for 1000 invalidations). So 1 invalidation took 75 us.

I think the value of the variable "elaps" is converted or displayed incorrectly with my printf function... for whatever reason.
If I measure it, then the LED toggles regularly every 200ms, see scope screenshot below and related code:

```
    while (1)
    {
    	//HAL_UART_Transmit(&huart3,"loop\n\r",8,1);
        //HAL_Delay(500);
    	//HAL_GPIO_TogglePin(GPIOI, GPIO_PIN_8); //LED 101
    	////////////////// TESTCODE ///////////////////////////

        if( ( millis()-lastMillis2 ) > 4){
        	lv_task_handler();
        	lastMillis2 = millis();
        }

        //lv_timer_handler_run_in_period(4);
/*
    	if( ( millis()-lastMillis1 ) >= 1){

    		//HAL_GPIO_TogglePin(GPIOI, GPIO_PIN_8); //LED 101

    		//lv_meter_set_indicator_value(guider_ui.screen_meter_1, screen_meter_1_scale_1_ndimg_0, lv_rand(0,1000) );
    		//lv_meter_set_indicator_value(guider_ui.screen_meter_1, screen_meter_1_scale_1_ndimg_0, 500 );
    		//lv_arc_set_value(guider_ui.screen_arc_1, lv_rand(0,1000));
    		//lv_arc_set_value(guider_ui.screen_arc_1, 500);
    		//lv_chart_set_value_by_id(guider_ui.screen_chart_1, screen_chart_1_0, countChart, lv_rand(0,1000));
    		//lv_chart_set_value_by_id(guider_ui.screen_chart_1, screen_chart_1_0, countChart, 500);

    		//countChart++;

    		lastMillis1 = millis();
    	}else{
    		__NOP();
    	}
*/

        for(i = 0; i < 1000; i++) {
        	lv_meter_set_indicator_value(guider_ui.screen_meter_1, screen_meter_1_scale_1_ndimg_0, lv_rand(0, 100));
        }
        HAL_GPIO_TogglePin(GPIOI, GPIO_PIN_8); //LED 101
}
```

[image]
[image]

epikao, 2022-05-25:
I'm starting to think I have a conflict with this millis() function....
Can it be that this SysTick_Handler(void) is not necessarily called once every 1ms? (see my code below)

```
void SysTick_Handler(void)
{
  HAL_IncTick();
  HAL_SYSTICK_IRQHandler();
  lv_tick_inc(1);
  counter++;
}

uint32_t millis() {
  return counter;
}
```

pete-pjb, 2022-05-25:
Hi @epikao ,

Can you put your  ```HAL_GPIO_TogglePin(GPIOI, GPIO_PIN_8); //LED 101``` in the tick handler?
```c
 void SysTick_Handler(void)
{
  HAL_IncTick();
  HAL_SYSTICK_IRQHandler();
  lv_tick_inc(1);
  counter++;
  HAL_GPIO_TogglePin(GPIOI, GPIO_PIN_8); //LED 101
}
```
This should confirm your suspicions.

Kind Regards,

Pete

epikao, 2022-05-25:
> It should be safe to do so.
> This should confirm your suspicions

just tried, does not seem to be the cause (clean every 1ms, no interruption...). 

But why is it regular (every 200ms) in the "for loop" but not in the millis() .... crazy...

pete-pjb, 2022-05-25:
Can you use ```lv_tick_get(void)''' or HAL_GetTick() instead of millis()?
Also it might be worth checking your stack and heap configuration and that they aren't overflowing if possible. 

Just my thoughts...

epikao, 2022-05-25:
> where and how is the variable 'counter' declared?

in the same file "stm32f7xx_it.c"
```
volatile uint32_t counter=0; //for millisfunction

uint32_t millis() {
  return counter;
}
```

I measured now following codes (lv_tick_get or millis() , is the same).
1st Code always around 50ms interruptions... 2th Code around 200ms interruptions...

**1st Code**
```
    while (1)
    {

        if( ( lv_tick_get()-lastMillis2 ) > 4){
        	lv_task_handler();
        	lastMillis2 = lv_tick_get();
        }
    		HAL_GPIO_TogglePin(GPIOI, GPIO_PIN_8); //LED 101
    		lv_meter_set_indicator_value(guider_ui.screen_meter_1, screen_meter_1_scale_1_ndimg_0, lv_rand(0,1000) );
}
```

[image]

**2th-code**

```
    while (1)
    {

        if( ( lv_tick_get()-lastMillis2 ) > 4){
        	lv_task_handler();
        	lastMillis2 = lv_tick_get();
        }

       uint32_t i;
       for(i = 0; i < 1000; i++) {
        	lv_meter_set_indicator_value(guider_ui.screen_meter_1, screen_meter_1_scale_1_ndimg_0, lv_rand(0, 1000));
    		HAL_GPIO_TogglePin(GPIOI, GPIO_PIN_8); //LED 101
        }
}
```

[image]

pete-pjb, 2022-05-25:
Hi @epikao ,

Looking back at [this](https://github.com/lvgl/lvgl/issues/2956) post you disabled the internal LVGL memory management, if that is still the case for the current project and you are still using the linker file from the repository [here](https://github.com/lvgl/lv_port_stm32f769_disco) you will only have 512 bytes of heap memory allocated and I expect this will almost certainly be overflowing.  This as you are probably aware can cause all sorts of problems.

Can I suggest you confirm this please?   If you like, share your 'lv_conf.h' file and 'LinkerScript.ld' file from your project here so we can take a look.

If you have checked these things, no problem just ignore me!

Kind Regards,

Pete

epikao, 2022-05-25:
> Can I suggest you confirm this please? If you like, share your 'lv_conf.h' file and 'LinkerScript.ld' file from your project here so we can take a look.

Thank you very much. I appreciate your support very much.

Your mentioned other post is a different project, and with Arduino.
Now I'm using released LVGL 8.2, stm32IDE and lv_port_stm32f746_disco.

Below my conf-file and linkerscript.

**lv_conf.h:**
`
/**
 * @file lv_conf.h
 * Configuration file for v8.2
 */

/*
 * Copy this file as `lv_conf.h`
 * 1. simply next to the `lvgl` folder
 * 2. or any other places and
 *    - define `LV_CONF_INCLUDE_SIMPLE`
 *    - add the path as include path
 */

/* clang-format off */
#if 1 /*Set it to "1" to enable content*/

#ifndef LV_CONF_H
#define LV_CONF_H

#include <stdint.h>

/*====================
   COLOR SETTINGS
 *====================*/

/*Color depth: 1 (1 byte per pixel), 8 (RGB332), 16 (RGB565), 32 (ARGB8888)*/
#define LV_COLOR_DEPTH     16

/*Swap the 2 bytes of RGB565 color. Useful if the display has an 8-bit interface (e.g. SPI)*/
#define LV_COLOR_16_SWAP   0

/*Enable more complex drawing routines to manage screens transparency.
 *Can be used if the UI is above an other layer, e.g. an OSD menu or video player.
 *Requires `LV_COLOR_DEPTH = 32` colors and the screen's `bg_opa` should be set to non LV_OPA_COVER value*/
#define LV_COLOR_SCREEN_TRANSP    0

/* Adjust color mix functions rounding. GPUs might calculate color mix (blending) differently.
 * 0: round down, 64: round up from x.75, 128: round up from half, 192: round up from x.25, 254: round up */
#define LV_COLOR_MIX_ROUND_OFS (LV_COLOR_DEPTH == 32 ? 0: 128)

/*Images pixels with this color will not be drawn if they are  chroma keyed)*/
#define LV_COLOR_CHROMA_KEY    lv_color_hex(0x00ff00)         /*pure green*/

/*=========================
   MEMORY SETTINGS
 *=========================*/

/*1: use custom malloc/free, 0: use the built-in `lv_mem_alloc()` and `lv_mem_free()`*/
#define LV_MEM_CUSTOM      0
#if LV_MEM_CUSTOM == 0
/*Size of the memory available for `lv_mem_alloc()` in bytes (>= 2kB)*/
#  define LV_MEM_SIZE    (64U * 1024U)          /*[bytes]*/

/*Set an address for the memory pool instead of allocating it as a normal array. Can be in external SRAM too.*/
#  define LV_MEM_ADR          0     /*0: unused*/
    /*Instead of an address give a memory allocator that will be called to get a memory pool for LVGL. E.g. my_malloc*/
    #if LV_MEM_ADR == 0
        //#define LV_MEM_POOL_INCLUDE your_alloc_library  /* Uncomment if using an external allocator*/
        //#define LV_MEM_POOL_ALLOC   your_alloc          /* Uncomment if using an external allocator*/
[... 791 lines cut ...]
  /* User_heap_stack section, used to check that there is enough RAM left */
  ._user_heap_stack :
  {
    . = ALIGN(8);
    PROVIDE ( end = . );
    PROVIDE ( _end = . );
    . = . + _Min_Heap_Size;
    . = . + _Min_Stack_Size;
    . = ALIGN(8);
  } >RAM

  

  /* Remove information from the standard libraries */
  /DISCARD/ :
  {
    libc.a ( * )
    libm.a ( * )
    libgcc.a ( * )
  }

  .ARM.attributes 0 : { *(.ARM.attributes) }
}

`

epikao, 2022-05-25:
> I've added these lines to the end of `lv_example_meter_1()`:
> 
> ```c
>     volatile uint32_t t = lv_tick_get();
>     uint32_t i;
>     for(i = 0; i < 1000; i++) {
>     	lv_meter_set_indicator_value(meter, indic, lv_rand(0, 100));
>     }
>     volatile uint32_t elaps = lv_tick_elaps(t);
>     printf("%d ms\n", elaps);
> ```
> 
> and tested it on STM32F769 Discovery.
> 
> It printed `75 ms` (for 1000 invalidations). So 1 invalidation took 75 us.

How much time does your lv_task_handler() need, always the same time?

pete-pjb, 2022-05-25:
hi @epikao ,

I have taken a quick look over the configuration and it seems okay to me, it was worth checking though.  I am afraid I am out of time today may be around tomorrow.  I hope you track it down soon!

Kind Regards,

Pete

kisvegabor, 2022-05-25:
> According my scope you can see, that lv_task_handler() sometime is fast, but sometimes with big delays (>50ms).

When LVGL renders (in every 30 ms by default) is lasts much longer. 

BTW, `HAL_UART_Transmit(&huart3,buf8,sizeof(buf8),1);` 
should be 
`HAL_UART_Transmit(&huart3,buf8,strlen(buf8),1);`

Have you tried the "manual profiling" I suggested [here](https://github.com/lvgl/lvgl/issues/3328#issuecomment-1137142609)?

epikao, 2022-05-25:
> When LVGL renders (in every 30 ms by default) is lasts much longer.
>
> BTW, `HAL_UART_Transmit(&huart3,buf8,sizeof(buf8),1);` should be `HAL_UART_Transmit(&huart3,buf8,strlen(buf8),1);`
>
> Have you [...]

Thank you, and no, i have not tried the "manuele profiling" because i don't understand exactly how or what is meant by it.

Anyway, may I come back to talk about your for-loop (1000 invalidations).
As you can see in the scope image below, my lv_set_indicator function (invalidations) is also executed very fast (within 30ms, so more than twice as fast as yours).

The problem is that the lv_task_handler needs 180ms afterwards??
I think from a certain number of the lv_set_indicator function is executed, the lv_task_handler comes to a halt.
With other function like the lv_chart_value_by_id function, I have not these big delays between....

Can you maybe evaluate the time your lv_task_handler needs after executing this lv_set_indicator always 1000 times in the while loop?

```
while (1)
{
    	////////////////// TESTCODE ///////////////////////////

        if( ( millis()-lastMillis2 ) > 4){
        	lv_task_handler();
        	lastMillis2 = millis();
        }

        for(i = 0; i < 1000; i++) {
        	lv_meter_set_indicator_value(guider_ui.screen_meter_1, screen_meter_1_scale_1_ndimg_0, lv_rand(0, 1000));
                HAL_GPIO_TogglePin(GPIOI, GPIO_PIN_8); //LED 101
        }

}
```

[image]

epikao, 2022-05-26:
> It might help to see in which functions the MCU is in most of the times (statistically).

Ok, thank you, will try this asap.

> I'm lost a little bit. Seeing your previous code snippets, you have tmeasured he time of `lv_task_handler` with the LED toggling, right?

the LED toggles in the for loop, so I measure both, the `lv_set_indicator` time + the `lv_task_handler` time.

> With such a large meter (`lv_obj_set_size(ui->screen_arc_1, 414, 540);`) and some not so good config rendering might take long.

Ok,, I do not understand very well about the rendering processes / optimizations. However, I'm now more focusing on the lv_meter function/code...

> * Are the buffer(s) if `lv_disp_draw_buf_t` stored in internal or external RAM?

Not sure, but when I comment out the `BSP_SDRAM_Init();`, I have a black image..
The problem remains the same even if I remove everything in the flush function, also the same if I disable the touch-function/driver. So flush and touch-driver are ok...

I'm using the lv_port_stm32f746_disco, and since I'm not using a 480x272 but a 1024x600 display, I simply changed the TFT_HOR_RES and TFT_VER_RES + Timing-Parameter.
Would I also need to adjust something on the buffer (buf1_1...)?

[image]

> Have you enabled -O2 or -O3 optimization?

Ok, I still have to look where I set this...

> I can check it on Monday, but I'm pretty sure it's ~20ms. With 180 ms there would be only 5 FPS which is certainly (hopefully) not the case.

Ah yes, the FPS is also an indication. I enabled it now, and it shows 5 FPS...

the STM32F746 board that I am using now is a modified version of the Discovery variant, but I am using the lv_port for the Discovery. Unfortunately I don't know what exactly is different, but I have the code configurations. And I think the system clock is correct. And I do not know what should run in the background? I have the problem only with lv_meter and lv_arc...
The demos also all run very smoothly...

But I also have the STM32F769 Discovery board here. Will test the whole thing with the corresponding lv_port and with this board....

epikao, 2022-05-26:
I think it is a buffer problem. Just remarked that I changed the origin value 68 to 10, because otherwise it does not fit in the RAM.
Hmm, now I changed 10 to 40... and FPS is 8... so better :-) ....

But is my RAM too small now? Or is the external RAM not used.... 
Unfortunately, I am not familiar with the RAM ...

epikao, 2022-05-26:
the RAM size was set to 16, but possible is 32. now i have changed this (see screenshot below), and still i can not increase the buf to 68....
Is there another place where I have to tell the SDRAM size?

[image]

embeddedt, 2022-05-26:
You probably need to adjust a linker script as well.

epikao, 2022-05-27:
> You probably need to adjust a linker script as well.

I think there is no setup for SDRAM in the linker script.
#define SDRAM_MEMORY_WIDTH FMC_SDRAM_MEM_BUS_WIDTH_16 or 32 has no influence,

I.e. the maximum framerate I can manage is 8-9 FPS, with a buf value of 61 (resolution 1024 x 600).
Would be interesting now how it looks like with the STM32F769 discovery board....

But if also in this range, then this is probably normal.
Then my question would be what more I could optimize to get more FPS here?

Thank you

kisvegabor, 2022-05-27:
With STM32F769 on 800x480 I got ~20 FPS at worst case for the most complex screens. Your screen is 1.6 times larger -> 20/1.6 = ~12 FPS. And maybe some not as-good-as-possible settings here and there -> 10 FPS. So it's somewhat there. 

Regarding the buffer size: usually 1/10 screen size is enough. Increasing the buffer size further results only minor change in performance.

epikao, 2022-05-27:
> With STM32F769 on 800x480 I got ~20 FPS at worst case for the most complex screens.

Ok thank you. To be sure, with the code below you will reach 20FPS ? (with the lv_demos like "stress, benchmark" I also reach 20 FPS and higher, but not with the code below).

```
while (1)
{
    	////////////////// TESTCODE ///////////////////////////

        if( ( millis()-lastMillis2 ) > 4){
        	lv_task_handler();
        	lastMillis2 = millis();
        }

        for(i = 0; i < 1000; i++) {
        	lv_meter_set_indicator_value(guider_ui.screen_meter_1, screen_meter_1_scale_1_ndimg_0, lv_rand(0, 1000));
                HAL_GPIO_TogglePin(GPIOI, GPIO_PIN_8); //LED 101
        }

}

```

kisvegabor, 2022-05-27:
I can try with a very large (480x480) meter on Monday. To have something simple for testing, let's use `lv_example_meter_*()` examples.

epikao, 2022-05-29:
> I can try with a very large (480x480) meter on Monday. To have something simple for testing, let's use `lv_example_meter_*()` examples.

Ok, thank you very much. With this lv_meter example 1 (code below) I achieve **11 FPS**.

```
    /// meter example ///

    meter = lv_meter_create(lv_scr_act());
    lv_obj_center(meter);
    lv_obj_set_size(meter, 480, 480);

    /*Add a scale first*/
    lv_meter_scale_t * scale = lv_meter_add_scale(meter);
    lv_meter_set_scale_ticks(meter, scale, 41, 2, 10, lv_palette_main(LV_PALETTE_GREY));
    lv_meter_set_scale_major_ticks(meter, scale, 8, 4, 15, lv_color_black(), 10);

    lv_meter_indicator_t * indic;

    /*Add a blue arc to the start*/
    indic = lv_meter_add_arc(meter, scale, 3, lv_palette_main(LV_PALETTE_BLUE), 0);
    lv_meter_set_indicator_start_value(meter, indic, 0);
    lv_meter_set_indicator_end_value(meter, indic, 20);

    /*Make the tick lines blue at the start of the scale*/
    indic = lv_meter_add_scale_lines(meter, scale, lv_palette_main(LV_PALETTE_BLUE), lv_palette_main(LV_PALETTE_BLUE),
                                     false, 0);
    lv_meter_set_indicator_start_value(meter, indic, 0);
    lv_meter_set_indicator_end_value(meter, indic, 20);

    /*Add a red arc to the end*/
    indic = lv_meter_add_arc(meter, scale, 3, lv_palette_main(LV_PALETTE_RED), 0);
    lv_meter_set_indicator_start_value(meter, indic, 80);
    lv_meter_set_indicator_end_value(meter, indic, 100);

    /*Make the tick lines red at the end of the scale*/
    indic = lv_meter_add_scale_lines(meter, scale, lv_palette_main(LV_PALETTE_RED), lv_palette_main(LV_PALETTE_RED), false,
                                     0);
    lv_meter_set_indicator_start_value(meter, indic, 80);
    lv_meter_set_indicator_end_value(meter, indic, 100);

    /*Add a needle line indicator*/
    indic = lv_meter_add_needle_line(meter, scale, 4, lv_palette_main(LV_PALETTE_GREY), -10);

    /// END meter example ///

    while (1)
    {
    	////////////////// TESTCODE ///////////////////////////

        if( ( millis()-lastMillis2 ) > 4){
        	lv_task_handler();
        	lastMillis2 = millis();
        }
       uint32_t i;
       for(i = 0; i < 1000; i++) {
        	lv_meter_set_indicator_value(meter, indic, lv_rand(0, 100));
    		HAL_GPIO_TogglePin(GPIOI, GPIO_PIN_8); //LED 101
        }
    }
```

kisvegabor, 2022-05-30:
I got 25 FPS with 80% CPU usage. 
However, I also got 11 FPS when I left this in my main loop:
```c

	while(1) {
	    lv_task_handler();
		HAL_Delay(5);
		lv_obj_invalidate(lv_scr_act());  //SHOULDN'T BE HERE
	}
```

Please enable `LV_USE_REFR_DEBUG` in `lv_conf.h` and take a video about the display.

kisvegabor, 2022-06-01:
The second video shows that the whole screen is refreshed continuously.  It can happen because you make 1000 invalidations and LVGL can't store that many invalid areas, so it rather refreshed the whole screen. 

Please take video with these code:
```c
while (1)
    {
        if( ( millis()-lastMillis2 ) > 50){
        	lv_meter_set_indicator_value(meter, indic, lv_rand(0, 100));
        	lv_task_handler();
        	lastMillis2 = millis();
        }
    }
```

epikao, 2022-06-01:
> The second video shows that the whole screen is refreshed continuously. It can happen because you make 1000 invalidations and LVGL can't store that many invalid areas, so it refreshes the whole screen instead.
> 
> Please take video with these code:
> 
> ```c
> while (1)
>     {
>         if( ( millis()-lastMillis2 ) > 50){
>         	lv_meter_set_indicator_value(meter, indic, lv_rand(0, 100));
>         	lv_task_handler();
>         	lastMillis2 = millis();
>         }
>     }
> ```

Ok thank you. See following the video link:

https://drive.google.com/file/d/1wsBhTEU12uxVSmk6UlYahTPI6LtEseU8/view?usp=sharing

kisvegabor, 2022-06-01:
Seems much better! Without `LV_USE_REFR_DEBUG` there should be a great FPS.

epikao, 2022-06-01:
> Seems much better! Without `LV_USE_REFR_DEBUG` there should be a great FPS.

Ok, but I think I had a comprehension problem. 

For example, if I want to get data every 5ms and display it in real time (e.g. with lv_chart), it doesn't work, or is difficult (with a Raspberry it would maybe work)....
But if the data would theoretically come only every >50ms, then it would be possible to display it in real time, right?

I.e. I have to cache this data. But the examples with lv_chart seems all based on a constant not changing data... I still have to familiarize myself with this...

 And "lv_task_handler" must be called theoretically only when there is a change on the display, right?

However, sorry for all the time, and thank you very much for this support.

kisvegabor, 2022-06-01:
> if I want to get data every 5ms and display it in real time 

To do this you need a display with 200 Hz refresh rate, 200 FPS and very good eyes to see the new data every 5 ms. :slightly_smiling_face: 

In reality your display probably has ~30 Hz refresh rate, so ~30 FPS -> ~33ms data rate would be reasonable. 

It doesn't make much sens to update the meter in every 5 ms because you will see only every 6th position. For a chart you can add data in every 5ms, but you will see 6 new data in every 30ms (assuming that the FPS doesn't drop)

closed 2022-06-02 by epikao