lvgl/lvgl #3329: OpenGL ES 2.0 Implementation

tanaytekin, 2022-05-05:
### Introduce the problem

There is no OpenGL implementation of LVGL.

### Examples and cases

LVGL forum link is [here](https://forum.lvgl.io/t/opengl-es-2-0-implementation-of-lvgl/8729).

#### Screenshot
[image]

### Suggested solution

We are currently trying to implement LVGL draw functions with OpenGL ES 2.0 to achieve GPU acceleration with PinePhone.
The project is currently **work in progress** because of that it may need heavy refactoring. Any thoughts and contributions are welcome :pray:. 

The drawing functions are [here](https://github.com/rainlab-inc/lvgl/tree/gles/src/draw/gles).

Example SDL driver is [here](https://github.com/rainlab-inc/lv_drivers/blob/gles/sdl/sdl_gles.c).

Example SDL application is [here](https://github.com/rainlab-inc/lvgl-opengl).

#### Curently implemented draw calls

More detailed list can be found in example [application](https://github.com/rainlab-inc/lvgl-opengl)

Some of the drawing code ported from [SDL implementation](https://github.com/lvgl/lvgl/tree/master/src/draw/sdl).

##### Rectangle 
- draw_bg : plain_rect, radius mask
##### Image
- draw_img : with texture caching
##### Letter
- draw_letter : with texture caching

###### Credits
- Thanks to @mariotaku for texture caching part in SDL drawing code.

Especially waiting comments, feedback, and contributions from folks with OpenGL(ES 2.0) knowledge :pray:.

mariotaku, 2022-05-05:
Awesome! Although I didn't take a closer look at it, it would certainly be really nice.

kisvegabor, 2022-05-20:
Are there any questions or updates on the OpenGL ES port?

tanaytekin, 2022-05-21:
I've done initial work to get it done, but I've used whatever means necessary to get it working. Implementation needs a lot of work which can be done over time, but mainlining support would be extremely welcome 🙏.  How much is it upstreamable in its current form, and what should we change/refactor/rewrite to get it merged in repositories (as a beta driver) so that more people can try/test/improve?

kisvegabor, 2022-05-28:
Please a PR if you have some time.

tanaytekin, 2022-05-29:
Hello,

Sorry for late reply. I haven't done a detailed test yet, but I tried it with the widget demo on the desktop with a dedicated graphics card(GeForce GTX 750 Ti) , and I got 300-400 fps. Of course, shadows etc. are not drawn in the drawing process yet, so these results may not be very healthy. However, there is no optimization at the moment. There may even be a slight drop in performance on some devices at first, until optimizations are made. We will also test it on a couple of devices.

kisvegabor, 2022-05-30:
>  I got 300-400 fps.

Not bad. :sunglasses: 

> Of course, shadows etc. are not drawn in the drawing process yet, so these results may not be very healthy. 

No problem with having these limitations.

kisvegabor, 2022-06-14:
Let's continue in the PR: https://github.com/lvgl/lvgl/pull/3389