lvgl/lvgl #3332: Recolor lost on wrapped label

andresovela, 2022-05-05:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
Recolor is lost on wrapped labels. Apparently this is a known issue but I couldn't find it in Github. I apologize if this is duplicated somewhere else.

### To Reproduce
See [LVGL forum post](https://forum.lvgl.io/t/recolor-lost-effect-on-wrapped-label/8671/2).

### Expected behavior
The recolor commands should still work after the label wraps.

### Screenshots or video
See [LVGL forum post](https://forum.lvgl.io/t/recolor-lost-effect-on-wrapped-label/8671/2).

andresovela, 2022-05-05:
I'm interested in fixing this issue, if I can get some guidance that'd be great :)

kisvegabor, 2022-05-05:
The problem with supporting it is that it's assumes that LVGL will parse all the lines to find out of there is recolor in the middle of the text. But LVGL just quickly finds the starting line which is visible now, and starts processing from there. If the text is very long it saves a lot of performance.

What about [lv_span](https://docs.lvgl.io/master/widgets/extra/span.html)? It's a more future rich implementation to change the style of text snippets.

andresovela, 2022-05-06:
I already have this working with lv_span, but to be honest it's much harder to use than a simple label. As a user I'd be willing to sacrifice performance for very long strings (which I'm not using anyway) in exchange for being able to use the recolor feature in wrapped strings.

kisvegabor, 2022-05-17:
Sorry for saying this but I think it'd add to much complexity for a small gain considering the whole user base. 

So I'd avoid supporting multi-line recolor for now, but I'll keep it in mind. Usually we find ways to support features like these as the architecture evolves.

andresovela, 2022-05-18:
Okay, I understand, in any case I have implemented my use case with lv_span, so this issue is not really a blocker for me, but I know it could have been much easier if labels supported this out-of-the-box.

kisvegabor, 2022-05-18:
Thanks for the understanding. :slightly_smiling_face: 
I close this issue for now.

closed 2022-05-18 by kisvegabor