lvgl/lvgl #3333: Feature request: LV_LABEL_LONG_WRAP_SCROLL_VERTICALLY

andresovela, 2022-05-05:
As discussed in the [LVGL forum post](https://forum.lvgl.io/t/how-to-make-a-label-wrap-and-scroll-vertically-and-possible-recoloring-bug/8664), I'm opening an issue for this feature.

I'm interested in implementing this too, if I can get some guidance that'd be best :)

Basically what I want is a label that wraps automatically according to its size constraints, and it scrolls in some way to show the rest of the label. The scrolling part may need to be defined better. Maybe a `LV_LABEL_LONG_WRAP_SCROLL_CIRCULAR_VERTICALLY` option would also be needed. Not sure.

I would also support animation templates for this option, as implemented in https://github.com/lvgl/lvgl/pull/3128.

kisvegabor, 2022-05-07:
Hi,

> I'm interested in implementing this too.

Glad to hear that! :slightly_smiling_face: 

I think adding `LV_LABEL_LONG_WRAP_SCROLL_VERTICALLY` and `LV_LABEL_LONG_WRAP_SCROLL_CIRCULAR_VERTICALLY` would be a little be too verbose and bloated. What do you think about allowing `LV_LABEL_LONG_WRAP | LV_LABEL_LONG_SCROLL`. It keeps the wrap behavior horizontally, but scrolls vertically if applicable. At first look it can be achieved by minimal code change.

andresovela, 2022-05-09:
That makes sense and it's a much clearer and straightforward solution :)

andresovela, 2022-05-24:
Since #3332 was closed, I have no use for this at the moment.