lvgl/lvgl #3339: glitch LCD Display

phucloi97, 2022-05-07:
hi there. i have problem register display. i config it
`    static lv_disp_draw_buf_t draw_buf_dsc_1;
    static lv_color_t buf_1[320 * 10];                          
    lv_disp_draw_buf_init(&draw_buf_dsc_1, buf_1, NULL, 320 * 10);   

    static lv_disp_drv_t disp_drv;                    
    lv_disp_drv_init(&disp_drv);                    

    disp_drv.hor_res = 320;
    disp_drv.ver_res = 240;`

but it glitch outpu as same as
[image]
please help me! thanks

embeddedt, 2022-05-08:
Were you able to solve this or did you close it by accident?

phucloi97, 2022-05-08:
Nobody help it. I close it

embeddedt, 2022-05-08:
That might be related to the fact that you posted it on the issue tracker instead of on the [forum](https://forum.lvgl.io/), and officially the issue tracker shouldn't be used for questions like this. If you post it on the forum and link to it here, I can try to take a look. Please be sure to include your `flush_cb` implementation as that is likely where the bug is.

phucloi97, 2022-05-08:
Okay. I will try it again

closed 2022-05-30 by lvgl-bot