lvgl/lvgl #3341: Error with the examples in the docs

kisvegabor, 2022-05-07:
### Describe the bug
All the the interactive examples are showing the same examle:
[image]

### To Reproduce

Go to any page with examples. E.g:
- https://docs.lvgl.io/master/widgets/core/bar.html#example
- https://docs.lvgl.io/master/examples.html

cc @embeddedt

embeddedt, 2022-05-07:
I made only [one change](https://github.com/lvgl/lv_web_emscripten/commit/36bc3ee67a783aac7f5791653a783dbf3c9f55d7) recently and it was just to fix the build system to compile LVGL as a proper library instead of the old globbing approach. I also went looking in DevTools and the URLs do seem to have the correct example name in them even though that's not what loads. Very strange!

kisvegabor, 2022-05-07:
Yes, I've also checked the inspector and all looks good.

And the URL really shows the actual demo: https://docs.lvgl.io/master/_static/built_lv_examples/?example=lv_example_bar_3&w=320&h=240

kisvegabor, 2022-05-07:
I found something. If in the inspector I change the version in the link from `8.3` to `8.1`, `8.2` or `master` is works. If I change back to `8.3` the incorrect example appears again.

embeddedt, 2022-05-07:
@kisvegabor I think I may have fixed it with https://github.com/lvgl/lvgl/commit/d367bb7cf17dc34863f4439bba9b66a820088951; can you check on your end?

kisvegabor, 2022-05-09:
I confirm it's working now. Thanks for the quick response!

closed 2022-05-09 by kisvegabor