lvgl/lvgl #3343: Style system resource usage

excitedbox, 2022-05-09:
### Introduce the problem
The style system uses data structures such as linked lists, array iteration loops that add together to use huge amounts of resources. I know I have posted this before in the markup language thread but I believe it is better to take things one step at a time and this change would make it easier to add a markup language at a later time.

### Examples and cases
`lv_res_t lv_obj_get_local_style_prop(lv_obj_t * obj, lv_style_prop_t prop, lv_style_value_t * value,
                                     lv_style_selector_t selector)
{
    uint32_t i;
    for(i = 0; i < obj->style_cnt; i++) {
        if(obj->styles[i].is_local &&
           obj->styles[i].selector == selector) {
            return lv_style_get_prop(obj->styles[i].style, prop, value);
        }
    }

    return LV_RES_INV;
}`

### Suggested solution
[The big O Cheat Sheet will illustrate what I mean](https://mezgitci9.medium.com/big-o-notation-and-algorithm-analysis-with-javascript-examples-12e6c8040715)

As you can see there is a huge difference between different ways of retrieving data, which is multiplied for each widget, style, style property etc. Switching to a system that offers O(1) time for search, insertion and removal will increase lvgl performance by multiple factors of magnitude.

my suggestion is storing an index of style properties (width, opacity, etc) and a list of styles, style properties for each widget being drawn. This would remove the loops from the search and insertion and allow retrieving them directly using pointers or array element ids. (ie. get style[id] from widget[id]). This can be done with either struct of struct or array of array or combination of arrays and structs. Much of this is already done by lvgl and mainly requires changing how the data is organized and retrieved.

Implementing this would require lists (dictionary/index) of all possible widgets, styles, and properties used in lvgl. Their element pointers or ids can then be used to maintain the lists of instances used by the draw loop. Each update cycle you would then loop through a list of changes to the instances and use the pointers/ids to access the elements directly.

kisvegabor, 2022-05-11:
Hi,

Be sure we are on the same page, please draft some pseudo code/data structure for this UI:

Styles:
```
- style1:
   - opa:50
   - border_width:3   
  
- style2:
   - border_width:5   
```
Widgets:
```
- btn1
   - style1 in default state
   - style2 in pressed state   

- btn2
   - style1 in default state
   - style2 in pressed state   

- slider1
   - style1 in default state of indicator part
   - style2 in pressed state of indicator part
```

embeddedt, 2022-05-17:
@kisvegabor If I understand correctly, the suggestion here is fairly straightforward. Currently styles and their properties are effectively stored as arrays within a widget. This leads to performance issues as the number of styles increase since retrieving a property can take `O(n*m)` time in the worst case, where n is the average number of properties per style and m is the number of styles on the object.

Changing the styles and properties to be stored using a hash table or similar structure should largely alleviate this problem and should not require any API-facing changes at all. The idea is to find a storage mechanism that allows a property to be looked up  in `O(1)` time from a style and, most importantly, does not require a loop over all the properties.

Personally, I wouldn't change anything about the styles yet as there are probably a lot less of them per object than there are properties.

embeddedt, 2022-05-17:
A hash map doesn't need to store every property. I believe most hash map implementations actually use several buckets behind the scenes and group the properties together. With an adequately chosen bucket size (maybe 4-16 properties, but we'd have to test), the cost of finding a property will still be very close to `O(1)`. If a bucket is not present, we know the style doesn't have any of those properties.

embeddedt, 2022-05-18:
The profiling results suggest to me that we really should solve both issues at the same time. I am curious about the `lv_color_mix` overhead, but it's an unrelated topic.

A cache style like you suggested could work to store a flattened representation of the object's current properties. That way only one style needs to be read from instead of checking all of them. We could also use a hashmap or sorting system on just that style, since the overhead would be less than keeping every style organized in that manner.

The issue now would be efficiently invalidating it when a non-local style gets changed. We currently have no mapping of styles back to widgets which are using them, so we would either need to add that, or scan the entire widget tree every time a style property changes, which will be too slow. Do you have another idea?

kisvegabor, 2022-05-18:
It's a problem in the current version too because LVGL needs to know if a style changes e.g to recalculate layouts. Now you need to call this function manually if a style is changed when object already using it:
```c
/**
 * Notify all object if a style is modified
 * @param style     pointer to a style. Only the objects with this style will be notified
 *                  (NULL to notify all objects)
 */
void lv_obj_report_style_change(lv_style_t * style);
```

kisvegabor, 2022-05-18:
(I tagged the wrong issue in a commit...)

kisvegabor, 2022-05-28:
Let's summarize it:
- we create a cached style for each part used by the widget
- it contains the current properties considering the current transition, state, etc (what `lv_obj_get_style_...` returns)
- add only the queried properties, so only the really used properties will be cached
- automatically invalidate cache styles only style change
- store the properties in this style in  smart way, (still not sure how) to make getting props as optimal as possible. 

Anything else?

embeddedt, 2022-05-28:
Yes. This won't necessarily speed up the process of modifying the styles, but rendering shouldn't be slowed down much anymore if accessing properties in the cached style is efficient. Nothing else comes to mind at the moment.

kisvegabor, 2022-05-29:
I've implemented the caching as described above (the cache styles are stored/managed normally  for now), and the widget demo now needs 120kB of RAM. Previously it was 60kB on my notebook. 

Quick calculations: 100 widgets, 2 parts per widget, 40 cached props per part = 8000 props. Let's say 8 byte per prop = 64kB, so it makes sense. During debugging I really saw that really there are about ~40 cached properties. 

The profiler has shown that we can gain about 5% performance. E.g. 30FPS to 31-32FPS. It's clearly not worth it in this form. :slightly_frowning_face:

embeddedt, 2022-05-29:
Interesting; that's not as much of an improvement as I'd expect. What lines are hotspots now?

kisvegabor, 2022-05-30:
This the new profiler result from PC (the style related functions are highlighted):
[image]

On STM32F769:
[image]

`lv_style_get_prop_inlined` is increased from 3% to 6% which is normal because when a lot of properties are stored in a style `group` is not effective and the property really can be searched. Counting with 32 props/style, with a simple logarithmic search we could lower O(32) to O(5) However 1 step of a logarithmic search is takes more time than a simple linear search, so in the end, probably 6% will be 1-2%. `get_prop_core` (which handles inheritance and filtering relevant styles) is almost disappeared (was 7%). So all in all, an optimistic gain can be 10% (30 FPS-> 33FPS).  

My conclusion is that we can achieve this 5-10% gain by
- using less styles (more optimal themes, or reconsider how they should work) -> faster `get_prop_core`
- store all style properties ordered to let `lv_style_get_prop` perform logarithmic search.

embeddedt, 2022-06-04:
> using less styles (more optimal themes, or reconsider how they should work) -> faster get_prop_core

I think the tweaks you proposed to themes in https://github.com/lvgl/lvgl/pull/3390#pullrequestreview-990710921 should address exactly that, as well as cutting down on RAM usage.

kisvegabor, 2022-06-19:
I added a note the [v9 reminder](https://github.com/lvgl/lvgl/issues/3392) about reconsidering the themes. 

I close this issue now and we can open a more targeted one when we get there.

commit 100197d4 (2022-05-18): fix(color): compensate rounding error during blending

fixes #3343
closed 2022-06-19 by kisvegabor