lvgl/lvgl #3350: Combination of chroma key + rotation does not work

nicusorcitu, 2022-05-11:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
Chroma key works, rotation works. But the combination of the two fails.

### To Reproduce
```c
lv_obj_t * obj = lv_img_create(lv_scr_act());
lv_obj_remove_style_all(obj);
lv_img_set_src(obj, &img_cogwheel_chroma_keyed);
/*Rotate 45.0 degrees*/
lv_img_set_angle(obj, 450);
lv_obj_center(obj);
```

### Expected behavior
Image shall be rotated but chroma key removed.

### Screenshots or video
[image]

kisvegabor, 2022-05-16:
I've fixed it in master.

kisvegabor, 2022-05-16:
Makes sense, updated. :slightly_smiling_face:

closed 2022-05-16 by kisvegabor

kisvegabor, 2022-05-16:
https://github.com/lvgl/lvgl/commit/ccbb41dedcf9c6424897fab9a1627bad63b7cc52 is tagged incorrectly to this issue. It's about #3351.