lvgl/lvgl #3356: flush_cb is called twice during one render pass.

AfflatusX, 2022-05-12:
### Describe the bug
I have a disp_drv using 1 full screen buffer, size set to screen size, and direct_mode set to true. In the render loop, I do two things - call lv_task_handler, and call lv_tick_inc. I put logging before and after the render pass, as well as a log in the flush_cb function. I noticed that flush_cb is called twice per loop run, each with the area being full screen size from 0,0.

### To Reproduce
see the bug description above

### Expected behavior
the flush_display should only be called once

### Screenshots or video
48525   LVGL  INFO render loop start
48530   LAPP  INFO flushing display with area: (0, 0), (379, 379)
48545   LAPP  INFO disp flush duration: 15 ms
48550   LAPP  INFO flushing display with area: (0, 0), (379, 379)
48565   LAPP  INFO disp flush duration: 15 ms
48565   LVGL  INFO finished task handler
48565   LVGL  INFO finished tick inc
48565   LVGL  INFO LVGL duration: 40 ms
48565   LVGL  INFO render loop end

Anything could cause this issue?

FASTSHIFT, 2022-05-17:
Add `lv_disp_flush_is_last()` judgment in `flush_cb()`.

AfflatusX, 2022-05-22:
> Add `lv_disp_flush_is_last()` judgment in `flush_cb()`.

Hmm I thought with direct mode it should only trigger flush once per tick?

AfflatusX, 2022-05-22:
> It's strange, it seems ok on my end. Do you use the latest LVGL version from the master branch?
> 
> Can you reproduce it in a simulator?

I haven't tried with the simulator since I suspect this is related to the timing on the hw. I'm changing the implementation anyway so I'll re-open this if I notice it happening again and try to repro in emulator.

AfflatusX, 2022-06-08:
> Add `lv_disp_flush_is_last()` judgment in `flush_cb()`.

Yeah actually adding some checking I noticed one of the call has lv_disp_flush_is_last return false. @kisvegabor  am I misunderstanding is_direct flag?

kisvegabor, 2022-06-11:
`direct` means only that LVGL draws directly into a screen sized buffer. With `direct=false` you will get rendered images which needs to be copied to the correct area of the frame buffer. With `direct=true` this copying can be saved.

However with `direct=true` `flush_cb` should be called only once per rendering cycle. It's not the case now which is bug and as a workaround you can use `lv_disp_flush_is_last()` to find which call is the last. I can't fix this bug now as some people might rely on this behavior but it'll be fixed in v9 for sure.

AfflatusX, 2022-06-11:
> `direct` means only that LVGL draws directly into a screen sized buffer. With `direct=false` you will get rendered images which needs to be copied to the correct area of the frame buffer. With [...]

Thanks for confirming - what's the rough timeline of v9?

kisvegabor, 2022-06-11:
I can foresee only the timeline of v8.3 now :slightly_smiling_face: It's planned for early July. After that we start to work on v9 which will take at least a half year.

AfflatusX, 2022-06-11:
cool, sounds good to me! only asking for planning purpose, the workaround seems to be fine.

closed 2022-06-11 by AfflatusX

bhushan131288, 2024-03-26:
is this issue has been resolved in v9? Also for direct mode do i need to allocate separate screen size buffer for LVGL other than the actual Frame buffer?

kisvegabor, 2024-03-27:
Buffering was improved in v9, see here: https://docs.lvgl.io/master/porting/display.html
Could you give it a try?

> Also for direct mode do i need to allocate separate screen size buffer for LVGL other than the actual Frame buffer?

Yes, if you used the active frame buffer, you would see as LVGL is drawing the widgets one after another.

bhushan131288, 2024-03-27:
But now with Direct mode also, i can see flush_cb is getting called multiple times. I have 5 bar graphs on screen with changing value at every main loop, so expected is that drawing shall happen for all five then only flush_cb get called in "LV_DISPLAY_RENDER_MODE_DIRECT" mode, but its called for all 5 objects.

kisvegabor, 2024-03-27:
LVGL calls the `flush_cb` for each updated area because the driver might need to know which areas where changed.
 
You can use `lv_display_flush_is_last()` in the `flush_cb` to see if that call was for the last update.

bhushan131288, 2024-03-27:
ok.. Thanks for your reply.. it means using double for LVGL will improve the performance and also will not see any drawing on screen one after another..