lvgl/lvgl #3357: slider on tileview

JustQ1989, 2022-05-14:
i place a slider on one of a tileview  page，the tileview don‘t only has one page，when i slide the slider，the page slide，slider work wrong

how to resolve this

kisvegabor, 2022-05-16:
Slider shouldn't propagate the scrolling to the parent. Please send a code snippet to reproduce the issue.

JustQ1989, 2022-05-17:
> Slider shouldn't propagate the scrolling to the parent. Please send a code snippet to reproduce the issue.
this is my code
[Lcd86Box.zip](https://github.com/lvgl/lvgl/files/8704377/Lcd86Box.zip)

kisvegabor, 2022-05-17:
It seems you are using LVGL v7. GuiGuider already supports LVGL v8 too. Is it an option for you to update to v8?

JustQY, 2022-05-17:
> It seems you are using LVGL v7. GuiGuider already supports LVGL v8 too. Is it an option for you to update to v8?

thinks，the v8 has no this issue

closed 2022-05-17 by kisvegabor