lvgl/lvgl #3358: Implementing box layout logic

rmantovani68, 2022-05-16:
### Introduce the problem
Would be nice to implement a box layout logic similar to the one used in gtk
Documentation can be found here : (https://hackthedeveloper.com/gtk-box-layout/) and here (https://docs.gtk.org/gtk3/class.Box.html).
This way would be possbile to easily manage the layout of the widgets.
I started to implement it and found that there's no way in lvgl to know the height and width of the widget (preferred height and width), because the dimensions are set in the creation funcitons.
In addition to the box layout there's the grid layout and others.

kisvegabor, 2022-05-16:
Hi,

Now we have [flexbox](https://docs.lvgl.io/master/layouts/flex.html) and [grid](https://docs.lvgl.io/master/layouts/grid.html). These are heavily inspired by CSS. 

I've checked the common UI libs (including GTK) and haven't found any layout systems more powerful then these. Of course, I might have missed something. 

Do you see something specific you can't realize with LVGL's flex and grid?

kisvegabor, 2022-05-17:
As I see GTK Box is like a subset of CSS (and LVGL) Flexbox. Can you mention a use case which is simpler to do with GTK Box? 

I'm very open to learn about new layout concepts, so please don't take it as nitpicking but rather curiosity.

rmantovani68, 2022-05-18:
ok

1 : you dont't need to know in advance how many columns or how many rows
2 : you can decide for every child: expand / filll / padding properties
3 : you can decide for the container : direction (normal / reverse) / orientation (horizontal / vertical) / spacing between children / homogeneus / non homogeneus.
4 : you can change the position of the children

- it's another way of managing layouts 
- can be used in substitution of grid and flex (different layout management methods can live together)
- it is useful in many cases where you don't know how many children you have and it is more flexible than grid or flexgrid.
- you can do everything with every layout management

I don't want to convince you to work on my suggestions, but if you agree I can submit a merge request on this.

kisvegabor, 2022-05-18:
> you dont't need to know in advance how many columns or how many rows

It's true for flex too. See [here](https://docs.lvgl.io/master/layouts/flex.html#a-simple-row-and-a-column-layout-with-flexbox).

> 2 : you can decide for every child: expand / filll / padding properties

Also works in flex. See [here](https://docs.lvgl.io/master/layouts/flex.html#demonstrate-flex-grow). Moreover you can set the portion to take from the free space relative to other "grow" children.

> you can decide for the container : direction (normal / reverse) / orientation (horizontal / vertical) / spacing between children / homogeneus / non homogeneus.

For flex see [this](https://docs.lvgl.io/master/layouts/flex.html#flex-align).

> I don't want to convince you to work on my suggestions, but if you agree I can submit a merge request on this.

I just want to be sure it's really a new functionality to avoid maintaining an extra layout without added value. If it really makes something possible which is not possible or complicated with flex I'm very to open to add Box.

rmantovani68, 2022-05-18:
Everything is possible with everything ... I mean ... IMHO it's about options and flexibility.
In any case I am already using a box layout in my lvgl applications, 
I will submit some examples and test cases.

kisvegabor, 2022-05-18:
> I will submit some examples and test cases.

It'd be great!

closed 2022-06-10 by lvgl-bot