lvgl/lvgl #3360: Image disappears when updating lv_label

hard-cat, 2022-05-17:
I updated the library in the project from 7.10 to 8.2, and got a strange effect.

I displayed 2 labels, and periodically change the text in them, sometimes the label image disappears, and the next time lv_label_set_text_fmt is called, it returns to the screen.
When using version 7.10, this is not observed.

kisvegabor, 2022-05-17:
Did you pay attention to [these](https://docs.lvgl.io/master/CHANGELOG.html#driver-changes) when you updated to v8?

hard-cat, 2022-05-17:
> Did you pay attention to [these](https://docs.lvgl.io/master/CHANGELOG.html#driver-changes) when you updated to v8?

Yes, I fixed the initialization as described

#define DISPLAY_BUFF_SIZE (240*40)

static void gui_task(void * p)
{

	lv_init();
    lvgl_driver_init();

    EXT_RAM_ATTR static lv_color_t buf1[DISPLAY_BUFF_SIZE];
    EXT_RAM_ATTR static lv_color_t buf2[DISPLAY_BUFF_SIZE];

	static lv_disp_draw_buf_t disp_buf;
	lv_disp_draw_buf_init(&disp_buf, buf1, buf2, DISPLAY_BUFF_SIZE);

    static lv_disp_drv_t disp_drv;
	lv_disp_drv_init(&disp_drv);
    disp_drv.flush_cb = disp_driver_flush;
	disp_drv.draw_buf = &disp_buf;
	disp_drv.hor_res = 240;
    disp_drv.ver_res = 240;
	disp_drv.antialiasing = 1;
    lv_disp_drv_register(&disp_drv);

    const esp_timer_create_args_t periodic_timer_args = {
        .callback = &lv_tick_task,
        .dispatch_method = ESP_TIMER_TASK,
        .name = "periodic_gui"
    };
    esp_timer_handle_t periodic_timer;
    ESP_ERROR_CHECK(esp_timer_create(&periodic_timer_args, &periodic_timer));
    ESP_ERROR_CHECK(esp_timer_start_periodic(periodic_timer, LV_TICK_PERIOD_MS * 1000));

    lv_widgets();
	set_lcd_bright(DEF_BRIGHT_LCD, 1000);

	while(1) 
	{
		vTaskDelay(10 / portTICK_PERIOD_MS);

		lv_task_handler();
		guiUserTask();
	}
}

pete-pjb, 2022-05-17:
Hi @kisvegabor,

cc @hard-cat 

Just to let you know I have updated to the latest master branch today and my splash screen images have stopped appearing on the screen when my systems boot up.  

Could this be a related problem?

Kind Regards,

Pete

kisvegabor, 2022-05-17:
AFAIK nothing critical was added in the past few days. Do you know where/how it halts?

hard-cat, 2022-05-17:
[image]

the problem was solved by disabling "Horizontal subpixel rendering".....

new library has worse performance?

pete-pjb, 2022-05-17:
Hi @kisvegabor,

cc @hard-cat

I hadn't read the post properly here sorry, I see @hard-cat is using v8.2 not latest master.

I am continuing to look in to this problem and I will open a new issue once I understand better what is happening.

Sorry for the noise on this thread.

Kind Regards,

Pete

kisvegabor, 2022-05-22:
Regarding the rendering issue please attach the converted font and "one label" example to reproduce the issue.

Regarding the performance, in which case do you see regression?

closed 2022-06-13 by lvgl-bot

Linzdigr, 2022-10-10:
I have the same issue on labels occurring here but with specific conditions:
- Fonts compiled from `npx github:lvgl/lv_font_conv` with option `--lcd` (horizontal subpixel rendering)
- Heavy interruptions on the rendering thread (i.e. short ISR routine triggered multiple times: >2k per second) on an nRF52840.