lvgl/lvgl #3361: Increase scale rendering accuracy in lv_meter

hard-cat, 2022-05-17:
Please tell me, is it possible to somehow improve the accuracy of drawing the scale?

[image]

    lv_obj_t * meter = lv_meter_create(backgrd);
    lv_obj_center(meter);
    lv_obj_set_size(meter, 230, 230);
    lv_obj_align(meter, LV_ALIGN_CENTER, 0, 0);

    /*Remove the circle from the middle*/
    lv_obj_remove_style(meter, NULL, LV_PART_INDICATOR);

    lv_meter_scale_t * scale = lv_meter_add_scale(meter);
    lv_meter_set_scale_ticks(meter, scale, 150, 2, 16, lv_palette_main(LV_PALETTE_GREY));

kisvegabor, 2022-05-17:
I made some experiments and could improve it. Please try thins patch:
```diff
diff --git a/src/extra/widgets/meter/lv_meter.c b/src/extra/widgets/meter/lv_meter.c
index 6f12a567..1ddd43fa 100644
--- a/src/extra/widgets/meter/lv_meter.c
+++ b/src/extra/widgets/meter/lv_meter.c
@@ -404,7 +404,7 @@ static void draw_ticks_and_labels(lv_obj_t * obj, lv_draw_ctx_t * draw_ctx, cons
     _LV_LL_READ_BACK(&meter->scale_ll, scale) {
         part_draw_dsc.sub_part_ptr = scale;
 
-        lv_coord_t r_out = r_edge + scale->r_mod;
+        lv_coord_t r_out = r_edge;
         lv_coord_t r_in_minor = r_out - scale->tick_length;
         lv_coord_t r_in_major = r_out - scale->tick_major_length;
 
@@ -473,29 +473,17 @@ static void draw_ticks_and_labels(lv_obj_t * obj, lv_draw_ctx_t * draw_ctx, cons
                 }
             }
 
-            /*`* 256` for extra precision*/
-            int32_t angle_upscale = ((i * scale->angle_range) << 8) / (scale->tick_cnt - 1);
-
-            int32_t angle_low = (angle_upscale >> 8);
-            int32_t angle_high = angle_low + 1;
-            int32_t angle_rem = angle_upscale & 0xFF;
-
-            /*Interpolate sine and cos*/
-            int32_t sin_low = lv_trigo_sin(angle_low + scale->rotation);
-            int32_t sin_high = lv_trigo_sin(angle_high + scale->rotation);
-            int32_t sin_mid = (sin_low * (256 - angle_rem) + sin_high * angle_rem) >> 8;
-
-            int32_t cos_low = lv_trigo_cos(angle_low + scale->rotation);
-            int32_t cos_high = lv_trigo_cos(angle_high + scale->rotation);
-            int32_t cos_mid = (cos_low * (256 - angle_rem) + cos_high * angle_rem) >> 8;
+            /*`* 10` for extra precision*/
+            int32_t angle_upscale = ((i * scale->angle_range) * 10) / (scale->tick_cnt - 1) +  + scale->rotation * 10;
 
             line_dsc.color = line_color;
             line_dsc.width = line_width;
             /*Use the interpolated angle to get the outer x and y coordinates.
              *Draw a little bit longer lines to be sure the mask will clip them correctly*/
             lv_point_t p_outer;
-            p_outer.x = (int32_t)(((int32_t)cos_mid * (r_out + line_width) + 127) >> (LV_TRIGO_SHIFT)) + p_center.x;
-            p_outer.y = (int32_t)(((int32_t)sin_mid * (r_out + line_width) + 127) >> (LV_TRIGO_SHIFT)) + p_center.y;
+            p_outer.x = p_center.x + r_out + LV_MAX(100, r_out);
+            p_outer.y = p_center.y;
+            lv_point_transform(&p_outer, angle_upscale, 256, &p_center);
 
             part_draw_dsc.p1 = &p_outer;
             part_draw_dsc.p1 = &p_center;
@@ -507,8 +495,9 @@ static void draw_ticks_and_labels(lv_obj_t * obj, lv_draw_ctx_t * draw_ctx, cons
                 lv_draw_mask_remove_id(outer_mask_id);
                 uint32_t r_text = r_in_major - scale->label_gap;
                 lv_point_t p;
-                p.x = (int32_t)((int32_t)((int32_t)cos_mid * r_text + 127) >> LV_TRIGO_SHIFT) + p_center.x;
-                p.y = (int32_t)((int32_t)((int32_t)sin_mid * r_text + 127) >> LV_TRIGO_SHIFT) + p_center.y;
+                p.x = p_center.x + r_text;
+                p.y = p_center.y;
+                lv_point_transform(&p, angle_upscale, 256, &p_center);
 
                 lv_draw_label_dsc_t label_dsc_tmp;
                 lv_memcpy(&label_dsc_tmp, &label_dsc, sizeof(label_dsc_tmp));
``` 

The performance might be slightly worse. I can fix it later if the changes look good.

Before:
[image]

After:
[image]

Still not perfect, but the current line drawer is a trade off between quality, speed and memory usage.

hard-cat, 2022-05-18:
@kisvegabor 

Thanks, I've applied these fixes.
Could you help me with the arrow too? I would like the arrow to start not from the center (1), but indented from the center (2).

[image]

hard-cat, 2022-05-18:
I tried to display the arrow as a picture, but it is not displayed correctly, it seems to me ...

lv_obj_t * meter_l = lv_meter_create(backgrd);
    lv_obj_center(meter_l);
    lv_obj_set_size(meter_l, 240, 240);
	lv_obj_align(meter_l, LV_ALIGN_CENTER, 0, 0);
	//Remove the circle from the middle
    lv_obj_remove_style(meter_l, NULL, LV_PART_INDICATOR);

	lv_meter_scale_t * scale_l = lv_meter_add_scale(meter_l);
	lv_meter_set_scale_range(meter_l, scale_l, 0, 40, 90, 135);	
        lv_meter_set_scale_ticks(meter_l, scale_l, 40, 2, 21, lv_color_hex(0x3D3D3D));

	lv_meter_indicator_t * indic = lv_meter_add_needle_line(meter_l, scale_l, 4, lv_palette_main(LV_PALETTE_GREY), -10);
 	lv_meter_set_indicator_end_value(meter_l, indic, 0);

	LV_IMG_DECLARE(arrow)
	lv_meter_indicator_t * indic_min = lv_meter_add_needle_img(meter_l, scale_l, &arrow, -70, -70);
	lv_meter_set_indicator_end_value(meter_l, indic_min, 0);

[image]

original picture of the arrow
[image]

kisvegabor, 2022-05-18:
> Thanks, I've applied these fixes.

And does it look acceptable on the device?

kisvegabor, 2022-05-18:
> Can you please tell me if this can be discussed here, or should I open a new topic?

Feel free to describe it here. Let's see the issues of the meter in one place :slightly_smiling_face:

hard-cat, 2022-05-18:
Fine :-)
Could you add a shift to the arrow like it's done in the arrow image function?

 * @param pivot_x       the X pivot point of the needle
 * @param pivot_y       the Y pivot point of the needle

kisvegabor, 2022-05-18:
Do you mean in `lv_arc_rotate_obj_to_angle`?

hard-cat, 2022-05-19:
Arrow display function

lv_meter_indicator_t * indic = lv_meter_add_needle_line(meter_l, scale_l, 4, lv_palette_main(LV_PALETTE_GREY), -10);

[image]
I would like to get the indent of the arrow from the center, as in Figure 2.

kisvegabor, 2022-05-19:
For this I suggest using a small line-like image and `lv_meter_add_needle_img()`.

hard-cat, 2022-05-19:
Yes, right now I'm using a picture, but there is a problem. The arrow has anti-aliasing and looks good. When using an image and rotating it, it looks grainy.

hard-cat, 2022-05-19:
red line - picture (lv_meter_add_needle_img)

[image]

kisvegabor, 2022-05-22:
Hm, rotation should  be anti aliased. 

Can you check [here](https://github.com/lvgl/lvgl/blob/master/src/extra/widgets/meter/lv_meter.c#L629) if `img_dsc.antialias == 1`?

hard-cat, 2022-05-23:
i checked, img_dsc.antialias is equal to 1
maybe something is wrong with my picture?
[red_line.txt](https://github.com/lvgl/lvgl/files/8751519/red_line.txt)

kisvegabor, 2022-05-23:
I've checked it and it was really anti-aliased but with a bad quality. It's because the SW rotate algorithm can do better job if there is at least 1px transparent "border" around the image. I added it manually to the C array and it look like this now:
[image]

hard-cat, 2022-05-23:
I added a transparent outline, now anti-aliasing works. thank!

hard-cat, 2022-05-23:
Can I have one more question about lv_meter?)

I want the last element (lv_meter_add_scale_lines) to be with a label. I added the following code to lv_meter.c

[image]

This works, but not for all values. For some values, 2 labels appear.
How to correctly determine the last tick in this cycle?

[image]

kisvegabor, 2022-05-26:
It seems working for me:
[image]

hard-cat, 2022-05-27:
this works with integers, you are using int32_t. When the number of ticks is, for example, 40, and the range is from 0 to 100, then there are problems with some values. If the range is equal to the number of indicator ticks, then there is no problem.

kisvegabor, 2022-05-28:
True. In this case I have better idea than what you already did. 

One optimization point I see though: you save the previous value of line in a `value_of_line_prev` variable, instead of always calculating the next value.

kisvegabor, 2022-06-14:
Are there any other remained here? If not I'll add the discussed improvement to master.

hard-cat, 2022-06-14:
hi,
I didn't do anything else, that's enough for my project

kisvegabor, 2022-06-19:
I pushed the fix to improve the line drawing precision.

commit 0255c6dd (2022-06-19): fix(meter): improve the precision of tick line drawing

fixes #3361
commit c241e830 (2022-09-20): fix(meter): improve the precision of tick line drawing

fixes #3361
closed 2022-06-19 by kisvegabor