lvgl/lvgl #3362: rendering error in objects with rounded corners

moinmoin-sh, 2022-05-17:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

When creating a button (or panel) with rounded corners adn the background of the button is set by using opacity, the aread within the avtive radius on the top and the bottom of the button is redered darker than the background of the remaining area. Please see attached screen shot for better explanation

### To Reproduce

```c
    lv_obj_t* act = lv_scr_act();

    lv_obj_t* screen = lv_obj_create(act);
    lv_obj_set_size(screen, 320, 480);
    lv_obj_set_style_bg_color(screen, lv_color_make(50, 0, 50), LV_PART_MAIN);

    lv_obj_t* btn = lv_btn_create(screen);
    lv_obj_set_size(btn, 100, 100);
    lv_obj_set_style_radius(btn, 10, LV_PART_MAIN);
    lv_obj_center(btn);
    lv_obj_set_style_bg_color(btn, lv_color_make(255, 255, 255), LV_PART_MAIN);
    lv_obj_set_style_bg_opa(btn, 50, LV_PART_MAIN);

```

### Expected behavior

The whole area should be colored consistent.

### Screenshots or video

[image]

kisvegabor, 2022-05-18:
Thanks for the report. I fixed it. 

I suggest setting `LV_COLOR_MIX_ROUND_OFS 0` to gain some performance.

closed 2022-05-18 by kisvegabor

moinmoin-sh, 2022-05-18:
Thanks for the quick fix !

moinmoin-sh, 2022-05-18:
Sorry for coming back, but the issue doesn't seam to be fully fixed. The mentioned upper and lower area are still darker.

[image]
[image]
[image]

moinmoin-sh, 2022-05-18:
The screenshots are taken from the PC.
Changed extension to .txt to be accepted here.

[lv_conf.txt](https://github.com/lvgl/lvgl/files/8720834/lv_conf.txt)

kisvegabor, 2022-05-18:
Thanks, I improved it.

moinmoin-sh, 2022-05-19:
Looks good now!
Thanks for fixing it!

kisvegabor, 2022-05-19:
Great, thanks!

kisvegabor, 2023-03-31:
It think it'd be easier to read like this:

```c
#if LV_COLOR_DEPTH == 16 && LV_COLOR_MIX_ROUND_OFS == 0
#if LV_COLOR_16_SWAP == 1
    c1.full = c1.full << 8 | c1.full >> 8;  // swap c1
    c2.full = c2.full << 8 | c2.full >> 8;  // swap c2
#endif

...
```

If you agree and it works well, could you open a pull request for the `release/v8.3` branch?

moinmoin-sh, 2023-03-31:
Will do.
Thanks for looking into this!

commit 523062b9 (2022-05-18): fix(color): color mix rounding error

fixes: #3362
commit 9d4ab8a0 (2022-05-18): fix(color): compensate rounding error during blending

fixes #3362"
commit 42d9c07e (2022-05-18): fix(color): compensate rounding error during blending

fixes #3362