lvgl/lvgl #3364: In arduino lv_obj_draw_part_dsc_t * dsc = lv_event_get_param(e); causes invalid conversion

elgerg, 2022-05-18:
### Describe the bug
When using the following line in Arduino:
lv_obj_draw_part_dsc_t * dsc = lv_event_get_param(e);
You get:
error: invalid conversion from 'void*' to 'lv_obj_draw_part_dsc_t*' [-fpermissive]
     lv_obj_draw_part_dsc_t * dsc = lv_event_get_param(e);

Casting does fix this issue:
lv_obj_draw_part_dsc_t * dsc = (lv_obj_draw_part_dsc_t *)lv_event_get_param(e);

It seems C++ is more picky than C (where it works fine).

### To Reproduce
As description

### Expected behavior
Code would compile

### Screenshots or video
[image]

elgerg, 2022-05-18:
Is it worth adding something in to the docs to explain (maybe in this case only as it's the only time I have seen this problem) how to overcome this "known" Arduino issue?

It took me a little while to get to this:

https://github.com/lvgl/lvgl/issues/1113

Which gave me a hint on how to overcome the problem.

Thanks!

embeddedt, 2022-05-18:
> can the function be changed from void * to lv_obj_draw_part_dsc_t * ?

I have just remembered that there is a wrapper (`lv_event_get_draw_part_dsc`) that also checks the event type to make sure you aren't casting incorrectly (since the type of the parameter depends on the event). Sorry for not remembering that before; that should solve your problem.

elgerg, 2022-05-18:
Oh, that is good news.
Should that go into the docs instead of whats there?

embeddedt, 2022-05-18:
Yes, I've fixed it.

commit 71d535de (2022-05-18): fix(examples) use type-safe function for retrieving event param

Fixes #3364