lvgl/lvgl #3366: lv_obj_t * tabView = lv_event_get_target(event) doesnt seem to work

elgerg, 2022-05-18:
### Describe the bug
When using:
lv_obj_t * tabView = lv_event_get_target(event);
then:
lv_obj_set_style_pad_bottom(parent, 7, LV_PART_MAIN);

I get a tab number of 16380

To workaround this I have had to create a global variable for tv and use that in the callback.

Is this intended behaviour?

### To Reproduce
As above (on Arduino)

### Expected behavior
I would expect 0 if I am going to tab 0 and 1 if 1 etc..

### Screenshots or video

kisvegabor, 2022-05-18:
I guess the event handler is attached to the tabview itself (and not the header or content).

If so you should use `lv_event_get_current_target()` as the event is sent by the header which "bubbles up" to the parent tabview, The terminology was copied from JaveScript's API. 

My personal opinion is that renaming 
- `lv_event_get_current_target` -> l`v_event_get_target`
- `lv_event_get_target` -> `lv_event_get_original_target`

would be more intuitive but we have decided to stick with the JS names. 

cc @embeddedt

embeddedt, 2022-06-02:
@elgerg Were you able to solve this?

elgerg, 2022-06-02:
I havent tried the fix yet, I just left the tv as a global variable.

I might try again shortly..

closed 2022-06-24 by lvgl-bot