lvgl/lvgl #3367: V8.2.0 lv_task_t unable to use

gellary, 2022-05-19:
### Perform all steps below and tick them with [x]
- [ ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [√] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
i can't use lv_task.h in lvgl  8.2.0
 lv_task.h and  lv_task.c is missing in misc folder

### IDE
running on arduino IDE 

### Develop board
 esp32_c3

### To Reproduce
Void task_cb(lv_task_t * t)
{
  uint32_t * user_data = (uint32_t*)t->user_data;
  printf("my_task called with user data: %d\n", *user_data);
}
 
### ERROR MESSAGE
 
Aurora_screen:319:14: error: variable or field 'task_cb' declared void
 void task_cb(lv_task_t * t)
              ^~~~~~~~~
Aurora_screen:319:14: error: 'lv_task_t' was not declared in this scope
D:\Desktop\Aurora_screen\Aurora_screen.ino:319:14: note: suggested alternative: 'lv_bar_t'
 void task_cb(lv_task_t * t)
              ^~~~~~~~~

gellary, 2022-05-19:
oh ~
nice thank you so much

pete-pjb, 2022-05-19:
No problem, you're welcome.

closed 2022-05-19 by pete-pjb