lvgl/lvgl #3368: Request for pre-multiplied alpha ARGB format in the LVGL online image converter

StefanBabatie, 2022-05-19:
**Introduction**
The LVGL online image converter tool does not currently support converting a given image into a C array using the pre-multiplied ARGB format. This feature would be necessary because some NXP i.MX RT boards need this format in order to work properly.

This is because some of these platforms do support hardware multiplication while others don't. If one does support this feature, using non-pre-multiplied ARGB images would be no problem. On the other hand, using such an image on a platform without hardware premultiplication results in unexpected image blending results, so the pre-multiplication in this case has to be done in the LVGL image converter.

**Example** 
For showing how the problem manifests, let's take into consideration two platforms:
- the NXP I.MX RT1170, featuring a 2DGPU which supports hardware pre-multiplication;
- the NXP I.MX RT500, featuring a 2DGPU which does not support hardware pre-multiplication.

The test with which we can showcase this issue is: use the 2DGPU to blit an ARGB image of a cogwheel onto an orange background. The results are as follows:

Using non-pre-multiplied ARGB image on RT500:
[image]

The lack of hardware pre-multiplication on the RT500 translates into the fact that the transparent pixels on the outer edge of the cog will not be multiplied with their alpha and instead will be shown as opaque white pixels.

Using the CPU or a pre-multiplied ARGB image on RT500:
[image]

The presence of hardware pre-multiplication on the RT1170 ensures that the outer edge pixels of the cog are multiplied with their respective alphas and do not show in the final result, resulting in an expected blend of the cog and the background.

A better understanding of the two situations can be achieved with the following formulas for the Porter Duff SRC OVER operation:

Porter-Duff SRC OVER without hardware premultiplication (as seen on the RT500)
`Result_colour = source_colour + (1 - source_alpha) * destination_colour`

Porter-Duff SRC OVER featuring hardware premultiplication (as seen on the RT1170)
`Result_colour = source_colour  * source_alpha + (1 - source_alpha) * destination_colour * destination_alpha`

**Suggested solution**
As such, in order to reproduce the formula used by the RT1170 and to achieve the correct results on screen irrespective of hardware pre-multiplication support,  any color used in ARGB images in the RT500 or other board that lacks hardware pre-multiplication should be replaced in the formula by the product between the said color and the alpha of the pixel in question. 
As this can not always be done by hardware itself and doing it by software (in the LVGL code, whenever the image is blitted) is not efficient, we need to do it in the image of the cog itself: the asset itself should already be stored in a pre-multiplied ARGB format if it’s intended to be used on a board that lacks hardware pre-multiplication. 

This is why I’m asking about adding the pre-multiplied alpha option into the LVGL image converter. It’s not a question of performance but rather of adapting the range of asset formats to the range of limitations of the different NXP platforms on which LVGL will run.  Currently, the customers should have no problems with creating assets for the NXP boards that do feature hardware pre-multiplication using the ‘True color with alpha’ option in the converter. In the same way, I think that a ‘True color with pre-multiplied alpha’ option should be added so they can also easily create assets for the NXP boards that lack this feature.

kisvegabor, 2022-05-20:
I see the problem, and thanks for the detailed explanation.

I think there is no problem with supporting it in the image converter (@embeddedt ?)

However it should be supported in SW rendering too. It can result in somewhat higher performance as 1 multiplication is saved  during blending.

embeddedt, 2022-05-20:
I think we can support this in the converter fairly easily, but to avoid doubling the number of image formats, we need a bit in the header that indicates that alpha has been premultiplied. (Another option would be to always premultiply the alpha, however, this breaks backwards compatibility.)

kisvegabor, 2022-05-22:
> I think we can support this in the converter fairly easily, but to avoid doubling the number of image formats, we need a bit in the header that indicates that alpha has been premultiplied. 

It doesn't double the number of formats. Only `LV_IMG_CF_TRUE_COLOR_ALPHA` has a pre-multiplied version as this the only format where alpha and colors are present together.

embeddedt, 2022-05-22:
We don't need this feature for RGB565A8?

kisvegabor, 2022-05-22:
I forgot about this new format, but yes, it required there too. And if we will add `ARGB4444`, it will be needed there too. And who knows what will be in the future. So you are right, probably a flag would be more appropriate. To be more consequent I'm also thinking about renaming `LV_IMG_CF_TRUE_COLOR*` formats as:
- `LV_IMG_CF_RGB`
- `LV_IMG_CF_RGB_CHROMA_KEYED`
- `LV_IMG_CF_ARGB`

They would mean `232`, `565`, `888` based on the `LV_COLOR_DEPTH` configuration.

embeddedt, 2022-06-06:
Not stale, I have been doing some refactoring to make this cleaner to implement.

kisvegabor, 2023-10-26:
The original request can be solved by a custom post process callback: https://docs.lvgl.io/master/overview/img.html#image-post-processing

closed 2023-10-26 by kisvegabor