lvgl/lvgl #3369: Support for 12-bit RGB444

acumartini, 2022-05-19:
I have no idea how difficult this is, but could be a big benefit to people running smaller displays that do not support 8-bit RBG332 but still want to speed up their frame rate. For example the ST7789V driver, which only supports 12/16/32-bit RGB.

acumartini, 2022-05-19:
I can take a stab at this with some pointers 👍

kisvegabor, 2022-05-22:
Are you sure 12 bit would speed up frame rate? If the pixels are tightly packed (i.e. on 13th bit the next pixel starts) it will be slow do to the complicated addressing of pixels and color channels. If the pixels are byte aligned it will be like 16 bit.

Or the speed up could come from less data via SPI (assuming tight packing here too)?

acumartini, 2022-05-23:
Yeah I think there would be some overhead due to complicated addressing although I do not understand the problem as well as you do. But the primary speed increase would be from sending less data over SPI, assuming you could compensate for addressing costs with frame buffers and/or DMA.

closed 2022-06-15 by lvgl-bot