lvgl/lvgl #3373: Usage of anonymous enums

corporate-dev, 2022-05-20:
### Introduce the problem

LVGL uses anonymous enums in several places which makes it impossible to identify which enum its values belong to.
I assume this might be reasonable to do in C but when trying to wrap the code in C++ it would be really helpful to collectively identify the defined values (e.g. to implement wrappers with strong type guarantees).

Is the reason for using anonymous enums to make sure that no variable of an enum is created and we use a well defined type like uint32_t to store values?

My suggestion is to to name the anonymous enums and maybe do so by using "internal names", i.e. underscore prefixed names. 

What is your stance on this?

Btw.: In C++(!) not naming enums is considered unfavorable. (cmp. [core guidelines](http://isocpp.github.io/CppCoreGuidelines/CppCoreGuidelines#Renum-unnamed))

### Examples and cases

[anonymous enums in lvgl](https://github.com/lvgl/lvgl/blob/7bb09e358026aff3d55d881237624baac77db890/src/core/lv_obj.h#L89)
(The enums are anonymous while they are processed and stored in lv_obj_flag_t)

```c
enum {
    LV_OBJ_FLAG_HIDDEN          = (1L << 0),  /**< Make the object hidden. (Like it wasn't there at all)*/
    LV_OBJ_FLAG_CLICKABLE       = (1L << 1),  /**< Make the object clickable by the input devices*/
    ...
    LV_OBJ_FLAG_USER_4          = (1L << 30), /**< Custom flag, free to use by user*/
};

typedef uint32_t lv_obj_flag_t;
```

could be turned into 

```c
typedef enum {
    LV_OBJ_FLAG_HIDDEN          = (1L << 0),  /**< Make the object hidden. (Like it wasn't there at all)*/
    LV_OBJ_FLAG_CLICKABLE       = (1L << 1),  /**< Make the object clickable by the input devices*/
    ...
    LV_OBJ_FLAG_USER_4          = (1L << 30), /**< Custom flag, free to use by user*/
} _lv_obj_flag_enum_t;

typedef uint32_t lv_obj_flag_t;
```

Of course I see that this duplicates names and might confuse users. Perhaps other ideas exist to solve this.

### Suggested solution

Naming all API-exposed enums in the LVGL codebase.

kisvegabor, 2022-05-22:
We introduced this practice to use allow using enums in struct as bitfields with a meaningful name. E.g. 
```c
typedef struct {
    [...]
    lv_scrollbar_mode_t scrollbar_mode : 2; /**< How to display scrollbars*/
    lv_scroll_snap_t scroll_snap_x : 2;     /**< Where to align the snappable children horizontally*/
    lv_scroll_snap_t scroll_snap_y : 2;     /**< Where to align the snappable children vertically*/
    [...]
} _lv_obj_spec_attr_t;
```

But I also some problems with this approach:
- Enum names are no shown during debug, only the integer values
- Unnamed enums followed by a typed is simply odd

Obviously the best would be to use normal enums as bit fields but it seems not that safe. See [here ](https://stackoverflow.com/a/26728946/1999969).

We could do this:
```c
/*Named enum*/
enum {
    LV_SCROLLBAR_MODE_OFF,      /**< Never show scrollbars*/
    LV_SCROLLBAR_MODE_ON,       /**< Always show scrollbars*/
    LV_SCROLLBAR_MODE_ACTIVE,   /**< Show scroll bars when object is being scrolled*/
    LV_SCROLLBAR_MODE_AUTO,     /**< Show scroll bars when the content is large enough to be scrolled*/
} lv_scrollbar_mode_t;

[...]

typedef struct {
    [...]
    uint8_t scrollbar_mode : 2; /**< How to display scrollbars. Elements of lv_scrollbar_mode_t*/
    uint8_t scroll_snap_x : 2;     /**< Where to align the snappable children horizontally, Elements of lv_scroll_snap_t*/
    uint8_t scroll_snap_y : 2;     /**< Where to align the snappable children vertically. Elements of lv_scroll_snap_t*/
    [...]
} _lv_obj_spec_attr_t;
```

But it still wouldn't give debug information. So I have no good idea... :slightly_frowning_face:

closed 2022-06-13 by lvgl-bot

embeddedt, 2022-06-13:
The only technically correct solution is to use enum types for the fields and remove the bitfields, but I think this would lead to an astronomical increase in memory usage if it were applied across the entire library. I think for now, the current solution is the best we have, barring some new development in C11 I'm not aware of.

kisvegabor, 2022-06-15:
We can also review where the current "hack" is required. Probably the memory impacts is not relevant in all cases.