lvgl/lvgl #3376: Slow lv_obj_create

AfflatusX, 2022-05-23:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [N/A] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
I noticed when creating a batch of LV objects, it takes significant amount of CPU time. My current example creates about 60 lv_line objects, which takes about 15ms on my hardware.

### To Reproduce
Simply create 60 line objects using lv_line_create in a loop, do a LV_LOG_INFO before and after, check the time spent.

### Expected behavior
Not sure what to expect here - but want to understand what contributes to the CPU time during object creation? Is it mainly memory speed/bandwidth because of all the malloc happening? Assuming we want to draw something at 60FPS, that gives us around 16ms per frame - if creating objects is quite slow, it makes hitting the target frame time quite hard.

AfflatusX, 2022-05-23:
Btw I already disabled all the trace/assert in lv_conf.

kisvegabor, 2022-05-23:
A few questions and ideas:
- How fast your MCU is?
- Have you enabled -O2 or -O3 optimization?
- For lines do you really need 60 widgets? Can't you use a larger "point_array"?
- You can try using the compilers `malloc`. See [here](https://github.com/lvgl/lvgl/blob/master/lv_conf_template.h#L49)
- Can you run a profiler to see where the bottleneck?
- What speed would you prefer? Now it's  0.25 ms/widgets which seems ok to me.

AfflatusX, 2022-05-23:
> A few questions and ideas:
> 
> * How fast your MCU is?
It's an M55 MCU @ 200Mhz
> * Have you enabled -O2 or -O3 optimization?
I need to double-check on this, but do we expect perf impact on obj creation with different optimization level? Reading the source code, it feels like the creation call is really just allocating memory and initializing them.
> * For lines do you really need 60 widgets? Can't you use a larger "point_array"?
We don't - we can definitely do something else to avoid this cost. However, I'm trying to understand the perf expectation of object creation in general so that we can either optimize it (including change of hw) or factor that time in when designing UI.
> * You can try using the compilers `malloc`. See [here](https://github.com/lvgl/lvgl/blob/master/lv_conf_template.h#L49)
I've already tried that actually, both yields similar timing, hence I suspect memory bandwidth being the cause. We use TLSF in other places and itself doesn't seem to cause more latency.
> * Can you run a profiler to see where the bottleneck?
Yeah, I'll try that - this issue is only noticeable on actual HW, so I need to figure out how to JTAG it...
> * What speed would you prefer? Now it's  0.25 ms/widgets which seems ok to me.
I think this is more of an extreme case - but I also noticed on avg it takes 0.25-0.5ms to create a widget. While it's not too bad, but we do need to factor in the time taken when new components are created so that it does not slow down the 16ms rendering window. It's more about understanding what is contributing to the latency.

For example, if it's about memory allocation, then maybe if a UI structure is known, we can allocate one big chunk for it and make it a linear memory block for its children?

Or if it's because of the initialization (memset) we can maybe optimize it by pre-initializing the chunk of memory allocated.

kisvegabor, 2022-05-23:
>> Have you enabled -O2 or -O3 optimization?

> I need to double-check on this, but do we expect perf impact on obj creation with different optimization level? Reading the source code, it feels like the creation call is really just allocating memory and initializing them.

Some events are also sent meanwhile, so optimization can make some difference.

> For example, if it's about memory allocation, then maybe if a UI structure is known, we can allocate one big chunk for it and make it a linear memory block for its children?

I'd prefer making it work normally without specific hacks :slightly_smiling_face:

closed 2022-06-14 by lvgl-bot