lvgl/lvgl #3377: performance problem with LVGL and STM32F746 (lv_task_handler)?

epikao, 2022-05-24:
Hello

I have the following simple code running below. I’m measuring how regularly the LED toggles.
If I exclude lv_task_handler(), it runs clean (the LED toggles all ms). 
If lv_task_handler is enabled there are about 4mS interruptions all 300ms, see image1 and 2?

And without lv_task_handler see image3 and 4.

I use a STM32F746 with LVGL 8.2.
Is this all normal? Or do I have a performance/setting problem ??
It is very urgent for me, I’am glad for any hint…

Thank you

```
while (1){
    	////////////////// TESTCODE ///////////////////////////

        if( ( millis()-lastMillis2 ) > 4){
        	lv_task_handler();
        	lastMillis2 = millis();
        }

    	if( ( millis()-lastMillis1 ) >= 1){
    		HAL_GPIO_TogglePin(GPIOI, GPIO_PIN_8); //LED 101
    		lastMillis1 = millis();
    	}
}
```

**Image1 and 2:**

[image]
[image]

**Image3 and 4:**

[image]
[image]

kisvegabor, 2022-05-24:
LVGL's internal tasks are running in every ~30 ms or so. I hav no idea what could happen in each 300 ms. 

This with the [other performance issue](https://github.com/lvgl/lvgl/issues/3328) suggests hat there is a problem with your HW configuration. Are you sure that the MCU is running at ~200 MHz?

pete-pjb, 2022-05-24:
Hi @epikao ,

Just a thought do you have any lv_timers configured for this period?

Cheers,

Pete

kisvegabor, 2022-05-24:
And how/where do you call `lv_tick_inc`? Or do you use `LV_TICK_CUSTOM`?

epikao, 2022-05-24:
> And how/where do you call `lv_tick_inc`? Or do you use `LV_TICK_CUSTOM`?

here (stm32f7xx_it.c):
void SysTick_Handler(void)
{
  HAL_IncTick();
  HAL_SYSTICK_IRQHandler();
  lv_tick_inc(1);
  counter++;
}

> This with the [other performance issue](https://github.com/lvgl/lvgl/issues/3328) suggests hat there is a problem with your HW configuration. Are you sure that the MCU is running at ~200 MHz?

Not sure how to check this 100% but below the scope of the toggle LED with following code:
```
while (1){
        //HAL_Delay(500);
    	HAL_GPIO_TogglePin(GPIOI, GPIO_PIN_8); //LED 101
    	////////////////// TESTCODE ///////////////////////////
        if( ( millis()-lastMillis2 ) > 4){
        	lv_task_handler();
        	lastMillis2 = millis();
        }
    	if( ( millis()-lastMillis1 ) >= 1){

    		//HAL_GPIO_TogglePin(GPIOI, GPIO_PIN_8); //LED 101
    		lastMillis1 = millis();
    	}
}
```

Clock config:
```
static void SystemClock_Config(void)
{
    RCC_ClkInitTypeDef RCC_ClkInitStruct;
    RCC_OscInitTypeDef RCC_OscInitStruct;
    RCC_PeriphCLKInitTypeDef PeriphClkInitStruct = {0}; //i2c, uart etc.

    HAL_PWR_EnableBkUpAccess(); //
    /* Enable Power Control clock */
    __HAL_RCC_PWR_CLK_ENABLE();

    /* The voltage scaling allows optimizing the power consumption when the device is
     clocked below the maximum system frequency, to update the voltage scaling value
     regarding system frequency refer to product datasheet.  */
    //__HAL_PWR_VOLTAGESCALING_CONFIG(PWR_REGULATOR_VOLTAGE_SCALE1);

    /*##-1- System Clock Configuration #########################################*/
    /* Enable HSE Oscillator and activate PLL with HSE as source */

/*
    RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSE;
    RCC_OscInitStruct.HSEState = RCC_HSE_ON;
    RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
    RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSE;
    RCC_OscInitStruct.PLL.PLLM = 25;
    RCC_OscInitStruct.PLL.PLLN = 400;
    RCC_OscInitStruct.PLL.PLLP = RCC_PLLP_DIV2;
    RCC_OscInitStruct.PLL.PLLQ = 8;
*/

    //RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSI|RCC_OSCILLATORTYPE_HSE;
    RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSE;
    RCC_OscInitStruct.HSEState = RCC_HSE_ON;
    //RCC_OscInitStruct.HSIState = RCC_HSI_ON;
    RCC_OscInitStruct.HSICalibrationValue = RCC_HSICALIBRATION_DEFAULT;
    RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
    RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSE;
    RCC_OscInitStruct.PLL.PLLM = 25;
    RCC_OscInitStruct.PLL.PLLN = 432; //432
    RCC_OscInitStruct.PLL.PLLP = RCC_PLLP_DIV2;
    RCC_OscInitStruct.PLL.PLLQ = 9;              //USB PLLQ 432 / 9 = 48
    HAL_RCC_OscConfig(&RCC_OscInitStruct);

    /* Activate the Over-Drive mode */
    HAL_PWREx_EnableOverDrive();

    /* Select PLL as system clock source and configure the HCLK, PCLK1 and PCLK2
     clocks dividers */
    RCC_ClkInitStruct.ClockType = (RCC_CLOCKTYPE_SYSCLK | RCC_CLOCKTYPE_HCLK | RCC_CLOCKTYPE_PCLK1 | RCC_CLOCKTYPE_PCLK2);
    RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_PLLCLK;
    RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
    RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV4;
    RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV2;
    HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_6);

    //HAL_RCC_MCOConfig(RCC_MCO1, RCC_MCO1SOURCE_HSI, RCC_MCODIV_1);

    PeriphClkInitStruct.PeriphClockSelection = RCC_PERIPHCLK_I2C1 | RCC_PERIPHCLK_I2C3 | RCC_PERIPHCLK_USART3 | RCC_PERIPHCLK_USART6 | RCC_PERIPHCLK_CLK48;
    PeriphClkInitStruct.I2c3ClockSelection = RCC_I2C3CLKSOURCE_PCLK1;
    PeriphClkInitStruct.I2c1ClockSelection = RCC_I2C1CLKSOURCE_PCLK1;
    PeriphClkInitStruct.Usart3ClockSelection = RCC_USART3CLKSOURCE_PCLK1;
    PeriphClkInitStruct.Usart6ClockSelection = RCC_USART6CLKSOURCE_PCLK2;

    /* Select PLLSAI output as USB clock source 384 /8 */
    //PeriphClkInitStruct.PeriphClockSelection = RCC_PERIPHCLK_CLK48;
    //PeriphClkInitStruct.Clk48ClockSelection = RCC_CLK48SOURCE_PLL;
    if (HAL_RCCEx_PeriphCLKConfig(&PeriphClkInitStruct) != HAL_OK)
    {
    }

}
```

[image]

> Just a thought do you have any lv_timers configured for this period?

Should not, I use the port: lv_port_stm32f746_disco

**This is my init:**

```
int main(void)
{
    HAL_Init();

    /* Configure the system clock to 216 MHz */
    SystemClock_Config();

    /* Enable I-Cache */
    SCB_EnableICache();

    /* Enable D-Cache */
    SCB_EnableDCache();

    MX_GPIO_Init();

    lv_init();
    tft_init();

    static lv_indev_drv_t indev_drv;
    lv_indev_drv_init( &indev_drv );
    indev_drv.type = LV_INDEV_TYPE_POINTER;
    indev_drv.read_cb = my_touchpad_read;
    lv_indev_drv_register( &indev_drv );
```

pete-pjb, 2022-05-24:
Hi @epikao ,

Once ```lvgl_task_handler()``` is running it will be calling the ```my_touchpad_read()``` and the ```flush_cb()``` function periodically based on your settings in "lv_conf.h".  Have you done any checks in those functions to see if anything is holding up proceedings there?

Hope that helps...

Kind Regards,

Pete

epikao, 2022-05-24:
> Once `lvgl_task_handler()` is running it will be calling the `my_touchpad_read()` and the `flush_cb()` functions periodically based on the timings set in "lv_conf.h". Have you done any checks in those functions to see if anything is holding up proceedings there?

Thank you for your help.

my_touchpad_read() is not called (commented out),

Following is my flush function. If I comment out the content of this flush function, the program never enters the while loop.
So it is difficult to check if there is something wrong...

```
static void ex_disp_flush(lv_disp_drv_t *drv, const lv_area_t *area, lv_color_t * color_p)
{
	int32_t x1 = area->x1;
	int32_t x2 = area->x2;
	int32_t y1 = area->y1;
	int32_t y2 = area->y2;
    /*Return if the area is out the screen*/

    if(x2 < 0) return;
    if(y2 < 0) return;
    if(x1 > TFT_HOR_RES - 1) return;
    if(y1 > TFT_VER_RES - 1) return;

    /*Truncate the area to the screen*/
    int32_t act_x1 = x1 < 0 ? 0 : x1;
    int32_t act_y1 = y1 < 0 ? 0 : y1;
    int32_t act_x2 = x2 > TFT_HOR_RES - 1 ? TFT_HOR_RES - 1 : x2;
    int32_t act_y2 = y2 > TFT_VER_RES - 1 ? TFT_VER_RES - 1 : y2;

    x1_flush = act_x1;
    y1_flush = act_y1;
    x2_flush = act_x2;
    y2_fill = act_y2;
    y_fill_act = act_y1;
    buf_to_flush = color_p;

	SCB_CleanInvalidateDCache();
	SCB_InvalidateICache();
    /*##-7- Start the DMA transfer using the interrupt mode #*/
    /* Configure the source, destination and buffer size DMA fields and Start DMA Stream transfer */
    /* Enable All the DMA interrupts */
    HAL_StatusTypeDef err;
    uint32_t length = (x2_flush - x1_flush + 1);
#if LV_COLOR_DEPTH == 24 || LV_COLOR_DEPTH == 32
    length *= 2; /* STM32 DMA uses 16-bit chunks so multiply by 2 for 32-bit color */
#endif
    err = HAL_DMA_Start_IT(&DmaHandle,(uint32_t)buf_to_flush, (uint32_t)&my_fb[y_fill_act * TFT_HOR_RES + x1_flush],
             length);
    if(err != HAL_OK)
    {
        while(1);	/*Halt on error*/
    }
}
```

embeddedt, 2022-05-24:
You'll need to call `lv_disp_flush_ready` within the function if you comment everything else out, so LVGL thinks the display has been flushed.

pete-pjb, 2022-05-24:
Hi @epikao ,

Looking at your flush function...

I am no expert on the STM32F746 MCU but in general as far as my knowledge extends, it shouldn't be necessary to invalidate the I-cache for DMA purposes, so it may be worth commenting out the call to ```SCB_InvalidateICache();``` just to see if this is causing something strange to happen.

Kind Regards,

Pete

epikao, 2022-05-24:
> You'll need to call `lv_disp_flush_ready` within the function if you comment everything else out, so LVGL thinks the display has been flushed.

looks the flush function is not responsible for this interruption every 300ms... :-/ 

```
static void ex_disp_flush(lv_disp_drv_t *drv, const lv_area_t *area, lv_color_t * color_p)
{
	lv_disp_flush_ready(drv);
}
```

pete-pjb, 2022-05-24:
Hi @epikao ,

If you are using the STM32CubeIDE are you able to do some code profiling so you can see what is being executed?  I had a quick look on line and it looks like there should be a set of views available when you are in the debug perspective prefixed with SWV where you can see various function execution times etc.  Is this something that could be used to help track this down?  Sorry for the noise if this is an irrelevant comment 🙂 

Hope that is helpful.

Kind Regards,

Pete

GorgonMeducer, 2022-05-24:
What's the value of macro `LV_DISP_DEF_REFR_PERIOD` in your `lv_conf.h`?

How about replacing 
```c
        if( ( millis()-lastMillis2 ) > 4){
        	lv_task_handler();
        	lastMillis2 = millis();
        }
```
 with
```c
    lv_timer_handler_run_in_period(4);
```

epikao, 2022-05-25:
Thank you. I have now updated from stm IDE 1.3 to IDE 1.9.
After that i got some link errors like multiple definition regarding lv_meter_indicator_t * and lv_chart_series_t *. 
I have corrected that. (No idea why I did not get this error with IDE 1.3)
Now interestingly there is no interruption every 300ms....
So this problem looks solved...

> My measurement [here](https://github.com/lvgl/lvgl/issues/3328#issuecomment-1136906646) indicates that the bug is really in your HW config as it's ~1000 times faster for me.

Thank you. I get 29ms here.... 
I still have the problem with lv_meter/lv_arc. Next I will recopy the whole lvgl-library.

kisvegabor, 2022-05-25:
For the sake of simplicity let's track this in #3328.