lvgl/lvgl #3379: Should we add scenes in benchmark for RGB565-A8?

GorgonMeducer, 2022-05-24:
### Introduce the problem
We have introduced a new colour format RGB565-A8. The performance improvement for 16bit colours comes for free, as long as the resource images are converted into c arrays using this new format. 

I believe this colour format should be suggested as the recommended colour format when the target LCD is using RGB565 for all new applications. 

We can already see a lot of 2D GPU drivers are updated to take advantage of this new colour format, e.g. NXP PXP and arm-2d. 

The problem is that
1. in the benchmark, we don't have scenes to cover this new colour format. We definitely need scenes to cover this new colour format for helping people add new GPU and profiling.
2. How to handle the relationship between ARGB8565 and RGB565-A8 in the same benchmark? Exclusive from each other? or Co-existing when calculating the Weighted FPS?

### Examples and cases

### Suggested solution

kisvegabor, 2022-05-26:
> When LV_COLOR_DEPTH is 16, use a macro, for example, LV_BM_USE_RGB565A8 to switch between ARGB8565 and RGB565-A8 for all ARGB related scenes.

It makes sense. 

> I believe this colour format should be suggested as the recommended colour format when the target LCD is using RGB565 for all new applications.

I can imagine these places to mention this new format:
- [Color formats](https://docs.lvgl.io/master/overview/image.html#color-formats)
- [Image widget](https://docs.lvgl.io/master/widgets/core/img.html)
- [Image converter](https://lvgl.io/tools/imageconverter)

Now, when someone selects "True color with alpha" format in the image converter he/she gets all color depth options selected by `#if LV_COLOR_DEPTH == ...`. What about having an option in the converter to include the `RGB565-A8` format by default, instead of the `ARGB8565`? 

cc @embeddedt

GorgonMeducer, 2022-05-26:
@kisvegabor since we have no scene in the benchmark for RGB565-A8, I don't have such data also.
But as you can imagine, avoiding repeating iterations for decoding, the CPU time used by decoding can be removed, this is the source of performance improvement.

GorgonMeducer, 2022-05-27:
@embeddedt check cf?

embeddedt, 2022-05-27:
Right now `LV_IMG_CF_TRUE_COLOR_ALPHA` means that the image will either be RGB332, ARGB8565, or ARGB8888, depending on `LV_COLOR_DEPTH`. We would need to introduce a new cf value in order to make this change.

GorgonMeducer, 2022-05-27:
@embeddedt we already have such new cf.

kisvegabor, 2022-05-27:
> The converter can be adjusted easily, but how does the library differentiate between older images using ARGB8565 and newer ones using RGB565-A8?

If RGB565-A8 is used for 16 bit the converted image header could look like this:
```c
const lv_img_dsc_t some_image = {
  .header.always_zero = 0,
  .header.w = 100,
  .header.h = 100,
  .data_size = 10000 * LV_IMG_PX_SIZE_ALPHA_BYTE,
#if LV_COLOR_DEPTH == 16
  .header.cf = LV_IMG_CF_RGB565A8,
#else
  .header.cf = LV_IMG_CF_TRUE_COLOR_ALPHA,
#endif

  .data = some_image,
};
```

embeddedt, 2022-05-27:
Ah, now I understand. That's a clever approach. I will look into it!

I suppose that since the older 8.x versions are still supported, there needs to be an option to disable this so that images converted today will still work in those versions.

kisvegabor, 2022-05-30:
Yes, it should be optional. Maybe having a checkbox like "RGB565-A8 format for 16 bit" when `TRUE_COLOR_ALPHA` is selected.

embeddedt, 2022-06-06:
I have realized that we currently have an R-B-swapped option for ARGB8565. Are we going to need that feature for RGB565A8 as well?

GorgonMeducer, 2022-06-06:
@embeddedt No.

In fact, my suggestion is that the high-low-byte-swapping feature should be removed from the LVGL rendering and only do it right before calling disp_flush.

kisvegabor, 2022-06-07:
> In fact, my suggestion is that the high-low-byte-swapping feature should be removed from the LVGL rendering and only do it right before calling disp_flush.

It's faster to render with the desired (swapped) format than swapping as a separate step. I haven't measured the performance cost of manual swapping though. Probably we can find a good algorithm which work on 32 bit data to swap 2 pixels at once.

Counting with 320x240=76,800 pixels, and 2 instructions per pixel on a 80MHz MCU it's: 76,800x2/80,000,000 = ~2ms. 
30FPS->28.5FPS, so not big deal. And it should be slightly faster to render in normal  (not swapped) format. 

I added not a to https://github.com/lvgl/lvgl/issues/3298 to consider removing it.

GorgonMeducer, 2022-06-07:
@kisvegabor the biggest problem is, if we have a swapped image, and we need to apply transform, then the decoding overhead will be very high:
1) decoding: swap to get a normal rgb565 pixel
2) transform
3) swap back to a swapped pixel for LCD flush

If we only use normal rgb565 and do it once just before disp_flush, the swap process in decoding can be removed. When GPU is available, a rgb565a8 image can be accelerated directly without decoding.

kisvegabor, 2022-06-09:
Ok, we can even have a callback in `draw_ctx` to do the swapping.  So it can be implemented in assembly too.

Maybe we can call it `buffer_post_process_cb`. It could apply any other effect or conversion too.

kisvegabor, 2022-06-10:
Sounds good! It can be added even for v8.3 as we can still keep `LV_COLOR_16_SWAP` for backward compatibility. 

We need to be careful about placing these new callback(s) into `lv_disp_drv_t` or `lv_draw_ctx_t`. As I see `lv_disp_drv_t` should be responsible for how the rendered image is transferred to the display while `lv_draw_ctx_t` is for how the image is rendered. Another aspect is `lv_disp_drv_t` is board/display/system dependent, while `lv_draw_ctx_t` is only MCU/architecture/software dependent.

For v9 we should review this two struct in this regard.

kisvegabor, 2022-06-23:
I've add an option the select between the two formats.

commit afaa8c93 (2022-06-23): demo(benchmark): add LV_DEMO_BENCHMARK_RGB565A8 option

fixes #3379
commit 7e604417 (2022-09-20): demo(benchmark): add LV_DEMO_BENCHMARK_RGB565A8 option

fixes #3379
closed 2022-06-23 by kisvegabor