lvgl/lvgl #3382: Message box button height fit is broken after v8

luka177, 2022-05-26:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
After v8 mbox got a bit broken and if you have more then one row of buttons it will have very small button size.

### To Reproduce

### Expected behavior
It must automatically fit buttons

### Screenshots or video
[image]

-->

kisvegabor, 2022-05-26:
You can solve it by 
```c
lv_obj_set_height(lv_msgbox_get_btns(mbox1), 80);
```

luka177, 2022-05-27:
Thanks will try, but in lvgl v7 it worked ok even without

pete-pjb, 2022-05-27:
Hi @luka177 ,

Can you confirm if you have changed the value of:

```c #define LV_DPI_DEF ```

to something lower than 130? in 'lv_conf.h'

Kind Regards,

Pete

pete-pjb, 2022-05-27:
Hi @luka177 ,

I have pushed a fix [here](https://github.com/lvgl/lvgl/commit/cb4da38e3ce0ffd67764f79e79c5e7677e3b3777) for this.

If you have reduced the LV_DPI_DEF value to something less than 130 with this patch the buttons will pad correctly.

cc @kisvegabor 

Kind Regards,

Pete

kisvegabor, 2022-05-27:
> Thanks will try, but in lvgl v7 it worked ok even without

We can improve v8 to count the rows and calculates with them.  Now LVGL just set height which makes sens for one row.

> If you have reduced the LV_DPI_DEF value to something less than 130 with this fix the buttons will pad correctly.

I'm not sure it's the right approach. In the original version DPI/10 means 25.4mm/10 = 2.54mm. Which makes sense (IMO) to always set the same  padding size in `mm` as the size of the user's finger is defined in `mm` too. In other words with a very small DPI, I think the user still wants a small extra area around the buttons' text.  Considering dpi=130 in this case would result in a larger than expected button size.

pete-pjb, 2022-06-06:
Hi @kisvegabor ,

Sorry for my late reply I have been away on holiday.  I have reverted my change to the padding, having revisited it I was probably a bit hasty.

Kind Regards,

Pete